wolfSSL/wolfssl #8250: Pkcs11AesCbcEncrypt/Pkcs11AesCbcDecrypt

embetrix, 2024-12-04:
### Version

master

### Description

`Pkcs11AesCbcEncrypt` `Pkcs11AesCbcDecrypt` do not make use of `C_EncryptUpdate` `C_DecryptUpdate` which make them unusable for lager buffer.

ColtonWilley, 2024-12-05:
@embetrix I am hoping to better understand your use case. The code within `Pkcs11AesCbcEncrypt` can of course be changed to use C_EncryptUpdate, but I am not sure how that helps you as I am not aware of any top level interface that takes partial data in an encryptupdate fashion. The most commonly used top level AES-CBC function wc_AesCbcEncrypt() takes all of the data at once, so you will still run into the same problem. 

Please let me know if you can provide any more information to clear this up so we can find a way to accommodate your use case.

embetrix, 2024-12-05:
@ColtonWilley: there are HSMs that have a limited stack size and cannot encrypt/decrypt large data buffer at once, this should be done by chunks using C_EncryptUpdate/C_DecryptUpdate. 
Ideally the chunck size should be set to something small e.g : 1K and be configurable at build time.

ColtonWilley, 2024-12-10:
@embetrix I understand your issue now. I will change the code to use C_EncryptUpdate/C_DecryptUpdate instead with a configurable chunk size. I will let you know on this issue when the PR is up.

closed 2026-03-10 by ColtonWilley