wolfSSL/wolfssl #8347: [Bug]: Fix the typo in aarch64 AES MPULL macro. (was #8346)

embhorn, 2025-01-09:
### Contact Details

@fraggerfox

### Version

@master

### Description

Currently the build fails in FreeBSD aarch64 with the following message

`FreeBSD main-arm64-default-job-07 15.0-CURRENT FreeBSD 15.0-CURRENT 1500030 arm64`

```
wolfcrypt/src/cpuid.c:262:28: error: use of undeclared identifier 'CPUID_AARCH64_FEAT_PMULL'
  262 |             if (features & CPUID_AARCH64_FEAT_PMULL)
      |                            ^
1 error generated.
--- wolfcrypt/src/src_libwolfssl_la-sp_int.lo ---
--- wolfcrypt/src/src_libwolfssl_la-cpuid.lo ---
*** [wolfcrypt/src/src_libwolfssl_la-cpuid.lo] Error code 1

make[2]: stopped making "all-am" in /wrkdirs/usr/ports/security/wolfssl/work/wolfssl-5.7.6
```

SparkiDev, 2025-01-09:
Hi @fraggerfox

I've made a fix for this in the PR: #8348.

Please let us know if this works for you and f there are further issues with the Aarch64 CPU feature identification on FreeBSD.

Thanks,
Sean

fraggerfox, 2025-01-11:
> Hi @fraggerfox
> 
> I've made a fix for this in the PR: #8348.
> 
> Please let us know if this works for you and f there are further issues with the Aarch64 CPU feature identification on FreeBSD.
> 
> Thanks, Sean

Thank you for the quick turn around, the build works fine with this patch, for now I shall include this patch into our build scripts  so that the build failures are fixed until the next release.

SparkiDev, 2025-01-13:
Hi @fraggerfox

The PR has been merged.
If you have any problems with this reopen this ticket.

Thanks,
Seab

PR #8348 (2025-01-10): Aarch64 CPU Id: FreeBSD/OpenBSD fix
# Description

Fix name and flags set.

Fixes #8347

# Testing

Customer tested.

# Checklist

 - [ ] added tests
 - [ ] updated/added doxygen
 - [ ] updated appropriate READMEs
 - [ ] Updated manual and documentation