wolfSSL/wolfssl #8364: [Bug]: Documentation issue with ocsp support page, there doesn't seem to be information on how it operates (e.g. fetching)

ell1e, 2025-01-21:
### Version

latest documentation here: https://www.wolfssl.com/wolfssl-online-certificate-status-protocol-ocsp-support/

### Description

The page for OCSP staping that explains how it works here https://www.wolfssl.com/wolfssl-online-certificate-status-protocol-ocsp-support/ doesn't seem to give any information on how it operates behind the scenes, specifically as to how it obtains the intermediate certificates.

I am assuming it uses its own custom networking and HTTPS stack to do some sort of request to fetch the latest revocation information from somewhere if needed. My deepest apologies if this is incorrect and I'm just misunderstanding OCSP stapling, if this is not how it works then this ticket is likely invalid. (Although perhaps then it could still give you some idea on how to clarify this misunderstanding on the docs page as well.)

If above assumption is true, it leads to the following questions in my head that I think the page linked above should answer:

1. When talking to downstream users with regards to privacy policies and legalities regarding that, it's not obvious what the involved parties here are because it's not clear what machines and networks WolfSSL will contact to get the stapled information. Basically I don't know anymore when enabling this who my program is potentially talking to. A start would be to have the documentation state which call in WolfSSL leads to it prompting what party with what information in the background, and how that party is determined.

2. It can be useful for the application to own any sort of DNS lookups being made. For example, the application may want to use a different DNS resolver if instructed by the user, or different approaches like DNS-over-TLS, and so on. In this case, it would be useful if any DNS lookup WolfSSL might make for these background requests could be hooked into via some sort of callback, and provided by the application instead. The docs page doesn't seem to give any hints whether that is possible, and perhaps it should.

3. It can be useful for the application to own any sort of UDP or TCP or HTTP(S) requests or download being made. For example, the application may be instructed by the user to use a different proxy setup than the system proxy for outgoing requests, or it may need to add headers like some specific user agent to be allowed to bypass some firewall, and so on. Same as point 2, some sort of callback from WolfSSL where the app gets the info on what party it wants to contact and letting the app take over this request, would be potentially helpful here. If the protocol is too complicated for the user to tap into, it would be helpful to at least be able to pass in a custom I/O socket interface to control or own the connection instead. The docs page doesn't seem to have any potential input on this either.

4. For legal or technical purposes, an application may be expected to be able to log all networking activity in some place, including the exact targets involved. This may be in a custom log format controlled by the app. Again for point 2 & 3, it would in this case be kind of required to hook into these background requests for the app to understand what is happening.

My apologies again if I'm misunderstanding how OCSP stapling works and this is all moot. In this case, feel free to simply close this ticket.

### Reproduction steps

1. Open up https://www.wolfssl.com/wolfssl-online-certificate-status-protocol-ocsp-support/ to read about OCSP stapling and how to interact with it

2. Figure out that you can't really see any info about 1. what parties it is contacting, 2. which exact call (e.g. opening up a certain certificate) is going to prompt those contact attempts, 3. how to proxy or otherwise own or wrap the sockets or system dns lookup queries involved in those operations, 4. or to even merely know what requests are happening for the purpose of logging.

### Relevant log output

```shell

```

anhu, 2025-01-21:
Hi @ell1e , 

This is Anthony again.  I believe you are on the right track.  However, I think an example would be much better than documentation in this case.  Please have a look at  https://github.com/wolfSSL/wolfssl-examples/blob/master/ocsp/ocsp_nonblock/ocsp_nonblock.c .  I think that will clarify all your questions.  Please let me know if you need further clarification. 

Warm regards, Anthony

ell1e, 2025-01-21:
Are the callbacks in the example needed or optional? I'm asking since the documentation page makes it look like they wouldn't be needed and that there is some default callback handling this if no user callback is said, but since the callbacks don't seem to be mentioned at all I'm not sure. Perhaps the documentation page could feature a paragraph outlining the general approach with callbacks and their use and when they are needed, and then link to the example in question?

anhu, 2025-01-21:
Those callbacks are optional.  You can see that in the code for `wolfSSL_CertManagerEnableOCSP()` default values for those  callbacks are set. 

```
    #ifndef WOLFSSL_USER_IO
        /* Set built-in OCSP lookup. */
        cm->ocspIOCb = EmbedOcspLookup;
        cm->ocspRespFreeCb = EmbedOcspRespFree;
        cm->ocspIOCtx = cm->heap;
    #endif /* WOLFSSL_USER_IO */
```

Warm regards, Anthony

anhu, 2025-01-22:
@ell1e , 

I think I've answered your questions to your satisfaction.  I'll now close this issue. 
In another issue, you mentioned licensing issues. We'd love to know more about your thoughts on this. Please do send a message to facts@wolfssl.com to have a private and confidential conversation about this. 

Warm regards, Anthony

closed 2025-01-22 by anhu