wolfSSL/wolfssl #8372: 'config.h' file not found

Nolannn, 2025-01-24:
### Version

5.7.6

### Description

When I executed make, a 'config.h' file not found error occurred, but I executed it correctly./ autogen.sh and./ configure, prompt when making, config.status: creating config.h, config.status: config.h is unchanged, but I didn't find this file. What is the reason?
this is my config:
Configuration summary for wolfssl version 5.7.6
   * Installation prefix:        /usr/local
   * System type:                apple-darwin23.6.0
   * Host CPU:                   aarch64
   * C Compiler:                 gcc
   * C Flags:                       -Werror -Wno-pragmas -Wall -Wextra -Wunknown-pragmas --param=ssp-buffer-size=1 -Waddress -Warray-bounds -Wbad-function-cast -Wchar-subscripts -Wcomment -Wfloat-equal -Wformat-security -Wformat=2 -Wmissing-field-initializers -Wmissing-noreturn -Wmissing-prototypes -Wnested-externs -Woverride-init -Wpointer-arith -Wpointer-sign -Wshadow -Wshorten-64-to-32 -Wsign-compare -Wstrict-overflow=1 -Wstrict-prototypes -Wswitch-enum -Wundef -Wunused -Wunused-result -Wunused-variable -Wwrite-strings -fwrapv

Nolannn, 2025-01-24:
I think the key reason is

[image]

anhu, 2025-01-24:
Hi @Nolannn , 

Can you send us a log showing us all the commands you execute and all the output from those commands? 

Can you also let us know a bit about yourself and your project?  Where are you located?  Is this project out of academic, professional or personal interest?  What are your goals? 

Warm regards, Anthony