wolfSSL/wolfssl #8374: Unable to bring PQC to an Arduino sketch

jhhdez, 2025-01-24:
### Version

5.7.4

### Description

Hi folks!!!

I would appreciate any help or guidance. I am working on Arduino IDE with the official library wolfssl, which was installed using the Arduino Library Manager. However, I am able to use  <wolfssl/wolfcrypt/aes.h> but not <wolfssl/wolfcrypt/wc_kyber.h>. I do not know whether the wolfssl in Arduino was compiled with libqos or not, or if the way to work with PQC is different from what I using. 

I will leave my code below. This is just a simple example to set things up and build upon.

`#define WOLFSSL_EXPERIMENTAL_SETTINGS
#define WOLFSSL_HAVE_KYBER
#define WOLFSSL_WC_KYBER
#define WOLFSSL_SHAKE256
#define WOLFSSL_SHA3

#include <wolfssl.h>
#include <wolfssl/wolfcrypt/wc_kyber.h>

KyberKey kyberkey;

void setup() {
    Serial.begin(115200);
    Serial.println("WolfSSL library is included successfully!");

    // Initialize WolfSSL
    Serial.println("Initializing WolfSSL...");
    if (wolfSSL_Init() != WOLFSSL_SUCCESS) {
        Serial.println("Failed to initialize WolfSSL");
        return;
    }
    Serial.println("WolfSSL initialized successfully");

    // Initialize Kyber
    Serial.println("Initializing Kyber...");
  
    // Set key type to KYBER512, KYBER768, or KYBER1024
    kyberkey.type = KYBER512;
    kyber_init();

}

void loop() {
    // Your code here
}`

The error that I am getting is:

user/.arduino15/packages/esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld:user/.cache/arduino/sketches/83F0BC3D0F174449E307A31B46415E99/sketch/first_test_pqc_test.ino.cpp.o:(.literal._Z5setupv+0x18): undefined reference to `kyber_init'
user/.arduino15/packages/esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld:user/.cache/arduino/sketches/83F0BC3D0F174449E307A31B46415E99/sketch/first_test_pqc_test.ino.cpp.o: in function `_Z5setupv':
user/Arduino/first_test_pqc_test/first_test_pqc_test.ino:23: undefined reference to `kyber_init'
collect2: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1

Thanks in advance !!

anhu, 2025-01-24:
Hi, 

You're trying to call `kyber_init()` but that is not considered a public API; it is marked with `WOLFSSL_LOCAL`.  You'll want to use the following instead: `wc_KyberKey_Init()`

Thanks for using wolfSSL!  Can you please let us know a bit about yourself and your project?  Where are you located?  Is this out of personal, academic or professional interest? What are you trying to make?  What are your goals? 

Warm regards, Anthony

jhhdez, 2025-01-24:
> Hi,
>
> You're trying to call `kyber_init()` but that is not considered a public API; it is marked with `WOLFSSL_LOCAL`. You'll want to use the following instead: `wc_KyberKey_Init()`
>
> Thanks for using [...]

Thank you for your fast reply @anhu !!

I am trying to do some tests with the fantastic wolfssl library for academics purposes. I am deeply passionate about cryptography ;) and IoT.

Your response help me to understand better the library, but unfortunately a similar error arise when I try to use wc_KyberKey_Init(). If wc_KyberKey_Init() is in kyber.h and I am adding it as a header to my Arduino sketch, why I get:

/Users/xxxxx/Library/Arduino15/packages/esp32/tools/esp-x32/2405/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /Users/xxxxx/Library/Caches/arduino/sketches/FD34271CDEDC16EB7F5BB38A293FFDAA/sketch/test.ino.cpp.o:(.literal._Z5setupv+0x8): undefined reference to `wc_KyberKey_Init'
/Users/xxxxx/Library/Arduino15/packages/esp32/tools/esp-x32/2405/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /Users/xxxxx/Library/Caches/arduino/sketches/FD34271CDEDC16EB7F5BB38A293FFDAA/sketch/test.ino.cpp.o: in function `_Z5setupv':
/Users/xxxxx/Documents/Arduino/test/test.ino:24:(.text._Z5setupv+0x36): undefined reference to `wc_KyberKey_Init'
collect2: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1

This is my code:

`#define WOLFSSL_EXPERIMENTAL_SETTINGS
#define WOLFSSL_HAVE_KYBER
#define WOLFSSL_WC_KYBER
#define WOLFSSL_SHAKE256
#define WOLFSSL_SHA3

#include <wolfssl.h>
#include <wolfssl/wolfcrypt/kyber.h>
#include <wolfssl/wolfcrypt/wc_kyber.h>

KyberKey myKey;

void setup() {
  Serial.begin(115200);
  
  // Initialize WolfSSL
  if (wolfSSL_Init() != WOLFSSL_SUCCESS) {
      return;
  }

  // Call to wc_KyberKey_Init
  if (wc_KyberKey_Init(KYBER512, &myKey, NULL, -1) != 0) {
      return;
  }
}

void loop() {
}`

Thanks!!!!

anhu, 2025-01-25:
That is a linker error. That means when you finally combine your object files into the finally binary, its can't find that function. You need to tell your linker tool where to find the wolfssl library.

jhhdez, 2025-01-25:
Thanks for answer!! 

Yes, it seems to be what you are saying, but I can not understand why. I just installed the oficial wolfssl library by Arduino Library Manager.

gojimmypi, 2025-01-25:
Hi @jhhdez  - That's cool you are looking into PQ on the Arduino! 

The wolfSSL settings *must* be visible to *all* wolfSSL source code, not just the main sketch.

The is accomplished with the wolfSSL `user_settings.h` file. 

Please move your settings from the sketch, to the `user_settings.h` in the wolfSSL Arduino library directory. 

The exact path varies. See the [Arduino docs](https://docs.arduino.cc/retired/hacking/software/Libraries/).

In my case, I removed these lines from your sample sketch:

```
#define WOLFSSL_EXPERIMENTAL_SETTINGS
#define WOLFSSL_HAVE_KYBER
#define WOLFSSL_WC_KYBER
#define WOLFSSL_SHAKE256
#define WOLFSSL_SHA3
```

And as I am using Windows, I added those lines to this file  in my installed wolfSSL Arduino library directory:

```
C:\Users\%USERNAME%\Documents\Arduino\libraries\wolfssl\src\user_settings.h
```

Give that a try and let me know how it goes.

I look forward to hearing more about your progress using PQ on various Arduino platforms! Please do not hesitate to reach out if you encounter any other problems.

jhhdez, 2025-01-25:
Hi @gojimmypi !!!

Thank you for your answer !!

Finally, that was what I did and can confirm that works. I am in Mac. 

Thanks for the support !!

closed 2025-01-25 by jhhdez

gojimmypi, 2025-01-26:
@jhhdez yay! that's so cool you have PQC working on the Arduino environment!

I've created https://github.com/wolfSSL/wolfssl/pull/8381 that will hopefully clarify where & how to make wolfSSL user configuration changes.