wolfSSL/wolfssl #8404: [Bug]:  WolfSSL code references outdated OQS constants

rishabhamar, 2025-01-31:
### Contact Details

rishabhamar1234@gmail.com

### Version

v5.7.6-stable

### Description

The issue states that the WolfSSL library is referencing outdated liboqs constants line **_ipd**, but my wolfssl version is v5.7.6-stable, my liboqs library version is 0.12.0 and OQS's OpenSSL Fork version is OQS-OpenSSL-1_1_1-stable-snapshot-2023-07

### Reproduction steps

1. git clone https://github.com/wolfssl/wolfssl
2. cd wolfssl
3. git checkout v5.7.6-stable
4. ./autogen.sh
5. ./configure --enable-lighty --enable-curl --enable-experimental --with-liboqs
6. make all

### Relevant log output

```shell
root@9907783168cf:~/oqs/wolfssl# sudo make check
      make -j13  check-recursive
      make[1]: Entering directory '/root/oqs/wolfssl'
      make[2]: Entering directory '/root/oqs/wolfssl'
      make[2]: warning: -j13 forced in submake: resetting jobserver mode.
        CC       wolfcrypt/benchmark/benchmark.o
        CC       wolfcrypt/src/src_libwolfssl_la-asn.lo
        CC       wolfcrypt/src/src_libwolfssl_la-dilithium.lo
        CC       wolfcrypt/src/src_libwolfssl_la-sphincs.lo
        CC       wolfcrypt/src/src_libwolfssl_la-ext_kyber.lo
        CC       wolfcrypt/src/port/liboqs/src_libwolfssl_la-liboqs.lo
        CC       src/libwolfssl_la-internal.lo
        CC       src/libwolfssl_la-wolfio.lo
        CC       src/libwolfssl_la-keys.lo
        CC       src/libwolfssl_la-ssl.lo
        CC       src/libwolfssl_la-tls.lo
        CC       src/libwolfssl_la-tls13.lo
        CC       src/libwolfssl_la-ocsp.lo
      In file included from wolfcrypt/benchmark/benchmark.c:206:
      ./wolfssl/wolfcrypt/dilithium.h:539:39: error: ‘OQS_SIG_ml_dsa_87_ipd_length_public_key’ undeclared here (not in a function); did you mean ‘OQS_SIG_ml_dsa_87_length_public_key’?
        539 | #define DILITHIUM_LEVEL5_PUB_KEY_SIZE OQS_SIG_ml_dsa_87_ipd_length_public_key
            |                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      ./wolfssl/wolfcrypt/dilithium.h:578:36: note: in expansion of macro ‘DILITHIUM_LEVEL5_PUB_KEY_SIZE’
        578 | #define DILITHIUM_MAX_PUB_KEY_SIZE DILITHIUM_LEVEL5_PUB_KEY_SIZE
            |                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      ./wolfssl/wolfcrypt/dilithium.h:636:12: note: in expansion of macro ‘DILITHIUM_MAX_PUB_KEY_SIZE’
        636 |     byte p[DILITHIUM_MAX_PUB_KEY_SIZE];
            |            ^~~~~~~~~~~~~~~~~~~~~~~~~~
      ./wolfssl/wolfcrypt/dilithium.h:537:39: error: ‘OQS_SIG_ml_dsa_87_ipd_length_secret_key’ undeclared here (not in a function); did you mean ‘OQS_SIG_ml_dsa_87_length_secret_key’?
        537 | #define DILITHIUM_LEVEL5_KEY_SIZE     OQS_SIG_ml_dsa_87_ipd_length_secret_key
            |                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      ./wolfssl/wolfcrypt/dilithium.h:576:36: note: in expansion of macro ‘DILITHIUM_LEVEL5_KEY_SIZE’
        576 | #define DILITHIUM_MAX_KEY_SIZE     DILITHIUM_LEVEL5_KEY_SIZE
            |                                    ^~~~~~~~~~~~~~~~~~~~~~~~~
      ./wolfssl/wolfcrypt/dilithium.h:637:12: note: in expansion of macro ‘DILITHIUM_MAX_KEY_SIZE’
        637 |     byte k[DILITHIUM_MAX_KEY_SIZE];
            |            ^~~~~~~~~~~~~~~~~~~~~~
      wolfcrypt/benchmark/benchmark.c: In function ‘bench_dilithiumKeySign’:
      ./wolfssl/wolfcrypt/dilithium.h:538:39: error: ‘OQS_SIG_ml_dsa_87_ipd_length_signature’ undeclared (first use in this function); did you mean ‘OQS_SIG_ml_dsa_87_length_signature’?
        538 | #define DILITHIUM_LEVEL5_SIG_SIZE     OQS_SIG_ml_dsa_87_ipd_length_signature
            |                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      ./wolfssl/wolfcrypt/dilithium.h:577:36: note: in expansion of macro ‘DILITHIUM_LEVEL5_SIG_SIZE’
        577 | #define DILITHIUM_MAX_SIG_SIZE     DILITHIUM_LEVEL5_SIG_SIZE
            |                                    ^~~~~~~~~~~~~~~~~~~~~~~~~
      wolfcrypt/benchmark/benchmark.c:13941:16: note: in expansion of macro ‘DILITHIUM_MAX_SIG_SIZE’
      13941 |     byte   sig[DILITHIUM_MAX_SIG_SIZE];
[... 602 lines cut ...]
      make[2]: *** [Makefile:7862: wolfcrypt/src/src_libwolfssl_la-asn.lo] Error 1
      ./src/ssl_load.c: In function 'ProcessBufferCertPublicKey':
      ./wolfssl/wolfcrypt/dilithium.h:519:39: error: 'OQS_SIG_ml_dsa_44_ipd_length_secret_key' undeclared (first use in this function); did you mean 'OQS_SIG_ml_dsa_44_length_secret_key'?
        519 | #define DILITHIUM_LEVEL2_KEY_SIZE     OQS_SIG_ml_dsa_44_ipd_length_secret_key
            |                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      ./src/ssl_load.c:1711:21: note: in expansion of macro 'DILITHIUM_LEVEL2_KEY_SIZE'
       1711 |             keySz = DILITHIUM_LEVEL2_KEY_SIZE;
            |                     ^~~~~~~~~~~~~~~~~~~~~~~~~
      ./wolfssl/wolfcrypt/dilithium.h:528:39: error: 'OQS_SIG_ml_dsa_65_ipd_length_secret_key' undeclared (first use in this function); did you mean 'OQS_SIG_ml_dsa_65_length_secret_key'?
        528 | #define DILITHIUM_LEVEL3_KEY_SIZE     OQS_SIG_ml_dsa_65_ipd_length_secret_key
            |                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      ./src/ssl_load.c:1721:21: note: in expansion of macro 'DILITHIUM_LEVEL3_KEY_SIZE'
       1721 |             keySz = DILITHIUM_LEVEL3_KEY_SIZE;
            |                     ^~~~~~~~~~~~~~~~~~~~~~~~~
      src/ssl.c: At top level:
      src/ssl.c:7374:5: error: expected identifier or '(' before 'if'
       7374 |     if (ret == 0 && type == PRIVATEKEY_TYPE) {
            |     ^~
      make[2]: *** [Makefile:8261: src/libwolfssl_la-internal.lo] Error 1
      make[2]: *** [Makefile:8282: src/libwolfssl_la-ssl.lo] Error 1
      make[2]: Leaving directory '/root/oqs/wolfssl'
      make[1]: *** [Makefile:9077: check-recursive] Error 1
      make[1]: Leaving directory '/root/oqs/wolfssl'
      make: *** [Makefile:9546: check] Error 2
```

anhu, 2025-01-31:
Hi @rishabhamar , 

Thanks for trying wolfSSL!!  Fortunately, we have recently added our own implementation of ML-DSA (Dilithium) and ML-KEM (Kyber).  

I noticed you are using the following configure command: 

`./configure --enable-lighty --enable-curl --enable-experimental --with-liboqs`

Please change that to: 

`./configure --enable-lighty --enable-curl --enable-experimental --enable-kyber --enable-dilithium`

Let me know how that goes. 
Warm regards, Anthony

rishabhamar, 2025-01-31:
Thank you for replying 
I was trying to implement PQC algorithms through wolfssl and earlier I tried the wolfssl implementation but it didn't work with python and even ctype then I switched to liboqs implementation by following your tutorial on YouTube (https://m.youtube.com/watch?v=KrNTYniz-ek&t=1727s&pp=ygULV29sZnNzbCBwcWM%3D). In this video they also mentioned the patch by you for lighttpd_ssl.privatekey patch but that didn't work either.
It would be great if you could guide in how to use pqc algorithms for certificate generation and key exchanges for TLS connection.
I am trying to build a software with detects that if a particular website aor domain is using post quantum algorithms or not and for that I was trying the wolfssl and liboqs.
Thank You Tha

rishabhamar, 2025-01-31:
Thank you for replying
I was trying to implement PQC algorithms through wolfssl and earlier I
tried the wolfssl implementation but it didn't work with python and even
ctype then I switched to liboqs implementation by following your tutorial
on YouTube (
https://m.youtube.com/watch?v=KrNTYniz-ek&t=1727s&pp=ygULV29sZnNzbCBwcWM%3D).
In this video they also mentioned the patch by you for
lighttpd_ssl.privatekey patch but that didn't work either.
It would be great if you could guide in how to use pqc algorithms for
certificate generation and key exchanges for TLS connection.
I am trying to build a software with detects that if a particular website
aor domain is using post quantum algorithms or not and for that I was
trying the wolfssl and liboqs.
Thank You

anhu, 2025-01-31:
In that video, I specify 5.7.0-stable for wolfssl and various other versions for software.  Please go back and watch that video very carefully. 

Warm regards, Anthony

rishabhamar, 2025-02-01:
@anhu I tried following the tutorial with exact same steps in the videos and still got error

    crypto/ec/oqs_meth.c: In function 'get_oqs_alg_name':
    crypto/ec/oqs_meth.c:204:14: error: 'OQS_SIG_alg_dilithium_2_aes' undeclared (first use in this function); did you mean 'OQS_SIG_alg_dilithium_2'?
      204 |       return OQS_SIG_alg_dilithium_2_aes;
          |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~
          |              OQS_SIG_alg_dilithium_2
    crypto/ec/oqs_meth.c:204:14: note: each undeclared identifier is reported only once for each function it appears in
    crypto/ec/oqs_meth.c:207:14: error: 'OQS_SIG_alg_dilithium_3_aes' undeclared (first use in this function); did you mean 'OQS_SIG_alg_dilithium_3'?
      207 |       return OQS_SIG_alg_dilithium_3_aes;
          |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~
          |              OQS_SIG_alg_dilithium_3
    crypto/ec/oqs_meth.c:210:14: error: 'OQS_SIG_alg_dilithium_5_aes' undeclared (first use in this function); did you mean 'OQS_SIG_alg_dilithium_5'?
      210 |       return OQS_SIG_alg_dilithium_5_aes;
          |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~
          |              OQS_SIG_alg_dilithium_5
    crypto/ec/oqs_meth.c:221:14: error: 'OQS_SIG_alg_picnic_L1_full' undeclared (first use in this function)
      221 |       return OQS_SIG_alg_picnic_L1_full;
          |              ^~~~~~~~~~~~~~~~~~~~~~~~~~
    crypto/ec/oqs_meth.c:225:14: error: 'OQS_SIG_alg_picnic3_L1' undeclared (first use in this function)
      225 |       return OQS_SIG_alg_picnic3_L1;
          |              ^~~~~~~~~~~~~~~~~~~~~~
    crypto/ec/oqs_meth.c:229:14: error: 'OQS_SIG_alg_rainbow_I_classic' undeclared (first use in this function)
      229 |       return OQS_SIG_alg_rainbow_I_classic;
          |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    crypto/ec/oqs_meth.c:232:14: error: 'OQS_SIG_alg_rainbow_V_classic' undeclared (first use in this function)
      232 |       return OQS_SIG_alg_rainbow_V_classic;
          |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    crypto/ec/oqs_meth.c:236:14: error: 'OQS_SIG_alg_sphincs_haraka_128f_robust' undeclared (first use in this function); did you mean 'OQS_SIG_alg_sphincs_shake_128f_simple'?
      236 |       return OQS_SIG_alg_sphincs_haraka_128f_robust;
          |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
          |              OQS_SIG_alg_sphincs_shake_128f_simple
    crypto/ec/oqs_meth.c:240:14: error: 'OQS_SIG_alg_sphincs_sha256_128f_robust' undeclared (first use in this function); did you mean 'OQS_SIG_alg_sphincs_sha2_128f_simple'?
      240 |       return OQS_SIG_alg_sphincs_sha256_128f_robust;
          |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
          |              OQS_SIG_alg_sphincs_sha2_128f_simple
    crypto/ec/oqs_meth.c:244:14: error: 'OQS_SIG_alg_sphincs_shake256_128f_robust' undeclared (first use in this function); did you mean 'OQS_SIG_alg_sphincs_shake_128f_simple'?
      244 |       return OQS_SIG_alg_sphincs_shake256_128f_robust;
          |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
          |              OQS_SIG_alg_sphincs_shake_128f_simple
    crypto/ec/oqs_meth.c:274:14: error: 'OQS_KEM_alg_ntru_hps2048509' undeclared (first use in this function)
      274 |       return OQS_KEM_alg_ntru_hps2048509;
          |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~
    crypto/ec/oqs_meth.c:277:14: error: 'OQS_KEM_alg_ntru_hps2048677' undeclared (first use in this function)
      277 |       return OQS_KEM_alg_ntru_hps2048677;
          |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~
    crypto/ec/oqs_meth.c:280:14: error: 'OQS_KEM_alg_ntru_hps4096821' undeclared (first use in this function)
      280 |       return OQS_KEM_alg_ntru_hps4096821;
          |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~
    crypto/ec/oqs_meth.c:283:14: error: 'OQS_KEM_alg_ntru_hrss701' undeclared (first use in this function)
      283 |       return OQS_KEM_alg_ntru_hrss701;
          |              ^~~~~~~~~~~~~~~~~~~~~~~~
    crypto/ec/oqs_meth.c:286:14: error: 'OQS_KEM_alg_saber_lightsaber' undeclared (first use in this function)
      286 |       return OQS_KEM_alg_saber_lightsaber;
          |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
    crypto/ec/oqs_meth.c:289:14: error: 'OQS_KEM_alg_saber_saber' undeclared (first use in this function); did you mean 'OQS_KEM_alg_kyber_768'?
      289 |       return OQS_KEM_alg_saber_saber;
          |              ^~~~~~~~~~~~~~~~~~~~~~~
          |              OQS_KEM_alg_kyber_768
    crypto/ec/oqs_meth.c:292:14: error: 'OQS_KEM_alg_saber_firesaber' undeclared (first use in this function)
      292 |       return OQS_KEM_alg_saber_firesaber;
          |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~
    crypto/ec/oqs_meth.c:295:14: error: 'OQS_KEM_alg_sidh_p434' undeclared (first use in this function); did you mean 'OQS_KEM_alg_bike_l3'?
      295 |       return OQS_KEM_alg_sidh_p434;
          |              ^~~~~~~~~~~~~~~~~~~~~
          |              OQS_KEM_alg_bike_l3
    crypto/ec/oqs_meth.c:298:14: error: 'OQS_KEM_alg_sidh_p503' undeclared (first use in this function); did you mean 'OQS_KEM_alg_bike_l3'?
      298 |       return OQS_KEM_alg_sidh_p503;
          |              ^~~~~~~~~~~~~~~~~~~~~
          |              OQS_KEM_alg_bike_l3
[... 38 lines cut ...]
          |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
          |              OQS_KEM_alg_ntruprime_sntrup761
[... cut ...]
    crypto/ec/oqs_meth.c:358:14: error: 'OQS_KEM_alg_ntruprime_sntrup857' undeclared (first use in this function); did you mean 'OQS_KEM_alg_ntruprime_sntrup761'?
      358 |       return OQS_KEM_alg_ntruprime_sntrup857;
          |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
          |              OQS_KEM_alg_ntruprime_sntrup761
    make[1]: *** [Makefile:3139: crypto/ec/oqs_meth.o] Error 1
    make[1]: Leaving directory '/root/oqs/liboqs/openssl'
    make: *** [Makefile:174: all] Error 2

Here is the full log file

[wolfssl.txt](https://github.com/user-attachments/files/18628075/wolfssl.txt)

rishabhamar, 2025-02-01:
@anhu I tried following the steps from the tutorial video but i still got the error (all the versions are according to the video).

  crypto/ec/oqs_meth.c: In function 'get_oqs_alg_name':
  crypto/ec/oqs_meth.c:204:14: error: 'OQS_SIG_alg_dilithium_2_aes' undeclared (first use in this function); did you mean 'OQS_SIG_alg_dilithium_2'?
    204 |       return OQS_SIG_alg_dilithium_2_aes;
        |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~
        |              OQS_SIG_alg_dilithium_2
  crypto/ec/oqs_meth.c:204:14: note: each undeclared identifier is reported only once for each function it appears in
  crypto/ec/oqs_meth.c:207:14: error: 'OQS_SIG_alg_dilithium_3_aes' undeclared (first use in this function); did you mean 'OQS_SIG_alg_dilithium_3'?
    207 |       return OQS_SIG_alg_dilithium_3_aes;
        |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~
        |              OQS_SIG_alg_dilithium_3
  crypto/ec/oqs_meth.c:210:14: error: 'OQS_SIG_alg_dilithium_5_aes' undeclared (first use in this function); did you mean 'OQS_SIG_alg_dilithium_5'?
    210 |       return OQS_SIG_alg_dilithium_5_aes;
        |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~
        |              OQS_SIG_alg_dilithium_5
  crypto/ec/oqs_meth.c:221:14: error: 'OQS_SIG_alg_picnic_L1_full' undeclared (first use in this function)
    221 |       return OQS_SIG_alg_picnic_L1_full;
        |              ^~~~~~~~~~~~~~~~~~~~~~~~~~
  crypto/ec/oqs_meth.c:225:14: error: 'OQS_SIG_alg_picnic3_L1' undeclared (first use in this function)
    225 |       return OQS_SIG_alg_picnic3_L1;
        |              ^~~~~~~~~~~~~~~~~~~~~~
  crypto/ec/oqs_meth.c:229:14: error: 'OQS_SIG_alg_rainbow_I_classic' undeclared (first use in this function)
    229 |       return OQS_SIG_alg_rainbow_I_classic;
        |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  crypto/ec/oqs_meth.c:232:14: error: 'OQS_SIG_alg_rainbow_V_classic' undeclared (first use in this function)
    232 |       return OQS_SIG_alg_rainbow_V_classic;
        |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  crypto/ec/oqs_meth.c:236:14: error: 'OQS_SIG_alg_sphincs_haraka_128f_robust' undeclared (first use in this function); did you mean 'OQS_SIG_alg_sphincs_shake_128f_simple'?
    236 |       return OQS_SIG_alg_sphincs_haraka_128f_robust;
        |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        |              OQS_SIG_alg_sphincs_shake_128f_simple
  crypto/ec/oqs_meth.c:240:14: error: 'OQS_SIG_alg_sphincs_sha256_128f_robust' undeclared (first use in this function); did you mean 'OQS_SIG_alg_sphincs_sha2_128f_simple'?
    240 |       return OQS_SIG_alg_sphincs_sha256_128f_robust;
        |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        |              OQS_SIG_alg_sphincs_sha2_128f_simple
  crypto/ec/oqs_meth.c:244:14: error: 'OQS_SIG_alg_sphincs_shake256_128f_robust' undeclared (first use in this function); did you mean 'OQS_SIG_alg_sphincs_shake_128f_simple'?
    244 |       return OQS_SIG_alg_sphincs_shake256_128f_robust;
        |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        |              OQS_SIG_alg_sphincs_shake_128f_simple
  crypto/ec/oqs_meth.c:274:14: error: 'OQS_KEM_alg_ntru_hps2048509' undeclared (first use in this function)
    274 |       return OQS_KEM_alg_ntru_hps2048509;
        |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~
  crypto/ec/oqs_meth.c:277:14: error: 'OQS_KEM_alg_ntru_hps2048677' undeclared (first use in this function)
    277 |       return OQS_KEM_alg_ntru_hps2048677;
        |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~
  crypto/ec/oqs_meth.c:280:14: error: 'OQS_KEM_alg_ntru_hps4096821' undeclared (first use in this function)
    280 |       return OQS_KEM_alg_ntru_hps4096821;
        |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~
  crypto/ec/oqs_meth.c:283:14: error: 'OQS_KEM_alg_ntru_hrss701' undeclared (first use in this function)
    283 |       return OQS_KEM_alg_ntru_hrss701;
        |              ^~~~~~~~~~~~~~~~~~~~~~~~
  crypto/ec/oqs_meth.c:286:14: error: 'OQS_KEM_alg_saber_lightsaber' undeclared (first use in this function)
    286 |       return OQS_KEM_alg_saber_lightsaber;
        |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
  crypto/ec/oqs_meth.c:289:14: error: 'OQS_KEM_alg_saber_saber' undeclared (first use in this function); did you mean 'OQS_KEM_alg_kyber_768'?
    289 |       return OQS_KEM_alg_saber_saber;
        |              ^~~~~~~~~~~~~~~~~~~~~~~
        |              OQS_KEM_alg_kyber_768
  crypto/ec/oqs_meth.c:292:14: error: 'OQS_KEM_alg_saber_firesaber' undeclared (first use in this function)
    292 |       return OQS_KEM_alg_saber_firesaber;
        |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~
  crypto/ec/oqs_meth.c:295:14: error: 'OQS_KEM_alg_sidh_p434' undeclared (first use in this function); did you mean 'OQS_KEM_alg_bike_l3'?
    295 |       return OQS_KEM_alg_sidh_p434;
        |              ^~~~~~~~~~~~~~~~~~~~~
        |              OQS_KEM_alg_bike_l3
  crypto/ec/oqs_meth.c:298:14: error: 'OQS_KEM_alg_sidh_p503' undeclared (first use in this function); did you mean 'OQS_KEM_alg_bike_l3'?
    298 |       return OQS_KEM_alg_sidh_p503;
        |              ^~~~~~~~~~~~~~~~~~~~~
        |              OQS_KEM_alg_bike_l3
[... 38 lines cut ...]
        |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        |              OQS_KEM_alg_ntruprime_sntrup761
[... cut ...]
  crypto/ec/oqs_meth.c:358:14: error: 'OQS_KEM_alg_ntruprime_sntrup857' undeclared (first use in this function); did you mean 'OQS_KEM_alg_ntruprime_sntrup761'?
    358 |       return OQS_KEM_alg_ntruprime_sntrup857;
        |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        |              OQS_KEM_alg_ntruprime_sntrup761
  make[1]: *** [Makefile:3139: crypto/ec/oqs_meth.o] Error 1
  make[1]: Leaving directory '/root/oqs/openssl'
  make: *** [Makefile:174: all] Error 2

Here is my terminal log file

[wolfssl.txt](https://github.com/user-attachments/files/18628300/wolfssl.txt)