wolfSSL/wolfssl #8430: [Bug]: wc_static_assert used in header files breaks MSVC/VS2019 C++ compilation

vaintroub, 2025-02-10:
### Contact Details

vvaintroub@gmail.com

### Version

5.7.4

### Description

First reported in https://jira.mariadb.org/browse/MDEV-36054

Does not happen on MSVC included into VS2022

> 

D:\a\mroonga\mroonga\mariadb\extra\wolfssl\wolfssl\wolfssl/wolfcrypt/error-crypt.h(321,56): error C2429: language feature 'terse static assert' requires compiler flag '/std:c++17'
--
wc_static_assert((int)WC_LAST_E <= (int)WC_SPAN2_LAST_E);
^
D:\a\mroonga\mroonga\mariadb\extra\wolfssl\wolfssl\wolfssl/wolfcrypt/error-crypt.h(322,51): error C2429: language feature 'terse static assert' requires compiler flag '/std:c++17'
wc_static_assert((int)MIN_CODE_E <= (int)WC_LAST_E);
^
D:\a\mroonga\mroonga\mariadb\extra\wolfssl\wolfssl\wolfssl/wolfcrypt/error-crypt.h(323,61): error C2429: language feature 'terse static assert' requires compiler flag '/std:c++17'
wc_static_assert((int)MIN_CODE_E <= (int)WC_SPAN2_MIN_CODE_E);
^
D:\a\mroonga\mroonga\mariadb\extra\wolfssl\wolfssl\wolfssl/error-ssl.h(244,55): error C2429: language feature 'terse static assert' requires compiler flag '/std:c++17'
wc_static_assert((int)WC_LAST_E <= (int)WOLFSSL_LAST_E);
^

### Reproduction steps

compile a C++ file that uses wolfssl 5.7.4, using MSVC included into Visual Studio 2019 without /std:c++17

### Relevant log output

```shell

```

kareem-wolfssl, 2025-02-11:
Hi @vaintroub ,

We have had issues with static_assert on MSVC/Visual Studio in the past, these issues were resolved in our most recent release 5.7.6.  Please try upgrading and let me know if it helps.

vaintroub, 2025-02-11:
hi @kareem-wolfssl , sorry my mistake. We use the latest 5.7.6, and this error stems directly from it. 
My humble opinion is that maybe wc_asserts do not belong to public headers, because now your have to think whether this 

wc_assert is included into C++, which version of C++, which version of compiler and so on.
The idea behind wc_assert is to catch errors during wolfssl compilation, not to cause errors for product that includes public headers.

Just my interpretation, I could be wrong of course. Anyway, it causes errors in some environments, as I reported.

douzzer, 2025-02-11:
@vaintroub , you can compile with `WC_NO_STATIC_ASSERT` defined in your `CFLAGS` or `CPPFLAGS` to confirm there are no other problems with the wolfSSL headers.

Do you have a particular test clause in mind to detect and accommodate the toolchain version where the problem arise?

By the way, I can't make sense of the `C2429` error code you're seeing.  The compiler documentation says

>If the compiler is in the default C++14 mode and `/permissive-` or `/Zc:static_assert` is specified, it uses `/Zc:static_assert` behavior. However, if it evaluates a `static_assert` in a template body, it also reports off-by-default warning C5254, "language feature 'terse static assert' requires compiler flag '`/std:c++17`'", since this behavior isn't required until C++17.

(https://learn.microsoft.com/en-us/cpp/build/reference/zc-static-assert?view=msvc-170)

So the expected code here is C5254, and it is off by default.

C2429 appears to be something completely different and unrelated (https://learn.microsoft.com/en-us/cpp/error-messages/compiler-errors-1/compiler-error-c2429?view=msvc-170).

Your comments regarding the complicated logic around `wc_static_assert()` in the wolfSSL header are well taken.  The setup for `wc_static_assert()` would be better in `wc_port.h`, but it needs to appear somewhere in the headers, and it needs to work correctly on all supported targets, even if only with dummy definitions that disable the static asserts.

vaintroub, 2025-02-11:
I'm reporting what I have seen, and what other people have seen (MariaDB users see the problem, while we have not, by using a later version of the compiler). I see it with the latest VS2019, and do not see it with VS2022, as I already reported.
The warning of course makes sense, because  you're using C++ static_assert with one argument, if C++ file includes wolfssl header. It is C++17

douzzer, 2025-02-11:
@vaintroub try https://github.com/wolfSSL/wolfssl/pull/8440