wolfSSL/wolfssl #8434: BIO_nwrite0 not implemented

FrothyB, 2025-02-10:
### Version

5.7.6

### Description

The OpenSSL extra compatibility API defines

`#define BIO_nwrite0                     wolfSSL_BIO_nwrite0`

However this is not actually implemented, there is only

`WOLFSSL_API int  wolfSSL_BIO_nwrite(WOLFSSL_BIO *bio, char **buf, int num);`

I have so far worked around this with `BIO_nwrite(.., .., 0);`

However I'm not sure how robust this is. Is it the intended approach? There is also the issue that without `BIO_nwrite0`, it is not clear to me how to cleanly and reliably know how much capacity the BIO has. Are there any plans to implement this other variant? Do you have any guidelines or recommendations? The example BIO code does not cover this at all.

For added context, I am trying to avoid copies when using SSL on top of io_uring and I set the BIOs for the sockets as follows:
```c
BIO_new_bio_pair(&internal_bio, internal_bio_sz, &network_bio, network_bio_sz);
SSL_set_bio(ssl, internal_bio, internal_bio);
```

For receiving, I read into the `network_bio`, then `BIO_nwrite `to register the write, then `SSL_read`. Vice versa I `SSL_write`, then send the contents of the `network_bio`

anhu, 2025-02-10:
Hi, I've assigned my colleague, Jacob, to comment on this.

JacobBarthelmeh, 2025-02-11:
Hi @FrothyB that macro was added back in 2016, likely at that time BIO_nread0 was required but not BIO_nwrite0. I think it was a duplicate, accidentally added to match BIO_nread0 in the header file. There is not currently plans to implement BIO_nwrite0 but I can add it to the feature request list.

Using BIO_nwrite(...,...,0) seems like a good work around. As you mention though -- I do not see a good way to get the BIO's capacity similar to the return value of BIO_nwrite0().

Can you tell us some more about the use case? If wanting to send more details in private contact the support@wolfssl.com email list. Knowing if this is for an open source project or more details in general could help with how we prioritize the feature request.

FrothyB, 2025-02-12:
I'm currently just performing some benchmarking of different approaches to networking and SSL. io_uring is nowadays showing some nice performance improvements in both latency and throughput vs the traditional sockets API, especially on more recent kernels. It's also nicer to use. I wanted to try and get the most out of it, which includes trying different SSL libraries, with wolfSSL showing a lot of promise as a more performant option.

io_uring does not really mesh with the traditional SSL APIs - you select a buffer at the time of preparing the read op in which to get the data (there is also the prepared buffer pool option but doesn't apply in the SSL use case anyway). So to avoid a copy in userspace, I need the SSL layer to give me a buffer to receive directly in to. As far as I'm aware, BIOs and nread/nwrite are the only way to do this, and this approach with the BIO pair is the only one I got working (there is shockingly little documentation or example code online, perhaps I could help by producing something small and self-contained but I am quite busy right now). 

BIO_nwrite0 is needed to do the above properly/cleanly - that said the hack with BIO_nwrite(..,..,0) can work with some caution from the user (e.g. have a network BIO with capacity for at least two frames, receive with a max of one frame size, etc).

To summarize, io_uring seems to me to be the way forward for Linux kernel networking, and BIO_nwrite0 is necesssary to avoid copies while using it and to do so robustly.