wolfSSL/wolfssl #8446: [Bug]: error return and heap use after free in X509 verify

ElinorCoding, 2025-02-13:
### Contact Details

xiangjiaying22@mails.ucas.ac.cn

### Version

commit_hash :=8f131ff3d

### Description

When I tested the x509 certificate verification functions of wolfssl, I found that an input was successfully verified by the wolfssl version with commit hash 8f131ff3d (ret is 0). However, the same input was not verified by OpenSSL or the old version of wolfssl (commit hash 7e01af0), where ret ≠ 0. At the same time, my AddressSanitizer shows a heap-use-after-free issue, which does not occur in the old version of wolfssl.

### Reproduction steps

> it can be easily reproducted by using the file below(The compilation commands have essentially been scripted.)

[bug_reports.zip](https://github.com/user-attachments/files/18779068/bug_reports.zip)

## prepare directory path and source
Extract the compressed package [bug_reports.zip](https://github.com/user-attachments/files/18779068/bug_reports.zip) to a directory, referred to as `yourpath2`. Modify the file `compile_commit_hash.sh` and the` Makefile` to replace `yourpath` with your actual working directory. Then, create the directories `src/` and `build/` in the working directory. In the `src` directory, execute the command `git clone https://github.com/wolfSSL/wolfssl `to clone the repository. Finally, modify the file `compile_commit_hash.sh` to replace `yourpath2` with the directory path where you extracted the compressed package.

## compile and test the error
In the `yourpath2` directory, execute the commands：
```
./compile_commit_hash.sh 8f131ff3d
```
and
```
./compile_commit_hash.sh 7e01af0
```
to compile two versions of wolfssl. 

Then, execute the following commands to generate the executable files for testing the two versions:
```
make verify COMMIT_HASH=8f131ff3d -B
```
and
```
make verify COMMIT_HASH=7e01af0 -B
```

In the `yourpath2` directory, execute the commands:
```
./verify_8f131ff3d error_input
```
and
```
./verify_7e01af0 error_input
```

to observe the different behaviors of wolfssl on certificate verification for the same input. You will also notice the error message `==312090==ERROR: AddressSanitizer: heap-use-after-free`

[image]

[image]

### Relevant log output

```shell
./verify_8f131ff3d error_input 
ret is 0
=================================================================
==311437==ERROR: AddressSanitizer: heap-use-after-free on address 0x620000894150 at pc 0x777b5ff645ba bp 0x7fffcc231690 sp 0x7fffcc231688
```

anhu, 2025-02-13:
Hi @ElinorCoding , 

Wow!!  Thank you so much for this detailed bug report. There is a lot to go through here so please give me some time to try your reproduction steps.  

In the meantime, can you let us know a bit about yourself and your project?  Here at wolfSSL we love learning about how the community uses our code. Some information we'd be interested in: 

- your location.  From your contact email address I assume you are in Beijing, but let me know if I am mistaken. 
- your goals for this project.  What are you trying to achieve? 
- is this project out of professional, personal, or academic interest? 
- is there an institution associated with this project? 
- anything else you care to share about yourself and your project. 

If you are concerned about your privacy and would rather share this information over a private channel, you can send your response to support@wolfssl.com . 

Please stay tuned for the results of my reproduction steps. 

Warm regards, Anthony

anhu, 2025-02-19:
The fix has been merged into the master branch.  As such, I will now close this issue.

closed 2025-02-19 by anhu