wolfSSL/wolfssl #8456: [Bug]: cmake - WOLFSSL_LEAN_TLS broken

jrolli, 2025-02-15:
### Contact Details

code@jrolli.dev

### Version

5.7.6-stable

### Description

Using `WOLFSSL_LEAN_TLS` causes a transient compilations issue when `WOLFSSL_MEMORY=OFF` is not provided.  The compilation will initially fail with the error given in the logs section, but touching any file that would trigger recompilation of WolfSSL will cause subsequent compilations to succeed.  At first glance, `CMakeLists.txt:1133` appears to be a bad assumption as the only difference between runs is that `WOLFSSL_MEMORY=OFF` is in the cache.  Both removing the conditional as well as adding the `list` command in place of this line fix the issue.

### Reproduction steps

1. Create a new directory with the attached files (`CMakeLists.txt` and save C code below as "repro.c" - cannot upload due to GitHub restrictions...)

```c
#include "wolfssl/options.h"
#include "wolfssl/wolfssl.h"

#include <sys/socket.h>

int
main (void)
{
    const SSL_METHOD *method = wolfTLS_client_method();
    SSL_CTX          *ctx    = wolfSSL_CTX_new(method);
    wolfSSL_CTX_free(ctx);
    return EXIT_FAILURE;
}
```

2. Run `cmake -B build -DWOLFSSL_LEAN_TLS=ON .`
3. Run `cmake --build build`

[CMakeLists.txt](https://github.com/user-attachments/files/18805786/CMakeLists.txt)

### Relevant log output

```shell
Undefined symbols for architecture arm64:
  "_wolfSSL_Free", referenced from:
      _wc_HashNew in hash.c.o
      _wc_HashDelete in hash.c.o
      _wc_rng_new in random.c.o
      __InitRng in random.c.o
      _wc_rng_new_ex in random.c.o
      _wc_rng_free in random.c.o
      _wc_FreeRng in random.c.o
      ...
  "_wolfSSL_Malloc", referenced from:
      _wc_HashNew in hash.c.o
      _wc_rng_new in random.c.o
      __InitRng in random.c.o
      _wc_rng_new_ex in random.c.o
      __sp_invmod_mont_ct in sp_int.c.o
      __sp_exptmod_nct in sp_int.c.o
      _wc_AesNew in aes.c.o
      ...
  "_wolfSSL_Realloc", referenced from:
      _wolfSSL_EVP_CipherUpdate_GCM in ssl.c.o
      _wolfSSL_EVP_PKEY_CTX_add1_hkdf_info in ssl.c.o
      _ECC_populate_EVP_PKEY in ssl.c.o
      _wolfssl_asn1_bit_string_grow in ssl.c.o
      _wolfSSL_BIO_set_conn_hostname in ssl.c.o
      _wolfSSL_BUF_MEM_resize in ssl.c.o
      _wolfSSL_X509_set_ext in ssl.c.o
      ...
ld: symbol(s) not found for architecture arm64
cc: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [_deps/wolfssl-build/libwolfssl.43.0.0.dylib] Error 1
make[1]: *** [_deps/wolfssl-build/CMakeFiles/wolfssl.dir/all] Error 2
make: *** [all] Error 2
```

jrolli, 2025-02-15:
I'm willing to create a merge request for either of the fixes, but I lack the larger context of why this comment was made (and don't know who else is depending on its current behavior).

embhorn, 2025-02-17:
Hi @jrolli 

Thanks for letting us know about the cmake build issue. I've opened a PR to fix this issue here #8460 

Could you tell us a bit about your project using wolfSSL?

Kind regards,
@embhorn - wolfSSL Support

jrolli, 2025-02-17:
@embhorn, the project that raised this issue is an attempt at a lightweight equivalent of `stunnel` and is mostly just some tinkering to get a feel for integrating wolfSSL into a project.  My longer-term side project is to poke at using it with `ngtcp2` for QUIC support to get more hands-on with QUIC and test out some other ideas.

PR #8460 (2025-02-17): Fix cmake lean_tls build
# Description

Cmake build with `-DWOLFSSL_LEAN_TLS=ON` should also add `NO_WOLFSSL_MEMORY`

Fixes #8456 

# Testing

```
wolfssl/build$ cmake .. -DWOLFSSL_LEAN_TLS=ON
wolfssl/build$ cmake --build .
```

# Checklist

 - [ ] added tests
 - [ ] updated/added doxygen
 - [ ] updated appropriate READMEs
 - [ ] Updated manual and documentation
closed 2025-02-17 by dgarske