wolfSSL/wolfssl #8477: [Bug]: Use of #warning in codebase causes error on Windows

res0nance, 2025-02-20:
### Version

v5.7.6

### Description

The codebase has heavy use of `#warning` e.g https://github.com/wolfSSL/wolfssl/blob/93000e5f1405bb44e1eeec1e9714ea87c5d1dbc1/wolfssl/wolfcrypt/settings.h#L349

This usage is invalid on windows as `cl` does not recognize the `#warning` directive. This will result in build failure to what should amount as a warning to the user which seems to be unintended by the developers.

### Relevant log output

```shell

```

embhorn, 2025-02-20:
Hi @res0nance 

We do support building in in VS and with MSVC tools:
https://github.com/wolfSSL/wolfssl/blob/93000e5f1405bb44e1eeec1e9714ea87c5d1dbc1/INSTALL#L29-L31

Could you tell us how you are building the library?

Thanks,
@embhorn - wolfSSL Support

res0nance, 2025-02-20:
> Hi [@res0nance](https://github.com/res0nance)
> 
> We do support building in VS and with MSVC tools:
> 
> [wolfssl/INSTALL](https://github.com/wolfSSL/wolfssl/blob/93000e5f1405bb44e1eeec1e9714ea87c5d1dbc1/INSTALL#L29-L31)
> 
> Lines 29 to 31 in [93000e5](/wolfSSL/wolfssl/commit/93000e5f1405bb44e1eeec1e9714ea87c5d1dbc1)
> 
>  4. Building on Windows 
>   
>      Use the Visual Studio Solution wolfssl64.sln 
> Could you tell us how you are building the library?
> 
> Thanks, [@embhorn](https://github.com/embhorn) - wolfSSL Support

In my case it has to do with importing a built library thru CMake. While the form of building is different. I think the codebase as a whole should properly handle `#warning` there are spots in the codebase where this is handled correctly but in other places it is not.

embhorn, 2025-02-20:
The `#warning` is handled as a `#pragma message()` in a few spots where we expect the compiler to hit. In that particular case, the library built with MSVC tools should always configure with `WOLFSSL_USER_SETTINGS` enabled, so it would be unexpected to hit that line. The application also needs to define `WOLFSSL_USER_SETTINGS` in order use the right wolfSSL configuration.
https://github.com/wolfSSL/wolfssl/blob/93000e5f1405bb44e1eeec1e9714ea87c5d1dbc1/examples/client/client.vcproj#L45

res0nance, 2025-02-26:
> The `#warning` is handled as a `#pragma message()` in a few spots where we expect the compiler to hit. In that particular case, the library built with MSVC tools should always configure with `WOLFSSL_USER_SETTINGS` enabled, so it would be unexpected to hit that line. The application also needs to define `WOLFSSL_USER_SETTINGS` in order use the right wolfSSL configuration.
> 
> [wolfssl/examples/client/client.vcproj](https://github.com/wolfSSL/wolfssl/blob/93000e5f1405bb44e1eeec1e9714ea87c5d1dbc1/examples/client/client.vcproj#L45)
> 
> Line 45 in [93000e5](/wolfSSL/wolfssl/commit/93000e5f1405bb44e1eeec1e9714ea87c5d1dbc1)
> 
>  PreprocessorDefinitions="WIN32;_DEBUG;_CONSOLE;WOLFSSL_USER_SETTINGS"

While compiling with the VS project works. Integrating a built library will cause these errors if all these hidden preprocessor definitions are undefined.