wolfSSL/wolfssl #8478: [Bug]: clang++-18 generates warning: struct has size 0 in C, size 1 in C++ [-Wextern-c-compat]

tatsuhiro-t, 2025-02-20:
### Version

v5.7.6-stable

### Description

When building with clang++-18, it emits warning:

```
<...>/include/wolfssl/wolfcrypt/hash.h:122:5: warning: struct has size 0 in C, size 1 in C++ [-Wextern-c-compat]
```

Can we ignore this warning?

### Reproduction steps

wolfssl.cc:
```cxx
#include <wolfssl/options.h>
#include <wolfssl/ssl.h>

int main(void) {
  WOLFSSL *ssl;
  (void)ssl;

  return 0;
}
```

1. clang++-18 -Wall -O2 -g wolfssl.cc

```
In file included from wolfssl.cc:2:
In file included from <...>/include/wolfssl/ssl.h:36:
In file included from <...>/include/wolfssl/wolfcrypt/asn_public.h:35:
In file included from <...>/include/wolfssl/wolfcrypt/dsa.h:33:
In file included from <...>/include/wolfssl/wolfcrypt/wolfmath.h:48:
In file included from <...>/include/wolfssl/wolfcrypt/sp_int.h:35:
<...>/include/wolfssl/wolfcrypt/hash.h:122:5: warning: struct has size 0 in C, size 1 in C++ [-Wextern-c-compat]
  122 |     WOLF_AGG_DUMMY_MEMBER;
      |     ^
<...>/include/wolfssl/wolfcrypt/types.h:177:13: note: expanded from macro 'WOLF_AGG_DUMMY_MEMBER'
  177 |             struct {                                                           \
      |             ^
1 warning generated.
```

### Relevant log output

```shell

```

embhorn, 2025-02-20:
@douzzer - do you test with Clang++ 18? What is your opinion here?

closed 2025-02-21 by JacobBarthelmeh

douzzer, 2025-02-21:
@tatsuhiro-t thanks for the report -- we've fixed the issue, and yes it is 100% safe to ignore the warning.

PR #8482 (2025-02-21): 20250220-misc-UnalignedWord64
`misc.c`: undo changes in 82b50f19c6 "when Intel x64 build, assume able to read/write unaligned" -- provokes sanitizer on amd64, and is not portable (e.g. different behavior on Intel vs AMD).  all performance-sensitive `word64` reads/writes should be on known-aligned data.

`wolfssl/wolfcrypt/types.h`: don't define `HAVE_EMPTY_AGGREGATES` when `defined(__cplusplus)` (fixes #8478).

`wolfcrypt/test/test.c`: fix return values in `camellia_test()` (also fixes some false positive `-Wreturn-stack-address`es from `clang++`).

tested with `wolfssl-multi-test.sh ... quant [... cut]
commit 8aa2799a (2025-02-21): wolfssl/wolfcrypt/types.h: don't define HAVE_EMPTY_AGGREGATES when defined(__cplusplus) (fixes #8478).