wolfSSL/wolfssl #8486: Unclear test status

ssooffiiaannee, 2025-02-21:
### Version

5.7.2-stable

### Description

After building the project, it is unclear if the tests are successful or not. And I am not sure if I am running the test correctly.

I build the project in debug mode (test building is enabled by default) :
> cmake -D WOLFSSL_DEBUG=y ..

RUn the unit tests
` ./tests/unit.test`

A unit test does not seem to succeed,  maybe an issue with a client reaching a host

> wolfSSL Entering SetSSL_CTX
> wolfSSL Entering wolfSSL_NewSession
> InitSSL done. return 0 (success)
> wolfSSL_new InitSSL success
> wolfSSL Leaving wolfSSL_new InitSSL =, return 0
> wolfSSL Entering wolfSSL_SetEnableDhKeyTest
> wolfSSL Leaving wolfSSL_SetEnableDhKeyTest, return 1
> wolfSSL error: tcp connect failed: No route to host

Any thoughts on what I am doing or went wrong ?

embhorn, 2025-02-21:
Hi @ssooffiiaannee 

The instructions for building with cmake are here:
https://github.com/wolfSSL/wolfssl/blob/8ae122584ca50216b231d9dc78ec4a90d5e464c7/INSTALL#L84-L139

From the root wolfSSL folder:
```
mkdir build
cd build
cmake .. -DCMAKE_BUILD_TYPE=Debug
cmake --build .
./tests/unit.test
```
At the end you should see:
```
 Success -- All results as expected.
 End Cipher Suite Tests

unit_test: Success for all configured tests.
```

Thanks,
@embhorn - wolfSSL Support

ssooffiiaannee, 2025-02-21:
Thank you @embhorn 

I still could not get a successful test, are you building commit tagged v5.7.2-stable ? 
gcc version : gcc (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0

> SSL version is TLSv1.2
> SSL cipher suite is TLS_ECDHE_ECDSA_WITH_CHACHA20_POLY1305_SHA256
> SSL curve name is SECP256R1
> SSL version is TLSv1.2
> SSL cipher suite is TLS_ECDHE_ECDSA_WITH_CHACHA20_POLY1305_SHA256
> SSL curve name is SECP256R1
> Client message: hello wolfssl!
> I hear you fa shizzle!
> trying server command line[703]: SuiteTest -v 3 -l ECDHE-RSA-AES128-GCM-SHA256 -k ./certs/server-key.pem -c ./certs/test/server-goodcn.pem -d -2 -p 0 
> trying client command line[704]: SuiteTest -v 3 -l ECDHE-RSA-AES128-GCM-SHA256 -h localhost -A ./certs/test/server-goodcn.pem -m -C -2 -p 34133 
> wolfSSL error: tcp connect failed: No route to host

embhorn, 2025-02-21:
I just retested with v5.7.2 successfully. 

The error log you shared is showing that the tcp connect is failing. Can you check that there are no other running instances of the test? Maybe from ctrl-C'ing the test previously?

ssooffiiaannee, 2025-02-22:
Ctrl-C did not work, I had to restart my OS to make it work, not sure what went wrong.

embhorn, 2025-02-24:
Great, I'll go ahead close out this issue. Feel free to reopen or create a new issue if there are questions.

closed 2025-02-24 by embhorn

ssooffiiaannee, 2025-02-27:
I managed to reproduce this issue, I found that I could not run the test because getHostByName() could not resolve localhost into the 127.0.0.1 ip. 
After switcing wifi off, I managed to make it run because it could resolve localhost.
it can also be tested with

>  nslookup localhost

Without wifi, localhost is resolved into some thing like : localhost.dhcp.xxx.xxx, 
But without wifi it is 127.0.0.1.