wolfSSL/wolfssl #8490: wolfssl binding for Ada/Spark is not in the index of Alire.

JulienGioux, 2025-02-22:
### Version

last

### Description

wolfssl binding for Ada/Spark is not in the index of Alire.
```
alr search wolfssl
No hits
```

embhorn, 2025-02-24:
Hi @JulienGioux 

Thanks for contacting wolfSSL Support. It looks as though this effort was started but never completed:
https://github.com/alire-project/alire-index/pull/1023

I can help you open a feature request if you'd like. Just email us at support@wolfssl.com to get started.

Kind regards,
@embhorn - wolfSSL Support

closed 2025-02-24 by embhorn

mgrojo, 2025-03-27:
I'd like to work on this, because I want to include my project [coap_spark](https://github.com/mgrojo/coap_spark) on the Alire index, and the first step should be having wolfssl also included.

I'd like also to add support of the WolfSSL build settings through [Alire crate configuration](https://alire.ada.dev/docs/#using-crate-configuration), so I can turn on "-DWOLFSSL_STATIC" in the configuration of my crate.

For that, I plan to include the Alire-generated `config/wolfssl_config.gpr` file in `wolfssl.gpr` as is usual practice in Alire crates. That could also be used for the automatic OS configuration.

@Irvise, I've seen you included this project file as part of 79235a469824e27c52f26b1a763a1fc2fac8bae2. Do you see any problem with this plan?

`default.gpr` is still present, and I suppose it could still be used by users of the library that aren't working with Alire. Other possibility is including the Alire-generated `config/wolfssl_config.gpr` in Git.

mgrojo, 2025-03-27:
It seems I have to make another change. I don't know why using the library through Alire introduces this change of behaviour, but the `pragma Restrictions (No_Secondary_Stack);` present in https://github.com/wolfSSL/wolfssl/blob/master/wrapper/Ada/gnat.adc was not applying to my project itself when I was extending the wolfssl.gpr project ([here](https://github.com/mgrojo/coap_spark/blob/main/wolfssl_coap.gpr)), but it is applying now when I use `wolfssl.gpr` through Alire. My project cannot work with that restriction, and I think the WolfSSL shouldn't force it on users, so I'm going to rename `gnat.adc` (otherwise is still used) and reference it only from `default.gpr`, not from the Alire project (`wolfssl.gpr`).

@joakim-strandberg `gnat.adc` was already in your original contribution, so I'd like also to know your opinion on this.

PR #8714 (2025-04-28): Ada: prepare inclusion of v5.8.0 in Alire index
# Description

This prepares the Alire manifest for publishing in the Alire index.

It can also be done with a fork, see draft PR: https://github.com/alire-project/alire-index/pull/1485, but I suppose it's better if the origin remits to the official repository. I will redo the Alire PR if the commit enters this repository.

Fixes #8490

# Testing

How did you test?

Automatic tests by `alr publish`

# Checklist

 - [ ] added tests
 - [ ] updated/added doxygen
 - [ ] updated appropriate READMEs
 - [ ] Updated manual and documentation