wolfSSL/wolfssl #8503: [Bug]: wolfssl verification error

dulanshuangqiao, 2025-02-25:
### Contact Details

bingxiaoganel@gmail.com

### Version

ubutun 5.6.4

### Description

 ./configure
  make
  sudo make install
  ./testsuite/testsuite.test
wolfSSL is configured and built by default

### Reproduction steps

Use this script for certificate verification
// gcc -g verify.c -o verify -lwolfssl
#include <stdlib.h>
#include <wolfssl/ssl.h>
#include <wolfssl/options.h>
#include <wolfssl/wolfcrypt/error-crypt.h>

int wolfSSL_Verify_PEM(char * cac, char * ec){
	int ret = 0;

	// to create a new wolfSSL cert manager
	WOLFSSL_CERT_MANAGER* cm;
	cm = wolfSSL_CertManagerNew();
	if (cm == NULL){
		printf("Creating a new wolfSSL_CertManager failed!\n");
		exit(1);
	}
	
	// to load cac to the created wolfSSL_CertManager
	ret = wolfSSL_CertManagerLoadCA(cm, cac, NULL);
	if (ret != SSL_SUCCESS){
		printf("Loading cac to the created wolfSSL_CertManager failed!\n");
		exit(2);
	}

	// to verify the ec in the created wolfSSL_CertManager
	ret = wolfSSL_CertManagerVerify(cm, ec, SSL_FILETYPE_PEM);
	if (ret != SSL_SUCCESS){
		printf("wolfSSL_CertManagerVerify filed and with return code %d and error message %s\n",
			 ret,
			 wolfSSL_ERR_reason_error_string(ret));
	}
	else{
		printf("The target cert has passed through verification.\n");
	}

	// to free cm
	wolfSSL_CertManagerFree(cm);

	return ret;
}

int main(int argc, char ** argv){
	char * cac = argv[1];
	char * ec = argv[2];
	wolfSSL_Verify_PEM(cac, ec);
	return 0;
}

Actual result:
./verify RootCA.pem Cert1732784624436A1.pem
The target cert has passed through verification.

Expected result:
Consistent verification result between wolfSSL and RFC 5280+other TLS implementations.

 RFC5280 mentions：A certificate MUST NOT include more than one instance of a particular extension.

### Relevant log output

```shell

```

dgarske, 2025-02-25:
Hi @dulanshuangqiao ,

Are you able to share the certificate or log output? I don't see either. If you'd like to send privately please email support at wolfssl dot com.

Thanks,
David Garske, wolfSSL

kareem-wolfssl, 2025-02-25:
Hi @dulanshuangqiao ,

To add to my colleague's response, please try retesting with our latest release 5.7.6 as 5.6.4 is outdated and we've over a year of bugfixes since then.

dulanshuangqiao, 2025-02-27:
> 你好[@dulanshuangqiao](https://github.com/dulanshuangqiao)，
> 
> 为了补充我同事的回复，请尝试使用我们最新版本的 5.7.6 重新测试，因为 5.6.4 已经过时，而且从那时起我们已经修复了一年多的错误。 还请重新排序您的 wolfSSL 包含，options.h 必须位于所有其他 wolfSSL 标头之前：
> 
> ```
> #include <wolfssl/options.h>
> #include <wolfssl/ssl.h>
> #include <wolfssl/wolfcrypt/error-crypt.h>
> ```
I upgraded WolfSSL and retested it with version 5.7.6, but the validation was still successful (RFC5280 specifies that certificates cannot contain multiple instances of specific extensions, and Wolfss should not pass validation).
The attachment is the test case I used: a leaf certificate with two AIA extensions and its CA certificate.

[case.zip](https://github.com/user-attachments/files/19005427/case.zip)

ColtonWilley, 2025-02-28:
@dulanshuangqiao 

So you are correct that we are verifying the certificate even though it is not in compliance with RFC 5280 section 4.2. At this time OpenSSL also handles duplicate extensions the same way (ie still verifies). We will be taking a look at enforcing this restriction, I will let you know when I have a PR available to test. Please let me know if you need anything else in the meantime.

Thanks,
Colton Willey, wolfSSL.

dulanshuangqiao, 2025-03-01:
> [@dulanshuangqiao](https://github.com/dulanshuangqiao)
>
> So you are correct that we are verifying the certificate even though it is not in compliance with RFC 5280 section 4.2. At this time OpenSSL [...]

Thanks for your reply. As you mentioned, I also reported the issue to OpenSSL and they accepted my report and classified it as a bug.