wolfSSL/wolfssl #8525: [Bug]: WolfSSL verification passed Serial Number with incorrect length

dulanshuangqiao, 2025-03-03:
### Contact Details

bingxiaoganel@gmail.com

### Version

ubutun 5.6.4

### Description

./configure
make
sudo make install
./testsuite/testsuite.test
wolfSSL is configured and built by default

### Reproduction steps

Use this script for certificate verification
// gcc -g verify.c -o verify -lwolfssl

```
#include <wolfssl/options.h>
#include <stdlib.h>
#include <wolfssl/wolfcrypt/error-crypt.h>
#include <wolfssl/ssl.h>

int wolfSSL_Verify_PEM(char * cac, char * ec){
int ret = 0;
// to create a new wolfSSL cert manager
WOLFSSL_CERT_MANAGER* cm;
cm = wolfSSL_CertManagerNew();
if (cm == NULL){
	printf("Creating a new wolfSSL_CertManager failed!\n");
	exit(1);
}

// to load cac to the created wolfSSL_CertManager
ret = wolfSSL_CertManagerLoadCA(cm, cac, NULL);
if (ret != SSL_SUCCESS){
	printf("Loading cac to the created wolfSSL_CertManager failed!\n");
	exit(2);
}

// to verify the ec in the created wolfSSL_CertManager
ret = wolfSSL_CertManagerVerify(cm, ec, SSL_FILETYPE_PEM);
if (ret != SSL_SUCCESS){
	printf("wolfSSL_CertManagerVerify filed and with return code %d and error message %s\n",
		 ret,
		 wolfSSL_ERR_reason_error_string(ret));
}
else{
	printf("The target cert has passed through verification.\n");
}

// to free cm
wolfSSL_CertManagerFree(cm);

return ret;
}

int main(int argc, char ** argv){
char * cac = argv[1];
char * ec = argv[2];
wolfSSL_Verify_PEM(cac, ec);
return 0;
```

RFC5280 states: Conforming CAs MUST NOT use serialNumber values longer than 20 octets.
GnuTLS performs this check during validation and refuses to accept such certificates.

```
certtool --verify --load-ca-certificate RootCA.pem --infile Cert174065870010.pem：
Loaded CAs (1 available)
|<2>| error: serial number value is longer than 20 octets
error parsing CRTs: Error in the certificate.
Setting log level to 10
```

Actual result:
Verifying certificate: Cert174065870010.pem
The target cert Cert174065870010.pem has passed verification.
Certificate Cert174065870010.pem passed verification.
Expected result:
Certificate Cert174065870010.pem failed verification.

### Relevant log output

```shell

```

ColtonWilley, 2025-03-03:
@dulanshuangqiao 

Again you are correct that we are verifying the certificate despite RFC5280 restrictions on serial number length. Along the same lines, openssl also verifies the certificate incorrectly. I will be discussing this issue with my colleagues, I will let you know how we will handle this once a decision has been made. We appreciate the time you have spent submitting these issues with well documented steps on on how to reproduce. Please continue to file these issues as you find them, and I will discuss the next best steps with the wolfSSL team.

Thanks,
Colton Willey, wolfSSL.

dulanshuangqiao, 2025-03-07:
> [@dulanshuangqiao](https://github.com/dulanshuangqiao)
>
> Again you are correct that we are verifying the certificate despite RFC5280 restrictions on serial number length. Along the same lines, [...]

OK, thank you very much for your reply, I will wait for your results.

dulanshuangqiao, 2025-04-01:
> [@dulanshuangqiao](https://github.com/dulanshuangqiao)
>
> Again you are correct that we are verifying the certificate despite RFC5280 restrictions on serial number length. Along the same lines, [...]

There is no staff to handle this report for a long time.I hope a developer can review my report，This is very important for my work.