wolfSSL/wolfssl #8529: [Bug]: CRL Issuer field with an empty DN

onepeople158, 2025-03-04:
### Contact Details

1784244538@qq.com

### Version

You are using version 0.1.7 of the wolfssl Command Line Utility. Linked to wolfSSL version 5.7.6

### Description

Main content：     
                I have a CRL file where the DN value in the Issuer field is empty, but RFC 5280 specifies that the size of the DN should be (1..MAX). I was able to successfully print the Issuer field of this CRL file using Bouncy Castle 1.79 without any errors. For comparison, when using GnuTLS to print this CRL file, the following error was displayed: error: get_issuer_dn: ASN1 parser: Value is not valid. 

Therefore, I believe this might be a bug.

Computer system：
Ubuntu

Test case:
[crl_zero_DN.zip](https://github.com/user-attachments/files/19066674/crl_zero_DN.zip)

### Reproduction steps

wolfssl crl -inform der -in crl_zero_DN.der -text

### Relevant log output

```shell

```

anhu, 2025-03-04:
HI @onepeople158 , 

I've reproduced what you are seeing by building wolfssl with the following  configuration `./configure --enable-wolfclu --enable-crl` and building wolfCLU with the following configuration: `./configure` and the following command: `./wolfssl crl -inform der -in crl_zero_DN.der -text` 

I'll note its interesting that OpenSSL does not show an error either: `openssl crl  -inform der -in crl_zero_DN.der -text`

I'll need to confer with my colleagues about this.  Please stay tuned. 

In the meantime, can you let me know a bit about yourself and your project?

- what are your goals?
- where you are located?
- is this out of personal, academic or professional interest?

Warm regards, Anthony

anhu, 2025-03-04:
Hi @onepeople158 , 

Please have a look at https://github.com/wolfSSL/wolfCLU/pull/172 . We believe this is the fix for what you are observing.  Let me know if this helps. 

Warm regards, Anthony

onepeople158, 2025-03-05:
> HI [@onepeople158](https://github.com/onepeople158) ,
> 
> I've reproduced what you are seeing by building wolfssl with the following configuration and building wolfCLU with the following configuration: and the following command: `./configure --enable-wolfclu --enable-crl``./configure``./wolfssl crl -inform der -in crl_zero_DN.der -text`
> 
> I'll note its interesting that OpenSSL does not show an error either: `openssl crl -inform der -in crl_zero_DN.der -text`
> 
> I'll need to confer with my colleagues about this. Please stay tuned.
> 
> In the meantime, can you let me know a bit about yourself and your project?
> 
> * what are your goals?
> * where you are located?
> * is this out of personal, academic or professional interest?
> 
> Warm regards, Anthony
Hello, it is for academic purposes.

onepeople158, 2025-03-05:
> Hi [@onepeople158](https://github.com/onepeople158) ,
> 
> Please have a look at [wolfSSL/wolfCLU#172](https://github.com/wolfSSL/wolfCLU/pull/172) . We believe this is the fix for what you are observing. Let me know if this helps.
> 
> Warm regards, Anthony

Hello, may I ask if the issue I raised will be considered a new bug?

anhu, 2025-03-05:
@onepeople158 , 
I'm not sure what you mean.  If you are looking for some sort of bug bounty we do not offer rewards for bug reports.  May I ask what you were hoping would happen? 
Warm regards, Anthony

onepeople158, 2025-03-05:
> [@onepeople158](https://github.com/onepeople158) , I'm not sure what you mean. If you are looking for some sort of bug bounty we do not offer rewards for bug reports. May I ask what you were hoping would happen? Warm regards, Anthony
Hello developer, I don't need rewards, I just want to confirm something.

anhu, 2025-03-05:
Hi @onepeople158 , 

 In the case of wolfCLU we see it as a tool with the same functionality as the `openssl` command line and so we want it to follow the same behaviour.   The job of the tool is to display CRLs and so we give it a bit of leeway.  When you build wolfSSL and use --enable-wolfclu , we define the macro `WOLFSSL_NO_ASN_STRICT`  

If you apply the following patch: 

```
--- a/configure.ac
+++ b/configure.ac
@@ -9382,7 +9382,7 @@ then
     # Uses alt name
     ENABLED_ALTNAMES="yes"
 
-    AM_CFLAGS="$AM_CFLAGS -DHAVE_OID_ENCODING -DWOLFSSL_NO_ASN_STRICT"
+    AM_CFLAGS="$AM_CFLAGS -DHAVE_OID_ENCODING"
 
 fi
 
```

and rebuild wolfSSL and wolfCLU you will find that it does return an error. 

The relevant code snippet is in `wolfssl/wolfcrypt/src/asn.c` in the function `GetCertName()` : 
```
            #ifndef WOLFSSL_NO_ASN_STRICT
                /* RFC 5280 section 4.1.2.4 lists a DirectoryString as being
                 * 1..MAX in length */
                if (ret == 0 && strLen < 1) {
                    WOLFSSL_MSG("Non conforming DirectoryString of length 0 was"
                                " found");
                    WOLFSSL_MSG("Use WOLFSSL_NO_ASN_STRICT if wanting to allow"
                                " empty DirectoryString's");
                    ret = ASN_PARSE_E;
                }
            #endif
```

So, to answer your question, no, we do not consider this a bug.   Can you please let us know about yourself and your project? 

Warm regards, Anthony

onepeople158, 2025-03-06:
> Hi [@onepeople158](https://github.com/onepeople158) ,
> 
> In the case of wolfCLU we see it as a tool with the same functionality as the command line and so we want it to follow the same behaviour. The job of the tool is to display CRLs and so we give it a bit of leeway. When you build wolfSSL and use --enable-wolfclu , we define the macro `openssl``WOLFSSL_NO_ASN_STRICT`
> 
> If you apply the following patch:
> 
> ```
> --- a/configure.ac
> +++ b/configure.ac
> @@ -9382,7 +9382,7 @@ then
>      # Uses alt name
>      ENABLED_ALTNAMES="yes"
>  
> -    AM_CFLAGS="$AM_CFLAGS -DHAVE_OID_ENCODING -DWOLFSSL_NO_ASN_STRICT"
> +    AM_CFLAGS="$AM_CFLAGS -DHAVE_OID_ENCODING"
>  
>  fi
>  
> ```
> 
> and rebuild wolfSSL and wolfCLU you will find that it does return an error.
> 
> The relevant code snippet is in in the function :`wolfssl/wolfcrypt/src/asn.c``GetCertName()`
> 
> ```
>             #ifndef WOLFSSL_NO_ASN_STRICT
>                 /* RFC 5280 section 4.1.2.4 lists a DirectoryString as being
>                  * 1..MAX in length */
>                 if (ret == 0 && strLen < 1) {
>                     WOLFSSL_MSG("Non conforming DirectoryString of length 0 was"
>                                 " found");
>                     WOLFSSL_MSG("Use WOLFSSL_NO_ASN_STRICT if wanting to allow"
>                                 " empty DirectoryString's");
>                     ret = ASN_PARSE_E;
>                 }
>             #endif
> ```
> 
> So, to answer your question, no, we do not consider this a bug. Can you please let us know about yourself and your project?
> 
> Warm regards, Anthony

Thank you for your response. My project is about certificate revocation checking.

anhu, 2025-03-06:
Is this an academic project? Will there be a paper published about your work?

onepeople158, 2025-03-06:
> Is this an academic project? Will there be a paper published about your work?

Currently, there are no papers published.

anhu, 2025-03-06:
Thanks for letting us know.  This comment will now close this ticket.

closed 2025-03-06 by anhu