wolfSSL/wolfssl #8545: [Bug]: OpenSSL compatibility SSL_get_cipher is incorrectly mapped

vuvova, 2025-03-10:
### Version

5.7.6

### Description

WolfSSL has its own cipher names, like "TLS13-AES128-GCM-SHA256" and IANA cipher names like "TLS_AES_128_GCM_SHA256", the latter generally matches what OpenSSL uses. There are function to retrieve both, `wolfSSL_get_cipher_name()` and `wolfSSL_get_cipher_name_iana()`. There's an OpenSSL compatibility header `openssl/ssl.h` which defines OpenSSL compatibility names, like
```c
#define SSL_get_cipher                  wolfSSL_get_cipher_name
```
This last definition is incorrect, OpenSSL compatible function name should return OpenSSL compatible cipher names, that is `SSL_get_cipher` should be mapped to `wolfSSL_get_cipher_name_iana`

cf. https://jira.mariadb.org/browse/MDEV-35580

### Relevant log output

```shell

```

anhu, 2025-03-11:
Hi, I've put up a proposed PR.  I'll need to discuss this internally with my engineering team.  Please stay tuned.

anhu, 2025-03-18:
Hi, I was wondering if you tried defining `WOLFSSL_CIPHER_INTERNALNAME` ?  Does that give you the names you're looking for?

vuvova, 2025-04-16:
meanwhile we've fixed it MariaDB as
```diff
+#undef SSL_get_cipher
+#define SSL_get_cipher                  wolfSSL_get_cipher
```

vuvova, 2025-04-17:
It seems that OpenSSL names aren't IANA names. Simply **some** WolfSSL cipher names don't match OpenSSL, while others do.
Anyway this original report is invalid, sorry for this.

closed 2025-04-17 by vuvova

vuvova, 2025-07-02:
FYI, if you still want a proper OpenSSL emulation of `SSL_get_cipher`, we do it like
```c
#define SSL_get_cipher(ssl) (SSL_version(ssl) == TLS1_3_VERSION ? wolfSSL_get_cipher(ssl) : wolfSSL_get_cipher_name(ssl))
```
I don't like it either, but that returns the same result as OpenSSL.

vuvova, 2025-07-02:
I still think it's incorrect. It's not some WolfSSL functionality where you can freely choose what to do. It's OpenSSL compatibility layer. It doesn't make sense to have it, unless it mimics OpenSSL behavior.