wolfSSL/wolfssl #8554: Build fails when HAVE_INTEL_QA enabled on missing file

mcollard0, 2025-03-12:
### Version

5.7.6+ (git pull current)

### Description

When I attempt to build wolfssl with HAVE_INTEL_QA enabled it fails stating a file is missing. I did search this file in the git codebase (on github.com/wolfssl and not my local copy) and did not find it. 

C:\repos\untitledMigrationService\packages\wolfssl\wolfssl\wolfcrypt\types.h(468,22): error C1083: Cannot open include file: 'wolfssl/wolfcrypt/port/intel/quickassist_mem.h': No such file or directory [...\packages\wolfssl\wolfssl-VS2022.vcxproj]
         (compiling source file 'src/crl.c')

Can include full output as desired. A list of the configuration Preprocessor Definitions:

NO_MD5;WOLFSSL_TLS13;NO_OLD_TLS;USE_FAST_MATH;WOLFSSL_SHA1;HAVE_SHA256;HAVE_SHA384;HAVE_SHA512;WOLFSSL_SHA3;WOLFSSL_AESNI;HAVE_ECC;HAVE_RSA;HAVE_AESGCM;OPENSSL_EXTRA;OPENSSL_ALL;WOLFSSL_ASYNC_CRYPT;WOLFSSL_ASYNC_CRYPT_SW;HAVE_CRL;HAVE_OCSP;HAVE_EX_DATA;KEEP_PEER_CERT;WOLFSSL_ASN_TEMPLATE;WOLFSSL_X509_STORE;HAVE_EXT_CACHE;WOLFSSL_CERT_GEN;WOLFSSL_CERT_REQ;HAVE_PKCS7;HAVE_X963_KDF;WOLFSSL_CUSTOM_OID;WOLFSSL_CERT_EXT;WOLFSSL_BIO_INCLUDED;HAVE_SUPPORTED_CURVES;HAVE_SNI;HAVE_SESSION_TICKET;HAVE_ALPN;HAVE_EXTENDED_MASTER;HAVE_SECURE_RENEGOTIATION;HAVE_THREAD_LS;WOLFSSL_DTLS_ALLOW_FUTURE;WOLFSSL_USER_IO;HAVE_CERTIFICATE_STATUS_REQUEST;HAVE_CERTIFICATE_STATUS_REQUEST_V2;HAVE_TRUSTED_CA;WOLFSSL_BASE64_ENCODE;BUILDING_WOLF3D;BUILDING_WOFSSL;WOLFSSL_CUSTOM_CONFIG;WOLFSSL_AES_DIRECT;HAVE_AES_KEYWRAP;

dgarske, 2025-03-13:
Hi @mcollard0 ,

Those files are in a separate public repository located here: https://github.com/wolfssl/wolfasynccrypt
https://github.com/wolfSSL/wolfAsyncCrypt/tree/master/wolfcrypt/src/port/intel

Please let me know if you have any issues. 

Thanks,
David Garske, wolfSSL

dgarske, 2025-03-26:
Hi @mcollard0 ,

Were you able to get the Intel QuickAssist code building and working? Can you tell us more about your project?

Thanks,
David Garske, wolfSSL