wolfSSL/wolfssl #8573: [Bug]: Coding validity judgment

dulanshuangqiao, 2025-03-20:
### Contact Details

bingxiaoganel@gmail.com

### Version

ubutun 5.6.4

### Description

./configure
make
sudo make install
./testsuite/testsuite.test
wolfSSL is configured and built by default

### Reproduction steps

Use this script for certificate verification

```
// gcc -g verify.c -o verify -lwolfssl
#include <stdlib.h>
#include <wolfssl/ssl.h>
#include <wolfssl/options.h>
#include <wolfssl/wolfcrypt/error-crypt.h>

int wolfSSL_Verify_PEM(char * cac, char * ec){
int ret = 0;

// to create a new wolfSSL cert manager
WOLFSSL_CERT_MANAGER* cm;
cm = wolfSSL_CertManagerNew();
if (cm == NULL){
	printf("Creating a new wolfSSL_CertManager failed!\n");
	exit(1);
}

// to load cac to the created wolfSSL_CertManager
ret = wolfSSL_CertManagerLoadCA(cm, cac, NULL);
if (ret != SSL_SUCCESS){
	printf("Loading cac to the created wolfSSL_CertManager failed!\n");
	exit(2);
}

// to verify the ec in the created wolfSSL_CertManager
ret = wolfSSL_CertManagerVerify(cm, ec, SSL_FILETYPE_PEM);
if (ret != SSL_SUCCESS){
	printf("wolfSSL_CertManagerVerify filed and with return code %d and error message %s\n",
		 ret,
		 wolfSSL_ERR_reason_error_string(ret));
}
else{
	printf("The target cert has passed through verification.\n");
}

// to free cm
wolfSSL_CertManagerFree(cm);

return ret;
}

int main(int argc, char ** argv){
char * cac = argv[1];
char * ec = argv[2];
wolfSSL_Verify_PEM(cac, ec);
return 0;
}
```
I use this program to verify certificates. For PolicyMappings extension with empty value, wolfSSL should throw an error, but here WolfSSL incorrectly verifies it. Because RFC5280 stipulates that PolicyMappings extension is a sequence of (1...MAX), so the extension value of zero byte is not a valid encoding.

./verify RootCA.pem PM.pem
The target cert has passed through verification.
./verify RootCA.pem PM1.pem
The target cert has passed through verification.

Expected result:
wolfSSL_CertManagerVerify failed with return code -140 and error message ASN parsing error, invalid input.

[PM.zip](https://github.com/user-attachments/files/19367221/PM.zip)

### Relevant log output

```shell

```

embhorn, 2025-03-20:
Hi @dulanshuangqiao 

Thanks for this report. Could you please test with the latest code in the repository? Let me know if the issue persists. Could you tell us about your project using wolfSSL?

Thanks,
@embhorn

dulanshuangqiao, 2025-03-21:
> Hi [@dulanshuangqiao](https://github.com/dulanshuangqiao)
> 
> Thanks for this report. Could you please test with the latest code in the repository? Let me know if the issue persists. Could you tell us about your project using wolfSSL?
> 
> Thanks, [@embhorn](https://github.com/embhorn)

Thank you for your reply. The WolfSSL version I tested is the latest 5.7.6.
I use the certificate verification script presented in this report for certificate verification.
Because I received feedback regarding the validity of the encoded values in question # [8564,](https://github.com/wolfSSL/wolfssl/issues/8564) I conducted relevant tests based on the feedback and discovered the issue reported here.

dulanshuangqiao, 2025-04-01:
> Hi [@dulanshuangqiao](https://github.com/dulanshuangqiao)
> 
> Thanks for this report. Could you please test with the latest code in the repository? Let me know if the issue persists. Could you tell us about your project using wolfSSL?
> 
> Thanks, [@embhorn](https://github.com/embhorn)

There is no staff to handle this report for a long time.I hope a developer can review my report，This is very important for my work.

embhorn, 2025-04-01:
Hi @dulanshuangqiao 

> This is very important for my work.

Could you tell us about your project? Feel free to use our support email if you'd prefer a more private exchange:
support@wolfssl.com

Thanks,
@embhorn

dulanshuangqiao, 2025-04-01:
> Hi [@dulanshuangqiao](https://github.com/dulanshuangqiao)
> 
> > This is very important for my work.
> 
> Could you tell us about your project? Feel free to use our support email if you'd prefer a more private exchange: [support@wolfssl.com](mailto:support@wolfssl.com)
> 
> Thanks, [@embhorn](https://github.com/embhorn)

I am conducting academic research aimed at enhancing TLS security. I apologize that I cannot share details about my project at this time, as it involves certain future review-related issues.

embhorn, 2025-04-01:
@dulanshuangqiao 

We need to better understand your goals for this project and your interest in opening so many PR's and issues in the wolfSSL project. Please email support@wolfssl.com. This will open a support ticket where we can discuss this in more detail.

Kind regards,
Eric Blankenhorn - wolfSSL Support

dulanshuangqiao, 2025-04-02:
> [@dulanshuangqiao](https://github.com/dulanshuangqiao)
> 
> We need to better understand your goals for this project and your interest in opening so many PR's and issues in the wolfSSL project. Please email [support@wolfssl.com](mailto:support@wolfssl.com). This will open a support ticket where we can discuss this in more detail.
> 
> Kind regards, Eric Blankenhorn - wolfSSL Support

OK, thank you for your reply. I will communicate with WolfSSL as soon as possible. 
Please deal with the problem I reported as soon as possible and give me the result. Thank you, this is very important to me.

dulanshuangqiao, 2025-04-16:
> [@dulanshuangqiao](https://github.com/dulanshuangqiao)
>
> We need to better understand your goals for this project and your interest in opening so many PR's and issues in the wolfSSL project. Please [...]

There is no staff to handle this report for a long time.I hope a developer can review my report，This is very important for my work. @embhorn

embhorn, 2025-04-16:
@dulanshuangqiao 

As previously requested, please email support@wolfssl.com to further discuss this.