wolfSSL/wolfssl #8585: [Bug]: DTLS 1.3 Server responds with empty Acknowledgement to Application data before handshake completion

c-southwest, 2025-03-22:
### Contact Details

GitHub should be good

### Version

wolfssl-5.7.6

### Description

## What happens
Before Client send Finished, I intentionally send Application, and server responds with empty Acknowledgement message.
[image]

## My expected behavior
1. Server should not respond with Acknowledgement to Application data, since Application data is not Handshake message type. 

2. I think the early Application data should be considered as unexpected message, and server should send Alert (unexpected_message)

## Reference
DTLS 1.3 RFC [rfc9147 7. ACK Message](https://www.rfc-editor.org/rfc/rfc9147.html#ack-msg)
> The ACK message is used by an endpoint to indicate which handshake records it has received and processed from the other side.

DTLS 1.3 RFC [rfc9147 7.1. Sending ACKs](https://www.rfc-editor.org/rfc/rfc9147.html#name-sending-acks)
> ACKs MUST NOT be sent for records of any content type other than handshake or for records which cannot be deprotected.

TLS 1.3 RFC [rfc8446 6.2.  Error Alerts](https://www.rfc-editor.org/rfc/rfc8446.html#section-6.2)
> unexpected_message:  An inappropriate message (e.g., the wrong
      handshake message, premature Application Data, etc.) was received.
      This alert should never be observed in communication between
      proper implementations.

## Other Information
### Environment
OS: Ubuntu 24.04

### How I build wolfSSL:
```
bash autogen.sh
AM_CFLAGS='-DHAVE_AES_CBC -DWOLFSSL_AES_128 -DWOLFSSL_DEBUG_TLS' ./configure --enable-dtls --enable-dtls13 --enable-keylog-export --enable-psk --enable-rsa --enable-sha --enable-debug C_EXTRA_FLAGS=-DWOLFSSL_STATIC_PSK
```

### How I launch wolfSSL server:
```
/dtls-fuzzer/suts/wolfssl-5.7.6/examples/server/server  -i -x -p 17830 -s -u -v 4 -l TLS13-AES128-GCM-SHA256
```

### Reproduction steps

If you don't want to do reproduction, then you can download **reproduction_ack_app.zip**, I collected network traffic captures, sslkeylog files, wolfssl server debug logs, and Docker files inside it.
[reproduction_ack_app.zip](https://github.com/user-attachments/files/19399734/reproduction_ack_app.zip)

---
In order to make the reproduction easy to run, I created [docker images](https://hub.docker.com/r/west2077/wolfssl-bug-reproduction/tags), so you can generate results with only one command.
If you are using Linux, you need to install [Docker](https://docs.docker.com/engine/install/ubuntu/) first.
If you are using Windows, you need to install [Docker Desktop](https://www.docker.com/products/docker-desktop/) first.

After you run the command, you can see a new folder `myTmpFolder`, inside the folder there will be three files:
* dtls_capture: network traffic captured by tcpdump, and you can check it with Wireshark
* running_output: Java program logs and wolfssl server debug logs
* sslkeylog.log: generated by wolfssl, you can use it to decrypt network traffic within Wireshark

## Command to run reproduction
Linux:
`docker run --rm -v $(pwd)/myTmpFolder:/output west2077/wolfssl-bug-reproduction:acknowledge-early-application-psk`
Windows with powershell:
`docker run --rm -v ${PWD}/myTmpFolder:/output west2077/wolfssl-bug-reproduction:acknowledge-early-application-psk`

## Dockerfile and How I build the docker image
I choose Docker for easy to run, portability, and stable reproduction, and you can also build the docker image from scratch.
The only file you need is Dockerfile [Dockerfile_acknowledge_early_application_psk](https://github.com/c-southwest/dtls-fuzzer/blob/add-DTLS-13-wolfssl-bug-reproduction/Dockerfile_acknowledge_early_application_psk)
After you download the Dockerfile, you can build the image with command:
`docker build -t wolfssl-bug-reproduction:acknowledge-early-application-psk -f Dockerfile_acknowledge_early_application_psk .`

### Relevant log output

```shell
wolfSSL Leaving SendTls13Finished, return 0
accept state ACCEPT_FINISHED_SENT
accept state  TICKET_SENT
wolfSSL Entering RetrySendAlert
growing input buffer
wolfSSL Leaving wolfSSL_dtls_get_current_timeout, return 320
wolfSSL Entering EmbedReceiveFrom
wolfSSL Entering wolfSSL_dtls_get_using_nonblock
Data received
	2e 8c 5b 00 15 8e 6c b7 1d 37 07 9f 9b f7 1c 83 |..[...l..7......
	32 d4 2b be 6f 11 3a 83 9f e4                   |2.+.o.:...
reconstructed epoch number: 2
wolfSSL Entering Dtls13SetEpochKeys
Provisioning DECRYPT key
	ca 68 f9 c1 fa 54 67 8d 65 5f e6 8f c9 cc 45 d8 |.h...Tg.e_....E.
Provisioning AES Record Number dec key:
	4f 64 1e af 8d 66 a3 63 78 ea 8c 9b fa c3 fc e3 |Od...f.cx.......
reconstructed seq number: 0
wolfSSL Entering DecryptTls13
Data to decrypt
	8e 6c b7 1d 37                                  |.l..7
Additional Authentication Data
	2e 00 00 00 15                                  |.....
Authentication tag
	07 9f 9b f7 1c 83 32 d4 2b be 6f 11 3a 83 9f e4 |......2.+.o.:...
Dtls13GetSeq(): using seq: 0
TLS 1.3 Write Sequence 0 0
Nonce
	a2 cc 77 a1 3b ac 91 7a f6 57 26 a2             |..w.;..z.W&.
Decrypted data
	54 65 73 74 17                                  |Test.
wolfSSL Entering Dtls13CheckWindow
wolfSSL Entering Dtls13UpdateWindow
received record layer msg
got app DATA
Dropping app data received before handshake complete
Shrinking input buffer
wolfSSL Entering Dtls13DoScheduledWork
wolfSSL Entering SendDtls13Ack
wolfSSL Entering Dtls13SetEpochKeys
Provisioning ENCRYPT key
	0c 0f 2b 1a 30 7c b8 b2 30 a0 ea 59 7f 31 6b 34 |..+.0|..0..Y.1k4
Provisioning AES Record Number enc key:
	30 85 2c 06 eb f0 17 dd 79 31 b4 cf 89 c1 e0 dd |0.,.....y1......
growing output buffer
write ack records
wolfSSL Entering BuildTls13Message
wolfSSL Entering EncryptTls13
Data to encrypt
	00 00 1a                                        |...
Additional Authentication Data
	2e 00 02 00 13                                  |.....
Dtls13GetSeq(): using seq: 2
TLS 1.3 Write Sequence 0 2
Nonce
	24 68 62 fe e5 fd 54 84 dd 34 60 91             |$hb...T..4`.
Encrypted data
	9b 33 a2                                        |.3.
Authentication Tag
	d8 65 34 fa 71 10 9b e4 99 46 25 56 ae 93 3e d2 |.e4.q....F%V..>.
wolfSSL Leaving BuildTls13Message, return 0
wolfSSL Entering Dtls13RtxFlushAcks
Data to send
	2e ee 99 00 13 9b 33 a2 d8 65 34 fa 71 10 9b e4 |......3..e4.q...
	99 46 25 56 ae 93 3e d2                         |.F%V..>.
wolfSSL Entering EmbedSendTo
Shrinking output buffer
wolfSSL Entering RetrySendAlert
growing input buffer
wolfSSL Leaving wolfSSL_dtls_get_current_timeout, return 320
wolfSSL Entering EmbedReceiveFrom
wolfSSL Entering wolfSSL_dtls_get_using_nonblock
Data received
```

embhorn, 2025-03-24:
Hi @c-southwest 

Thanks for sharing this report. Could you tell us a bit about your project using wolfSSL?

c-southwest, 2025-03-24:
Hi @embhorn 
We are testing the wolfSSL DTLS 1.3 server implemention using [DTLS-Fuzzer](https://github.com/assist-project/dtls-fuzzer), a testing tool for DTLS implmentations. 
We are in the process of extending DTLS-Fuzzer to support DTLS 1.3.

rizlik, 2025-04-01:
Hi @c-southwest,

Thank you for the detailed report and reproduction steps.

In this scenario, wolfSSL behaves correctly:

Sending the ACK is intentional because receiving AppData before the Finished messages can result from network issues (e.g., lost or re-ordered UDP frames). wolfSSL detects this disruption when it sees AppData before the Finished message. The empty ACK informs the sender of the disruption, prompting a quick retransmission of the Finished message.

Sending an alert would be inappropriate since this situation can occur under normal network conditions with a well-behaving peer due to UDP's unreliability.

Does this make sense to you? Am I missing something?

c-southwest, 2025-04-01:
@rizlik Yes, your explaination really makes sense, thank you for your detailed description. And I will close this issue.

closed 2025-04-01 by c-southwest