wolfSSL/wolfssl #8591: [Bug]: Two AKI extensions

onepeople158, 2025-03-26:
### Version

I am using version 0.1.7 of the wolfssl Command Line Utility. Linked to wolfSSL version 5.7.6

### Description

WolfSSL successfully parsed the AKI extension information from a CRL file that contains two AKI extensions,although only one of the AKI extension information was parsed.However, this test case does not comply with the RFC 5280 specification, which states that a specific extension can only have one instance.

### Reproduction steps

wolfssl crl -inform der -in crl_two_aki.der -text
[crl_two_aki.zip](https://github.com/user-attachments/files/19462556/crl_two_aki.zip)

### Relevant log output

```shell
Certificate Revocation List (CRL):
        Version: 2 (0x1)
        Signature Algorithm: sha256WithRSAEncryption
        Issuer:  C=US, ST=California, L=San Francisco, O=My Company, CN=My Root CA, OU=My Root CA
        Last Update: Sep  1 00:00:00 2024 GMT
        Next Update: Dec  1 00:00:00 2025 GMT
        CRL extensions:
            X509v3 Authority Key Identifier:
                keyid:EF:69:E0:F7:D5:1D:E6:99:EC:DC:6D:D0:F7:E2:B9:5C:64:71:83:35
No Revoked Certificates.
    Signature Algorithm: sha256WithRSAEncryption
         81:55:c3:88:eb:4b:8b:5a:49:74:7f:f6:03:f3:a8:6a:ea:15:
         e1:eb:21:70:a7:16:d1:c9:4c:bc:6a:6b:cb:d4:38:0a:92:c2:
         ac:ac:bc:da:5e:a5:52:a2:a0:c5:68:1a:b2:e3:14:c8:c0:36:
         35:90:a9:31:22:31:13:80:9c:b7:c9:ff:e2:fc:14:76:64:57:
         f9:99:c0:e2:80:5c:c8:48:0b:d3:2b:27:38:9f:c4:12:02:c2:
         6b:c3:d3:6f:2a:9f:04:26:d3:d1:0f:f3:3e:08:09:9a:6d:0f:
         ac:0f:75:14:c3:cd:db:9c:28:1b:80:5f:ec:e7:f2:93:6b:26:
         0c:ab:85:e4:4a:a4:f8:27:a3:d2:ac:6c:2c:ca:e0:3f:d1:22:
         dd:6c:4c:ad:8c:de:f5:4e:b5:8f:af:a6:c3:7b:d1:06:14:5e:
         f4:38:af:64:59:69:cd:e8:9a:54:f8:df:1e:65:75:d3:f1:22:
         57:28:b9:0a:70:59:c5:6f:af:f2:be:55:ae:fb:d0:c4:6b:8e:
         7f:3c:3c:1a:c9:9a:fb:04:c3:f4:18:54:15:e9:1b:1b:f6:6f:
         bf:07:14:36:9c:6d:af:34:b0:c2:26:f2:5b:51:76:a5:02:6b:
         9a:88:73:a3:80:4e:cc:fb:8b:30:90:5c:46:e4:4c:47:84:4a:
         b7:0a:d4:c0
-----BEGIN X509 CRL-----
MIICCDCB8QIBATANBgkqhkiG9w0BAQsFADB5MQswCQYDVQQGEwJVUzETMBEGA1UE
CAwKQ2FsaWZvcm5pYTEWMBQGA1UEBwwNU2FuIEZyYW5jaXNjbzETMBEGA1UECgwK
TXkgQ29tcGFueTETMBEGA1UEAwwKTXkgUm9vdCBDQTETMBEGA1UECwwKTXkgUm9v
dCBDQRcNMjQwOTAxMDAwMDAwWhcNMjUxMjAxMDAwMDAwWqBEMEIwHwYDVR0jBBgw
FoAU72ng99Ud5pns3G3Q9+K5XGRxgzUwHwYDVR0jBBgwFoAU72ng99Ud5pns3G3Q
9+K5XGRxgzUwDQYJKoZIhvcNAQELBQADggEBAIFVw4jrS4taSXR/9gPzqGrqFeHr
IXCnFtHJTLxqa8vUOAqSwqysvNpepVKioMVoGrLjFMjANjWQqTEiMROAnLfJ/+L8
FHZkV/mZwOKAXMhIC9MrJzifxBICwmvD028qnwQm09EP8z4ICZptD6wPdRTDzduc
KBuAX+zn8pNrJgyrheRKpPgno9KsbCzK4D/RIt1sTK2M3vVOtY+vpsN70QYUXvQ4
r2RZac3omlT43x5lddPxIlcouQpwWcVvr/K+Va770MRrjn88PBrJmvsEw/QYVBXp
Gxv2b78HFDacba80sMIm8ltRdqUCa5qIc6OATsz7izCQXEbkTEeESrcK1MA=
-----END X509 CRL-----
```

anhu, 2025-03-27:
Hi @onepeople158 

Interesting catch!  I'll have to look into this.  Please stay tuned. 

Warm regards, Anthony

onepeople158, 2025-06-27:
> Hi [@onepeople158](https://github.com/onepeople158) , can you please let us know how you generated that CRL?

Hello, I generated this CRL file using another ASN.1 library.

onepeople158, 2025-06-28:
> That is fine. Please share instructions for creating it. If you would like to keep it private email support at wolfssl dot com. Thanks

Hello, I have parsed the CRL file using the ASN.1 library and created two AKI extensions within it.

closed 2025-06-28 by douzzer