wolfSSL/wolfssl #8592: [Bug]: Verifying a certificate with SAN extension

dulanshuangqiao, 2025-03-26:
### Contact Details

bingxiaoganel@gmail.com

### Version

ubutun 5.6.4

### Description

./configure
make
sudo make install
./testsuite/testsuite.test
wolfSSL is configured and built by default

### Reproduction steps

```
// gcc -g verify.c -o verify -lwolfssl
#include <stdlib.h>
#include <wolfssl/ssl.h>
#include <wolfssl/options.h>
#include <wolfssl/wolfcrypt/error-crypt.h>

int wolfSSL_Verify_PEM(char * cac, char * ec){
int ret = 0;

// to create a new wolfSSL cert manager
WOLFSSL_CERT_MANAGER* cm;
cm = wolfSSL_CertManagerNew();
if (cm == NULL){
	printf("Creating a new wolfSSL_CertManager failed!\n");
	exit(1);
}

// to load cac to the created wolfSSL_CertManager
ret = wolfSSL_CertManagerLoadCA(cm, cac, NULL);
if (ret != SSL_SUCCESS){
	printf("Loading cac to the created wolfSSL_CertManager failed!\n");
	exit(2);
}

// to verify the ec in the created wolfSSL_CertManager
ret = wolfSSL_CertManagerVerify(cm, ec, SSL_FILETYPE_PEM);
if (ret != SSL_SUCCESS){
	printf("wolfSSL_CertManagerVerify filed and with return code %d and error message %s\n",
		 ret,
		 wolfSSL_ERR_reason_error_string(ret));
}
else{
	printf("The target cert has passed through verification.\n");
}

// to free cm
wolfSSL_CertManagerFree(cm);

return ret;
}

int main(int argc, char ** argv){
char * cac = argv[1];
char * ec = argv[2];
wolfSSL_Verify_PEM(cac, ec);
return 0;
}
```
I use this program to verify certificates. 
Actual result:
Verifying certificate: Cert174107799115M5.pem
The target cert Cert174107799115M5.pem has passed verification.
Expected result:
WolfSSL and other TLS implementations have the same verification results：Verify failed.
OpenSSL:
Verifying Cert174107799115M5.pem
error 1 at 0 depth lookup: unspecified certificate verification error
error Cert174107799115M5.pem: verification failed
40C7FF0127780000:error:0580009E:x509 certificate routines:ossl_x509v3_cache_extensions:reason(158):../crypto/x509/v3_purp.c:635:
GnuTLS:
Verifying Cert174107799115M5.pem
Loaded CAs (1 available)|<3>| ASSERT: x509_ext.c[gnutls_x509_ext_import_subject_alt_names]:234|<3>| ASSERT: x509.c[cache_alt_names]:397|<3>| ASSERT: x509.c[gnutls_x509_crt_import]:835|<3>| ASSERT: x509.c[gnutls_x509_crt_list_import]:3925
error parsing CRTs: ASN1 parser: Error in DER parsing.
I provided this test certificate

[SANtest.zip](https://github.com/user-attachments/files/19463973/SANtest.zip)

### Relevant log output

```shell

```

anhu, 2025-03-26:
hello @dulanshuangqiao , 

Thank you so much for helping us find these issues.  I will be looking into these problems you have found. 

It will take me some time to reproduce what you are seeing.  In the meantime, can you let us know a bit about yourself and your project: 

- Where are you geographically located? 
- What are the goals of your project? 
- Is your effort out of personal, professional or academic interest? 
- Any other information you care to share about yourself and your project. 

If you would prefer a secure and confidential channel to submit this personal information, you can send it to support@wolfssl.com. 

Warm regards, Anthony

dulanshuangqiao, 2025-03-29:
> 我不熟悉您的用户 ID。我们之前有互动过吗？
Hello, it seems we haven't interacted before.
This is my personal project. I am working on defect detection related to digital certificate verification.

dulanshuangqiao, 2025-03-31:
> hello [@dulanshuangqiao](https://github.com/dulanshuangqiao) ,
> 
> Thank you so much for helping us find these issues. I will be looking into these problems you have found.
> 
> It will take me some time to reproduce what you are seeing. In the meantime, can you let us know a bit about yourself and your project:
> 
> * Where are you geographically located?
> * What are the goals of your project?
> * Is your effort out of personal, professional or academic interest?
> * Any other information you care to share about yourself and your project.
> 
> If you would prefer a secure and confidential channel to submit this personal information, you can send it to [support@wolfssl.com](mailto:support@wolfssl.com).
> 
> Warm regards, Anthony

> *This is my research project. Due to review issues, I cannot provide you with information here.
> *Enhancing TLS security
> *Academic
> *As mentioned before, I will provide you with it when it is convenient.

dulanshuangqiao, 2025-03-31:
> hello [@dulanshuangqiao](https://github.com/dulanshuangqiao) ,
>
> Thank you so much for helping us find these issues. I will be looking into these problems you have found.
>
> It will take me some time [...]

1>This is my research project. Due to review issues, I cannot provide you with information here.
2>Enhancing TLS security
3>Academic
4>As mentioned before, I will provide you with it when it is convenient.

anhu, 2025-04-02:
Hello @dulanshuangqiao , 

This is Anthony again.  Please do understand that we use the information you provide to prioritize our work.  The more we understand the more we can figure out if our interests align and the higher a priority we will give. 

I did follow your steps and can confirm the following: 

```
./verify SANtest/RootCA.pem SANtest/SANtest.pem
The target cert has passed through verification.
```

When I further inspect the the certificate I can see there is a 150 byte octet string inside the SAN. I'll need to further look into why we are accepting this certificate. 

Please stay tuned. 

Warm regards, Anthony

dulanshuangqiao, 2025-04-02:
> 你好[@dulanshuangqiao](https://github.com/dulanshuangqiao)，
> 
> 我是 Anthony。请您理解，我们会使用您提供的信息来确定工作优先顺序。我们了解得越多，就越能确定我们的利益是否一致，我们就会给予更高的优先顺序。
> 
> 我确实按照您的步骤操作并可以确认以下内容：
> 
> ```
> ./verify SANtest/RootCA.pem SANtest/SANtest.pem
> The target cert has passed through verification.
> ```
> 
> 当我进一步检查证书时，我发现 SAN 中有一个 150 字节的八位字节字符串。我需要进一步研究我们接受此证书的原因。
> 
> 请继续关注。
> 
> 致以诚挚的问候，安东尼

Ok, looking forward to your results.

anhu, 2025-04-02:
Hi  @dulanshuangqiao ,

upon further thought, I think we are doing the correct thing here.  We are ignoring the content of the SAN extension.  It is our option to do so because the extension is NOT marked as `Critical`.  

Warm regards, Anthony

anhu, 2025-04-04:
Hi @dulanshuangqiao , 

I think my previous message expresses our position quite succinctly.  As such I will now close this issue.  That said, if you have further comments, please do re-open this issue.  

Warm regards, Anthony

dulanshuangqiao, 2025-04-15:
> 你好[@dulanshuangqiao](https://github.com/dulanshuangqiao)，
> 
> 我认为我之前的帖子已经非常简洁地表达了我们的立场。因此，我现在将关闭此问题。不过，如果您还有其他意见，请重新打开此问题。
> 
> 致以诚挚的问候，安东尼

I tested again using a certificate with the SAN extension marked as "critical" and WolfSSL still had differences with OpenSSL and GnuTLS.
I provided the certificate I used. [SANtest_.zip](https://github.com/user-attachments/files/19757720/SANtest_.zip)
I do not have permission to reopen this issue, so I am replying here. May I ask if you can handle the issue again.

jackctj117, 2025-05-02:
Hi @dulanshuangqiao ,

I have successfully reproduced the incorrectly passing of verification with your C code using wolfSSL API and getting a similar error code with openSSL that you got. But what I am seeing on the openSSL side of things is `error 20 at 0 depth lookup: unable to get local issuer certificate` where I see you're getting `error 1 at 0 depth lookup: unspecified certificate verification error`. I will continue looking into the differences.

Best Regards,
- Jack Tjaden

closed 2025-11-11 by jackctj117