wolfSSL/wolfssl #8593: [Bug]: Verifying a certificate with Nameconstrains extension

dulanshuangqiao, 2025-03-26:
### Contact Details

bingxiaoganel@gmail.com

### Version

ubutun 5.6.4 & WolfSSL 5.7.6

### Description

./configure
make
sudo make install
./testsuite/testsuite.test
wolfSSL is configured and built by default

### Reproduction steps

```
// gcc -g verify.c -o verify -lwolfssl
#include <stdlib.h>
#include <wolfssl/ssl.h>
#include <wolfssl/options.h>
#include <wolfssl/wolfcrypt/error-crypt.h>

int wolfSSL_Verify_PEM(char * cac, char * ec){
int ret = 0;

// to create a new wolfSSL cert manager
WOLFSSL_CERT_MANAGER* cm;
cm = wolfSSL_CertManagerNew();
if (cm == NULL){
	printf("Creating a new wolfSSL_CertManager failed!\n");
	exit(1);
}

// to load cac to the created wolfSSL_CertManager
ret = wolfSSL_CertManagerLoadCA(cm, cac, NULL);
if (ret != SSL_SUCCESS){
	printf("Loading cac to the created wolfSSL_CertManager failed!\n");
	exit(2);
}

// to verify the ec in the created wolfSSL_CertManager
ret = wolfSSL_CertManagerVerify(cm, ec, SSL_FILETYPE_PEM);
if (ret != SSL_SUCCESS){
	printf("wolfSSL_CertManagerVerify filed and with return code %d and error message %s\n",
		 ret,
		 wolfSSL_ERR_reason_error_string(ret));
}
else{
	printf("The target cert has passed through verification.\n");
}

// to free cm
wolfSSL_CertManagerFree(cm);

return ret;
}

int main(int argc, char ** argv){
char * cac = argv[1];
char * ec = argv[2];
wolfSSL_Verify_PEM(cac, ec);
return 0;
}
```
I use this program to verify certificates.
Actual result:
Verifying certificate: Cert174107808016M7.pem
wolfSSL_CertManagerVerify failed with return code -198 and error message Name Constraint error
Cert174107808016M7.pem failed verification.
Expected result:
WolfSSL and other TLS implementations have the same verification results：Pass verification.
OpenSSL:
Verifying Cert174107808016M7.pem
Cert174107808016M7.pem: OK
GnuTLS:
Verifying Cert174107808016M7.pem
Loaded CAs (1 available)|<3>| ASSERT: name_constraints.c[extract_name_constraints]:183|<3>| ASSERT: name_constraints.c[extract_name_constraints]:183|<3>| ASSERT: x509.c[gnutls_x509_crt_get_authority_key_id]:1691|<3>| ASSERT: x509.c[gnutls_x509_crt_get_authority_key_id]:1691|<3>| ASSERT: name_constraints.c[gnutls_x509_crt_get_name_constraints]:592
Output: Verified. The certificate is trusted. 
Chain verification output: Verified. The certificate is trusted. 
I provided this test certificate:[NCtest.zip](https://github.com/user-attachments/files/19464097/NCtest.zip)

### Relevant log output

```shell

```

anhu, 2025-03-26:
hello @dulanshuangqiao ,

Thank you so much! I will be looking into these problems you have found.

Please do take a moment to answer the questions on the other ticket that was recently opened. 

Warm regards, Anthony

dulanshuangqiao, 2025-04-01:
> hello [@dulanshuangqiao](https://github.com/dulanshuangqiao) ,
> 
> Thank you so much! I will be looking into these problems you have found.
> 
> Please do take a moment to answer the questions on the other ticket that was recently opened.
> 
> Warm regards, Anthony

There is no staff to handle this report for a long time.I hope a developer can review my report，This is very important for my work.

anhu, 2025-04-02:
Hi @dulanshuangqiao 

I am able to confirm that I see the following: 

```
$ ./verify NCtest/RootCA.pem   NCtest/NC.pem 
wolfSSL_CertManagerVerify filed and with return code -198 and error message Name Constraint error
```

When I use openssl to parse the certificate I see the follow: 

```
                Permitted:
                  email:oBy2Xk.wlSf.xi
                  DNS:LfdaOnX1BX.E=XLDf3.YNnm
                  IP:16.150.61.136/255.255.255.0
                Excluded:
                  email:VIPPUOz.7LB.Q34E
                  DNS:eKbAa9Sb.QzaCdk.OG.Sw
                  IP:254.78.38.30/255.255.255.0
```

Are you expecting us to return an  error because the email addresses and DNS fields are  not a properly formatted? 

Warm regards, Anthony

dulanshuangqiao, 2025-04-02:
> 你好[@dulanshuangqiao](https://github.com/dulanshuangqiao)
> 
> 我可以确认我看到了以下内容：
> 
> ```
> $ ./verify NCtest/RootCA.pem   NCtest/NC.pem 
> wolfSSL_CertManagerVerify filed and with return code -198 and error message Name Constraint error
> ```
> 
> 当我使用 openssl 解析证书时，我看到以下内容：
> 
> ```
>                 Permitted:
>                   email:oBy2Xk.wlSf.xi
>                   DNS:LfdaOnX1BX.E=XLDf3.YNnm
>                   IP:16.150.61.136/255.255.255.0
>                 Excluded:
>                   email:VIPPUOz.7LB.Q34E
>                   DNS:eKbAa9Sb.QzaCdk.OG.Sw
>                   IP:254.78.38.30/255.255.255.0
> ```
> 
> 您是否希望我们返回错误，因为电子邮件地址和 DNS 字段的格式不正确？
> 
> 致以诚挚的问候，安东尼

Thank you for your reply. I made this report based on the results of the difference test, that is, my expected result is that WolfSSL should pass the verification like OpenSSL and GnuTLS, but the current result is that the verification fails. Therefore, I made this bug report.

closed 2025-04-04 by anhu

dulanshuangqiao, 2025-04-15:
> 你好[@dulanshuangqiao](https://github.com/dulanshuangqiao)，
> 
> 我认为我之前的帖子已经非常简洁地表达了我们的立场。因此，我现在将关闭此问题。不过，如果您还有其他意见，请重新打开此问题。
> 
> 致以诚挚的问候，安东尼

OpenSSL believes that it is meaningless to have this extension in an EE certificate, so it ignores the extension and passes the verification. RFC5280 mentions the name constraints extension, which MUST be used only in a CA certificate.
I do not have permission to reopen this issue, so I am replying here. May I ask if you can handle the issue again.

anhu, 2025-04-15:
Hello @dulanshuangqiao , 

I think our position has not changed.  We feel our behaviour is correct, but if you would like to match openSSL's behaviour you can simply define `WOLFSSL_NO_ASN_STRICT` .  I see no need to re-open the ticket.