wolfSSL/wolfssl #8597: [Bug]:wolfSSL accepts the incorrect generalTime value.

onepeople158, 2025-03-27:
### Version

I am using version 0.1.7 of the wolfssl Command Line Utility. Linked to wolfSSL version 5.7.6

### Description

The RFC standard for X.509 CRLs restricts the thisUpdate field to only two formats, namely UTCTime (YYMMDDHHMMSSZ) and GeneralizedTime (YYYYMMDDHHMMSSZ) in ASN.1 representation, which are 13 and 15 characters wide, respectively. However,wolfSSL accepts the thisUpdate field type as generalTime with a length of 13 ("240123000000Z") in the CRL.

### Reproduction steps

wolfssl crl -inform der -in crl_file_13gn.der -text

[crl_file_13gn.zip](https://github.com/user-attachments/files/19478166/crl_file_13gn.zip)

### Relevant log output

```shell
wolfSSL Entering wolfSSL_Init
wolfSSL Entering wolfCrypt_Init
RNG_HEALTH_TEST_CHECK_SIZE = 128
sizeof(seedB_data)         = 128
opened /dev/urandom.
rnd read...
wolfSSL Entering wolfSSL_BIO_new_file
wolfSSL Entering wolfSSL_BIO_s_file
wolfSSL Entering wolfSSL_BIO_new
wolfSSL Entering wolfSSL_BIO_set_fp
wolfSSL Entering wolfSSL_BIO_get_len
wolfSSL Entering wolfSSL_BIO_get_fp
wolfSSL Entering wolfSSL_BIO_read
wolfSSL Entering wolfSSL_d2i_X509_CRL
wolfSSL Entering InitCRL
wolfSSL Entering BufferLoadCRL
InitDecodedCRL
ParseCRL
About to verify CRL signature
Did NOT find CRL issuer CA
ERR TRACE: wolfcrypt/src/asn.c L 38604 ASN_CRL_NO_SIGNER_E (-190)
wolfSSL Entering AddCRL
wolfSSL Entering InitCRL_Entry
wolfSSL Entering wolfSSL_d2i_X509_NAME
Getting Name
Getting Cert Name
wolfSSL Entering wolfSSL_X509_NAME_new_ex
wolfSSL Entering wolfSSL_X509_NAME_add_entry_by_NID
Found place for name entry
wolfSSL Entering wolfSSL_sk_X509_NAME_new
wolfSSL Entering wolfSSL_sk_new_node
wolfSSL Entering wolfSSL_sk_push
wolfSSL Entering wolfSSL_sk_insert
wolfSSL Entering wolfSSL_X509_NAME_ENTRY_free
wolfSSL Entering wolfSSL_X509_NAME_add_entry_by_NID
Found place for name entry
wolfSSL Entering wolfSSL_sk_push
wolfSSL Entering wolfSSL_sk_insert
wolfSSL Entering wolfSSL_sk_new_node
wolfSSL Entering wolfSSL_X509_NAME_ENTRY_free
wolfSSL Entering wolfSSL_X509_NAME_add_entry_by_NID
Found place for name entry
wolfSSL Entering wolfSSL_sk_push
wolfSSL Entering wolfSSL_sk_insert
wolfSSL Entering wolfSSL_sk_new_node
wolfSSL Entering wolfSSL_X509_NAME_ENTRY_free
wolfSSL Entering wolfSSL_X509_NAME_add_entry_by_NID
Found place for name entry
wolfSSL Entering wolfSSL_sk_push
wolfSSL Entering wolfSSL_sk_insert
wolfSSL Entering wolfSSL_sk_new_node
wolfSSL Entering wolfSSL_X509_NAME_ENTRY_free
wolfSSL Entering wolfSSL_X509_NAME_add_entry_by_NID
Found place for name entry
wolfSSL Entering wolfSSL_sk_push
wolfSSL Entering wolfSSL_sk_insert
wolfSSL Entering wolfSSL_sk_new_node
wolfSSL Entering wolfSSL_X509_NAME_ENTRY_free
wolfSSL Entering wolfSSL_X509_NAME_add_entry_by_NID
Found place for name entry
wolfSSL Entering wolfSSL_sk_push
wolfSSL Entering wolfSSL_sk_insert
wolfSSL Entering wolfSSL_sk_new_node
wolfSSL Entering wolfSSL_X509_NAME_ENTRY_free
wolfSSL Entering wolfSSL_X509_NAME_new_ex
wolfSSL Entering wolfSSL_X509_NAME_copy
wolfSSL Entering wolfSSL_sk_X509_NAME_new
wolfSSL Entering wolfSSL_sk_new_node
wolfSSL Entering wolfSSL_sk_push
wolfSSL Entering wolfSSL_sk_insert
wolfSSL Entering wolfSSL_sk_push
wolfSSL Entering wolfSSL_sk_insert
wolfSSL Entering wolfSSL_sk_new_node
wolfSSL Entering wolfSSL_sk_push
wolfSSL Entering wolfSSL_sk_insert
wolfSSL Entering wolfSSL_sk_new_node
wolfSSL Entering wolfSSL_sk_push
wolfSSL Entering wolfSSL_sk_insert
wolfSSL Entering wolfSSL_sk_new_node
wolfSSL Entering wolfSSL_sk_push
wolfSSL Entering wolfSSL_sk_insert
wolfSSL Entering wolfSSL_sk_new_node
wolfSSL Entering wolfSSL_sk_push
wolfSSL Entering wolfSSL_sk_insert
wolfSSL Entering wolfSSL_sk_new_node
wolfSSL Entering wolfSSL_X509_NAME_free
wolfSSL Entering wolfSSL_sk_free
FreeDecodedCRL
wolfSSL Entering wolfSSL_BIO_s_file
wolfSSL Entering wolfSSL_BIO_new
wolfSSL Entering wolfSSL_BIO_set_fp
wolfSSL Entering wolfSSL_BIO_write
Certificate Revocation List (CRL):
wolfSSL Entering wolfSSL_BIO_write
wolfSSL Entering wolfSSL_BIO_write
        Version: 2 (0x1)
wolfSSL Entering wolfSSL_X509_CRL_get_signature
wolfSSL Entering wolfSSL_X509_CRL_get_signature
wolfSSL Entering wolfSSL_BIO_write
wolfSSL Entering wolfSSL_OBJ_obj2txt
wolfSSL Entering wolfSSL_OBJ_nid2ln
wolfSSL Entering wolfSSL_BIO_write
        Signature Algorithm: sha256WithRSAEncryption
wolfSSL Entering wolfSSL_BIO_write
wolfSSL Entering wolfSSL_X509_NAME_print_ex
wolfSSL Entering wolfSSL_BIO_write
wolfSSL Entering wolfSSL_X509_NAME_entry_count
wolfSSL Leaving wolfSSL_X509_NAME_entry_count, return 6
wolfSSL Entering wolfSSL_X509_NAME_ENTRY_get_data
wolfSSL Entering wolfSSL_BIO_write
wolfSSL Entering wolfSSL_X509_NAME_ENTRY_get_data
wolfSSL Entering wolfSSL_BIO_write
wolfSSL Entering wolfSSL_X509_NAME_ENTRY_get_data
wolfSSL Entering wolfSSL_BIO_write
wolfSSL Entering wolfSSL_X509_NAME_ENTRY_get_data
wolfSSL Entering wolfSSL_BIO_write
wolfSSL Entering wolfSSL_X509_NAME_ENTRY_get_data
wolfSSL Entering wolfSSL_BIO_write
wolfSSL Entering wolfSSL_X509_NAME_ENTRY_get_data
wolfSSL Entering wolfSSL_BIO_write
wolfSSL Entering wolfSSL_BIO_write
        Issuer:  C=US, ST=US, L=US, O=US, CN=US, OU=US
wolfSSL Entering wolfSSL_BIO_write
wolfSSL Entering wolfSSL_BIO_write
wolfSSL Entering wolfSSL_BIO_write
        Last Update: Jan 23 00:00:00 2024 GMT
wolfSSL Entering wolfSSL_BIO_write
wolfSSL Entering wolfSSL_BIO_write
wolfSSL Entering wolfSSL_BIO_write
        Next Update: Jan 20 00:00:00 2034 GMT
wolfSSL Entering wolfSSL_BIO_write
        CRL extensions:
wolfSSL Entering wolfSSL_BIO_write
            X509v3 CRL Number:
wolfSSL Entering wolfSSL_BIO_write
                214884672
wolfSSL Entering wolfSSL_BIO_write
Revoked Certificates:
wolfSSL Entering wolfSSL_X509_REVOKED_get_serial_number
wolfSSL Entering wolfSSL_BIO_write
wolfSSL Entering wolfSSL_BIO_write
    Serial Number:
        1c80022ef81f2405ee96a612dcb61fe0ac701e5e
wolfSSL Entering wolfSSL_BIO_write
wolfSSL Entering wolfSSL_BIO_write
wolfSSL Entering wolfSSL_BIO_write
        Revocation Date: Mar 13 02:44:40 2025 GMT
wolfSSL Entering wolfSSL_X509_CRL_get_signature
wolfSSL Entering wolfSSL_X509_CRL_get_signature
wolfSSL Entering wolfSSL_BIO_write
wolfSSL Entering wolfSSL_OBJ_obj2txt
wolfSSL Entering wolfSSL_OBJ_nid2ln
wolfSSL Entering wolfSSL_BIO_write
    Signature Algorithm: sha256WithRSAEncryption
wolfSSL Entering wolfSSL_BIO_write
wolfSSL Entering wolfSSL_BIO_write
         68:36:3d:8c:17:40:26:20:2e:8d:49:1a:2d:82:d5:b7:33:56:
wolfSSL Entering wolfSSL_BIO_write
         17:fd:12:c4:3e:42:07:87:58:21:c6:4c:aa:d3:ca:2e:7e:72:
wolfSSL Entering wolfSSL_BIO_write
         91:cc:64:5e:f9:d1:6d:58:a1:27:e2:a7:00:0b:fd:16:49:f9:
wolfSSL Entering wolfSSL_BIO_write
         8b:08:fb:ec:41:b0:c4:d8:f2:66:4b:50:e2:00:26:70:c8:42:
wolfSSL Entering wolfSSL_BIO_write
         4c:11:1c:00:76:e6:8b:dd:ad:1e:db:68:b7:d4:ab:e3:8f:82:
wolfSSL Entering wolfSSL_BIO_write
         37:ed:0d:69:a4:03:39:f9:48:79:5c:3b:66:2a:fd:d1:35:ae:
wolfSSL Entering wolfSSL_BIO_write
         7e:34:9c:cb:cf:de:ec:59:15:9b:e6:83:e4:28:9c:ad:b2:56:
wolfSSL Entering wolfSSL_BIO_write
         aa:87:b6:d6:90:75:43:58:d2:e3:d8:8d:ad:9c:ea:67:6b:f0:
wolfSSL Entering wolfSSL_BIO_write
         1c:b8:aa:34:0c:e9:79:cc:70:52:28:7f:60:f7:b9:f8:20:64:
wolfSSL Entering wolfSSL_BIO_write
         e4:da:b8:bc:80:9e:89:e1:95:0b:f2:4e:f6:be:52:91:d0:f0:
wolfSSL Entering wolfSSL_BIO_write
         59:04:e5:d0:8d:e4:48:ae:a7:e0:98:7b:e7:71:66:21:e9:fc:
wolfSSL Entering wolfSSL_BIO_write
         d0:5d:99:66:e6:6e:e3:f7:e1:27:b0:b8:ae:5a:fa:5b:d8:ba:
wolfSSL Entering wolfSSL_BIO_write
         16:b2:b4:ea:ce:66:93:53:de:60:51:ca:84:29:30:23:cc:29:
wolfSSL Entering wolfSSL_BIO_write
         f1:c2:2e:74:94:03:94:bb:0a:da:ee:02:4d:cb:93:29:d6:c3:
wolfSSL Entering wolfSSL_BIO_write
         2e:cb:33:60
wolfSSL Entering wolfSSL_BIO_write
-----BEGIN X509 CRL-----
MIIB7DCB1QIBATANBgkqhkiG9w0BAQsFADBOMQswCQYDVQQGEwJVUzELMAkGA1UE
CAwCVVMxCzAJBgNVBAcMAlVTMQswCQYDVQQKDAJVUzELMAkGA1UEAwwCVVMxCzAJ
BgNVBAsMAlVTGA0yNDAxMjMwMDAwMDBaGA0zNDAxMjAwMDAwMDBaMDUwMwIUHIAC
LvgfJAXulqYS3LYf4KxwHl4XDTI1MDMxMzAyNDQ0MFowDDAKBgNVHRUEAwoBBqAc
MBowGAYDVR0UBBECDxnP/97adO3y9qRGDM7hQDANBgkqhkiG9w0BAQsFAAOCAQEA
aDY9jBdAJiAujUkaLYLVtzNWF/0SxD5CB4dYIcZMqtPKLn5ykcxkXvnRbVihJ+Kn
AAv9Fkn5iwj77EGwxNjyZktQ4gAmcMhCTBEcAHbmi92tHttot9Sr44+CN+0NaaQD
OflIeVw7Zir90TWufjScy8/e7FkVm+aD5CicrbJWqoe21pB1Q1jS49iNrZzqZ2vw
HLiqNAzpecxwUih/YPe5+CBk5Nq4vICeieGVC/JO9r5SkdDwWQTl0I3kSK6n4Jh7
53FmIen80F2ZZuZu4/fhJ7C4rlr6W9i6FrK06s5mk1PeYFHKhCkwI8wp8cIudJQD
lLsK2u4CTcuTKdbDLsszYA==
-----END X509 CRL-----
wolfSSL Entering wolfSSL_X509_CRL_free
wolfSSL Entering FreeCRL
wolfSSL Entering FreeCRL_Entry
wolfSSL Entering wolfSSL_sk_free
wolfSSL Entering wolfSSL_BIO_free
wolfSSL Entering wolfSSL_BIO_free
wolfSSL Entering wolfSSL_Cleanup
wolfSSL Entering wolfCrypt_Cleanup
```

anhu, 2025-03-27:
Hi @onepeople158 , 

Sorry, can you help me understand what is the difference between generalTime and GeneralizedTime?

anhu, 2025-03-27:
Ah...I think I understand.  You are saying that its GeneralizedTime but in the length/format of UTC Time.  Ok.  Let me look into this.

anhu, 2025-03-28:
https://github.com/wolfSSL/wolfssl/pull/8603