wolfSSL/wolfssl #8610: [Bug]: wolfSSL_PEM_write_PKCS8PrivateKey fails for no known reason

ribes96, 2025-03-31:
### Contact Details

aribes96@gmail.com

### Version

5.7.6

### Description

I am reading a private key in PEM format, both in PKCS#1 and in PKCS#8, both unencrypted and encrypted, using `wolfSSL_PEM_read_bio_PrivateKey`.

Then I write the private key in PEM format, in PKCS#8, unencrypted, using `wolfSSL_PEM_write_PKCS8PrivateKey`.

`wolfSSL_PEM_write_PKCS8PrivateKey` returns error 0 for some of the cases, but then `wolfSSL_ERR_get_error` reports no error.

The test cases are:
- Plain PKCS#1: Success
- Plain PKCS#8: Error writing the private key
- Encrypted PKCS#1: Error **reading** the private key
- Encrypted PKCS#8: Error writing the private key

### Reproduction steps

~~~sh
./configure --host=arm-linux --enable-opensslextra --enable-opensslall --enable-certgen --enable-libwebsockets --disable-tls13 --disable-examples CFLAGS="-DHAVE_EX_DATA -DNO_DES3 -DNO_DSA -DNO_OLD_TLS" LDFLAGS="-lrt"
~~~

Minimal example
~~~c
#include <wolfssl/options.h>
#include <wolfssl/ssl.h>
#include <stdio.h>

static const char *g_plain_pk1 =
"-----BEGIN RSA PRIVATE KEY-----"
"MIIEowIBAAKCAQEA6VPEs42n/PvfZXqG96QZw/RwROgqUqlsw8/2axg0XVovqHPn"
"zsKpNrMbaseZwu4JOPMQv6YxS/L2FdrXPPPfbLy4GNxO/wRUAzQSiXFqpWNeweNK"
"3JfIomeHZeIrIGnbN0P9Gb/kEZjsl1019A+xNEOJtbUZolwceis+ZXDQddpwDLRp"
"IsLt4mxEwaLRnIT6DMo1hlXHpn0Q8e1sTuih5icBD3E/3eC8CqHPKKONkttjIRAd"
"TjDc6Ld9NcCokddqYbMiq+PZHFYnhMYkFWho/KT45T+4aeM8bB1FiI9IieyzYryS"
"BhexaKZ6iGk53U7bl++d5Eh4zy+EXHpmB7LyAQIDAQABAoIBAAPpprPIHVgNtTpT"
"aq3+b3wsvjE5n3QRaUSDWUOEfWRPXds6TlnWQ9Yp/Ly2qiptqXPuB4rfM5ZtdZj6"
"OX/JKyb8py2rjBuUq//9wI3TfBFX8D/aGuhGF1Q53jhkKYr6m3V6y2TCJgXxhY4o"
"Pv5fwP36X/XPGqXxW3zk+r/iu2wufF1mptFH3DgiroSGcFeRZQ9p8UPBKWFBr6/j"
"f7Exf6z6VtbRarZVIVduvq2Xrh2BPvgnZFuEN1RqrtfgqGAdA8yE12GYrRkfVlUB"
"Ne4Tf8tSDRARi5RVWq4CyKEBGB2FKjEPihE6J0Bk81Qkmx2vtPpcMqlUfTd0VU2U"
"2Pe3/Z0CgYEA+u0Wirr/zLuYP9mJA2mf369FqcpfR8JhWJAwUweHVfYF3DdRwl2P"
"lkIkgM3wi2pIkEXDC+AsN2fsW9KD0uUrggUKLgduIHVihrSIUo2qdf3Ote2WJDcg"
"xF89u0oVduis5CAUFPKE9Sihijg8t7RlGIvKVMuMlAS2dsW84gXfnEMCgYEA7guU"
"gBStlSgEFdzuCzotZpS5Q9OflP23Vi43QaYkTA9W/taqbmsn/kkNPJKFa6MD6+vB"
"ayHyR7bVFhN0rlxzF0qTFSbkcEFJ8VSoSW1R5UtG6jCXQjtk0wteHBz/Q9weGb+T"
"fodK4O1HZA/l3hpHe0leHhzkTDqfFCGJcT13tmsCgYAPoQjQZ0lbhSu8fl4sWj3c"
"/tvF8Zbxr0fE7mY0JbnmPsaqqr9q7cztvPztEOa0qFAQvGejq7jVUqHELl6Lmj6V"
"gJ7bsxbG6sjSPxhmm/BMHRzs8/lwFAb0dEIyEjE4ENID+4OEsn5ZjshRBaOyhYfu"
"eXaGxY7iAVEkILjdEySUZQKBgC3G5lDXKSZk4pw9tz1Le82dTh9gA/FH9UjKbG/g"
"flxuGsG5EF9ZlgBNlhEO8D9pwuhMlw5+IbRu0X2EUkzxlGRnP++yYf3AYHLbX912"
"+2fH2SRSIdRnzZTUO83WObSYEOXMtJWBeQHWP4Nbl03l6dconcPWtISARXU+mcrA"
"ElRvAoGBALvyZZZSSdYOm3/5MuQAvKBrDL/6WDriCBs18t6OHvd+2NkWAPymOkQu"
"rRh63W8CZ2OMo9uGHRs7Vttn7N9m97drIjoJxMACzBmGByYwnW6WDOBltMWTXvCR"
"vTUreYym99Acf+mr0uiKcsV6od5Z07s4TAHEwMe9mvqH6j8R3QmS"
"-----END RSA PRIVATE KEY-----"
;

static const char *g_encrypted_pk1 =
"-----BEGIN RSA PRIVATE KEY-----"
"Proc-Type: 4,ENCRYPTED"
"DEK-Info: AES-128-CBC,C32F122386CC91D5A9E1E86406C22C7B"
""
"L8l2fpn0cYYm0Gigs2aku16hrRW6gdax9a5yCp55lBDAODS/9YjObxO6EQjEIIDb"
[... 174 lines cut ...]
LjdBpiRMD1b+1qpuayf+SQ08koVrowPr68FrIfJHttUWE3SuXHMXSpMVJuRwQUnx
VKhJbVHlS0bqMJdCO2TTC14cHP9D3B4Zv5N+h0rg7UdkD+XeGkd7SV4eHORMOp8U
IYlxPXe2awKBgA+hCNBnSVuFK7x+XixaPdz+28XxlvGvR8TuZjQlueY+xqqqv2rt
zO28/O0Q5rSoUBC8Z6OruNVSocQuXouaPpWAntuzFsbqyNI/GGab8EwdHOzz+XAU
BvR0QjISMTgQ0gP7g4SyflmOyFEFo7KFh+55dobFjuIBUSQguN0TJJRlAoGALcbm
UNcpJmTinD23PUt7zZ1OH2AD8Uf1SMpsb+B+XG4awbkQX1mWAE2WEQ7wP2nC6EyX
Dn4htG7RfYRSTPGUZGc/77Jh/cBgcttf3Xb7Z8fZJFIh1GfNlNQ7zdY5tJgQ5cy0
lYF5AdY/g1uXTeXp1yidw9a0hIBFdT6ZysASVG8CgYEAu/JlllJJ1g6bf/ky5AC8
oGsMv/pYOuIIGzXy3o4e937Y2RYA/KY6RC6tGHrdbwJnY4yj24YdGztW22fs32b3
t2siOgnEwALMGYYHJjCdbpYM4GW0xZNe8JG9NSt5jKb30Bx/6avS6IpyxXqh3lnT
uzhMAcTAx72a+ofqPxHdCZI=
-----END PRIVATE KEY-----
bash $ wolfkey plain 8
Error writing private key: ok
bash $ wolfkey encrypted 1
Error reading private key: ok
bash $ wolfkey encrypted 8
Error writing private key: ok
~~~

### Relevant log output

```shell

```

ribes96, 2025-03-31:
`PEM_write_bio_PrivateKey` works fine, but can only write in PKCS#1.

So, seems the problem only happens when PKCS#8 is read and also written.

The error code path crosses this:

~~~c
    /* sanity check: make sure the key doesn't already have a PKCS 8 header */
    if (ToTraditionalInline_ex(key, &keyIdx, keySz, &tmpAlgId) >= 0) {
        (void)tmpAlgId;
        return ASN_PARSE_E;
    }
~~~

This is function `wc_CreatePKCS8Key`, in `wolfcrypt/src/asn.c:7334`

The comment makes me think it has something to do to having already read a PKCS8 header during the read of the key. It if helps...

kareem-wolfssl, 2025-03-31:
Hi @ribes96 ,

Thank you for your in-depth report.  I've reproduced it here and I my colleague @JacobBarthelmeh has posted a fix here: https://github.com/wolfSSL/wolfssl/pull/8612 which I've confirmed fixes the issue for me.  Please try applying it on your end and let me know if it helps.

ribes96, 2025-04-01:
The posted fix works in my end. Thanks!