wolfSSL/wolfssl #8615: [Bug]: WolfSSL accepts a certificate whose serial number is zero

Jennifer-first, 2025-04-01:
### Contact Details

zhanglin7120@gamail.com

### Version

WolfSSL 5.6.4

### Description

./configure
make
sudo make install
./testsuite/testsuite.test
wolfSSL is configured and built by default

### Reproduction steps

```
// gcc -g verify.c -o verify -lwolfssl
// gcc -g verify.c -o verify -lwolfssl
#include <stdlib.h>
#include <wolfssl/ssl.h>
#include <wolfssl/options.h>
#include <wolfssl/wolfcrypt/error-crypt.h>

int wolfSSL_Verify_PEM(char * cac, char * ec){
int ret = 0;

// to create a new wolfSSL cert manager
WOLFSSL_CERT_MANAGER* cm;
cm = wolfSSL_CertManagerNew();
if (cm == NULL){
	printf("Creating a new wolfSSL_CertManager failed!\n");
	exit(1);
}

// to load cac to the created wolfSSL_CertManager
ret = wolfSSL_CertManagerLoadCA(cm, cac, NULL);
if (ret != SSL_SUCCESS){
	printf("Loading cac to the created wolfSSL_CertManager failed!\n");
	exit(2);
}

// to verify the ec in the created wolfSSL_CertManager
ret = wolfSSL_CertManagerVerify(cm, ec, SSL_FILETYPE_PEM);
if (ret != SSL_SUCCESS){
	printf("wolfSSL_CertManagerVerify filed and with return code %d and error message %s\n",
		 ret,
		 wolfSSL_ERR_reason_error_string(ret));
}
else{
	printf("The target cert has passed through verification.\n");
}

// to free cm
wolfSSL_CertManagerFree(cm);

return ret;
}

int main(int argc, char ** argv){
char * cac = argv[1];
char * ec = argv[2];
wolfSSL_Verify_PEM(cac, ec);
return 0;
}
```
Actual result:
The target cert has passed through verification.
WolfSSL accepts a certificate whose serial number is zero which is forbidden by RFC 5280 and its errata. The certificate should be rejected according to Sec. 4.1.2.2 in RFC 5280 "The serial number MUST be a positive integer assigned by the CA to each certificate" and the errata #3200 "The serial number MUST be a positive non-zero integer assigned by the CA to each certificate".
OpenSSL:
openssl verify -CAfile ca.pem 14.pem
14.pem：OK
GnuTLS:
certtool --verify --load-ca-certificate=ca.pem < 14.pem
Chain verification output: Verified. The certificate is trusted.
I provided this test certificate:

[test.zip](https://github.com/user-attachments/files/19552177/test.zip)

### Relevant log output

```shell

```

kareem-wolfssl, 2025-04-01:
Hi @Jennifer-first ,

Thanks for the report.  This was a known issue previously and has been resolved since 5.6.4.  Please upgrade to 5.7.6 and retry your test.  We have changed the behavior to fail on certificates with a serial number of zero by default, as long as WOLFSSL_ASN_ALLOW_0_SERIAL or WOLFSSL_NO_ASN_STRICT are not defined.

Thanks,
Kareem

kareem-wolfssl, 2025-12-16:
Hi @jrosiek,

My apologies for my delayed response on this.  You bring up a very good point, I've brought this up to the team for consideration.  I'll let you know if we end up modifying this.

anhu, 2026-01-14:
My bad, the PR in question has not merged yet. Re-opened.