wolfSSL/wolfssl #8625: [Bug]: The wolfSSL incorrectly handled the keyIdentifier field

onepeople158, 2025-04-02:
### Version

I am using version 0.1.7 of the wolfssl Command Line Utility. Linked to wolfSSL version 5.7.6

### Description

Hello developer, I have a CRL file where the keyIdentifier field in the AKI extension has the value 43:4C:73:65:4E:62:44:33:69:4D:73:43:59:31:46:43:4D:45:30:46:2F:4E:35:61:65:61:63:0A. However, when I parse it using wolfSSL, I get keyid:59:E0:46:AE:2C:90:C5:7A:04:5B:C4:BF:54:FE:D3:0E:85:93:48:37. According to RFC 5280, "The keyIdentifier is composed of the 160-bit SHA-1 hash of the value of the BIT STRING subjectPublicKey (excluding the tag, length, and number of unused bits)." So, I wonder if this might be a bug?

### Reproduction steps

wolfssl crl -inform der -in crl_keyid.der -text

[crl_keyid.zip](https://github.com/user-attachments/files/19563596/crl_keyid.zip)

### Relevant log output

```shell
wolfSSL Entering wolfSSL_Init
wolfSSL Entering wolfCrypt_Init
RNG_HEALTH_TEST_CHECK_SIZE = 128
sizeof(seedB_data)         = 128
opened /dev/urandom.
rnd read...
wolfSSL Entering wolfSSL_BIO_new_file
wolfSSL Entering wolfSSL_BIO_s_file
wolfSSL Entering wolfSSL_BIO_new
wolfSSL Entering wolfSSL_BIO_set_fp
wolfSSL Entering wolfSSL_BIO_get_len
wolfSSL Entering wolfSSL_BIO_get_fp
wolfSSL Entering wolfSSL_BIO_read
wolfSSL Entering wolfSSL_d2i_X509_CRL
wolfSSL Entering InitCRL
wolfSSL Entering BufferLoadCRL
InitDecodedCRL
ParseCRL
wolfSSL Entering ParseCRL_AuthKeyIdExt
About to verify CRL signature
Did NOT find CRL issuer CA
ERR TRACE: wolfcrypt/src/asn.c L 38604 ASN_CRL_NO_SIGNER_E (-190)
wolfSSL Entering AddCRL
wolfSSL Entering InitCRL_Entry
wolfSSL Entering wolfSSL_d2i_X509_NAME
Getting Name
Getting Cert Name
wolfSSL Entering wolfSSL_X509_NAME_new_ex
wolfSSL Entering wolfSSL_X509_NAME_add_entry_by_NID
Found place for name entry
wolfSSL Entering wolfSSL_sk_X509_NAME_new
wolfSSL Entering wolfSSL_sk_new_node
wolfSSL Entering wolfSSL_sk_push
wolfSSL Entering wolfSSL_sk_insert
wolfSSL Entering wolfSSL_X509_NAME_ENTRY_free
wolfSSL Entering wolfSSL_X509_NAME_add_entry_by_NID
Found place for name entry
wolfSSL Entering wolfSSL_sk_push
wolfSSL Entering wolfSSL_sk_insert
wolfSSL Entering wolfSSL_sk_new_node
wolfSSL Entering wolfSSL_X509_NAME_ENTRY_free
wolfSSL Entering wolfSSL_X509_NAME_add_entry_by_NID
Found place for name entry
wolfSSL Entering wolfSSL_sk_push
wolfSSL Entering wolfSSL_sk_insert
wolfSSL Entering wolfSSL_sk_new_node
wolfSSL Entering wolfSSL_X509_NAME_ENTRY_free
wolfSSL Entering wolfSSL_X509_NAME_add_entry_by_NID
Found place for name entry
wolfSSL Entering wolfSSL_sk_push
wolfSSL Entering wolfSSL_sk_insert
wolfSSL Entering wolfSSL_sk_new_node
wolfSSL Entering wolfSSL_X509_NAME_ENTRY_free
[... 140 lines cut ...]
MIICxTCCAa0CAQEwDQYJKoZIhvcNAQELBQAwTjELMAkGA1UEBhMCVVMxCzAJBgNV
BAgMAlVTMQswCQYDVQQHDAJVUzELMAkGA1UECgwCVVMxCzAJBgNVBAMMAlVTMQsw
CQYDVQQLDAJVUxcNMjUwMTAxMDAwMDAwWhcNMjUxMjAxMDAwMDAwWjA1MDMCFByA
Ai74HyQF7pamEty2H+CscB5eFw0yNTAzMjcwMjUxMTBaMAwwCgYDVR0VBAMKAQag
gfMwgfAwgdMGA1UdIwSByzCByIAcQ0xzZU5iRDNpTXNDWTFGQ01FMEYvTjVhZWFj
CqGBoqSBnzCBnDELMAkGA1UEBgwCVVMxEzARBgNVBAgMCkNhbGlmb3JuaWExFjAU
BgNVBAcMDVNhbiBGcmFuY2lzY28xETAPBgNVBAoMCEZOTVQtUkNNMSMwIQYDVQQL
DBpFUwpPPW1rY2VydCBkZXZlbG9wbWVudCBDQTEoMCYGA1UEAwwfU2VlIHd3dy5l
bnRydXN0Lm5ldC9sZWdhbC10ZXJtc4IDAeJAMBgGA1UdFAQRAg8Zz//e2nTt8vak
RgzO4UAwDQYJKoZIhvcNAQELBQADggEBAKT3KuNz8tPCfcDFk1yEE0NxN9AAgcxF
thPtx2efNfGfxSu/mGFeLpj2eSUFSDamKE26UaAa3YfFScidlPwdbvTQ4LTEb+4F
R2nfrst3IkRvp7zmvST+GtGtxjDeZ4ViMpIIR/SwWDi+PFgj+q96TrfIp28O20gd
uR0gyLhx3rgoXIQzoL/dujBZmE6YWhWO4erBUQ4x1UkhSRInEkcf7dDifJrX8Aal
i8Pe9ubuAOW9PBRgH0ldFARNuUKqqRP1uVbeLQ5g30fInRLEMkWwn6hEygBXdGuy
eTIiODDY9wmlFlG7KtZGbBX0kRajycqIl5VJvI8+p//Hi61d6wy4km4=
-----END X509 CRL-----
wolfSSL Entering wolfSSL_X509_CRL_free
wolfSSL Entering FreeCRL
wolfSSL Entering FreeCRL_Entry
wolfSSL Entering wolfSSL_sk_free
wolfSSL Entering wolfSSL_BIO_free
wolfSSL Entering wolfSSL_BIO_free
wolfSSL Entering wolfSSL_Cleanup
wolfSSL Entering wolfCrypt_Cleanup
```

lealem47, 2025-04-02:
Hi @onepeople158,

> The keyIdentifier is composed of the 160-bit SHA-1 hash of the value of the BIT STRING subjectPublicKey (excluding the tag, length, and number of unused bits)

As you mention above, the keyIdentifier will be 160 bits long (20 bytes). The value that you are expecting (`43:4C:73:65:4E:62:44:33:69:4D:73:43:59:31:46:43:4D:45:30:46:2F:4E:35:61:65:61:63:0A`), is 28 bytes long and isn't compliant with the RFC.

onepeople158, 2025-04-03:
> Hi [@onepeople158](https://github.com/onepeople158),
>
>
> As you mention above, the keyIdentifier will be 160 bits long (20 bytes). The value that you are expecting [...]

Hello developer, sorry, although wolfSSL follows the rule that "The keyIdentifier is composed of the 160-bit SHA-1 hash of the value of the BIT STRING subjectPublicKey (excluding the tag, length, and number of unused bits)," there is another common method in RFC5280: "The keyIdentifier is composed of a four-bit type field with the value 0100 followed by the least significant 60 bits of the SHA-1 hash of the value of the BIT STRING subjectPublicKey (excluding the tag, length, and number of unused bits)." The keyIdentifier generated by the second method is likely to be shorter than 20 bytes. When I use wolfSSL to parse a keyIdentifier with fewer than 20 bytes (for example, keyIdentifier=00:10:64:71:83:35), wolfSSL still parses it as a 20-byte length. Here is the test case:

[crl_less_keyid.zip](https://github.com/user-attachments/files/19587564/crl_less_keyid.zip)

onepeople158, 2025-04-25:
> Hi [@onepeople158](https://github.com/onepeople158),
> 
> > The keyIdentifier is composed of the 160-bit SHA-1 hash of the value of the BIT STRING subjectPublicKey (excluding the tag, length, and number of unused bits)
> 
> As you mention above, the keyIdentifier will be 160 bits long (20 bytes). The value that you are expecting (), is 28 bytes long and isn't compliant with the RFC.`43:4C:73:65:4E:62:44:33:69:4D:73:43:59:31:46:43:4D:45:30:46:2F:4E:35:61:65:61:63:0A`

Hi Developer, I've noticed that when wolfSSL parses a keyIdentifier field that isn't 20 bytes long, it seems to parse it into a 20-byte keyIdentifier value. Is this correct behavior? Could this lead to misinterpretation of the original keyIdentifier's content?

lealem47, 2025-04-25:
Hi @onepeople158 ,

Thank you for the report. I'll make sure to review this next week.

onepeople158, 2025-04-27:
> Hi [@onepeople158](https://github.com/onepeople158) ,
> 
> Thank you for the report. I'll make sure to review this next week.

Thanks.

lealem47, 2025-05-08:
Hi @onepeople158,

We should support this, but it looks like it will be a significant refactor. I'll spend time on it during spare cycles but I don't expect to have many of any time soon. I'll keep you posted.

Thanks