wolfSSL/wolfssl #8627: [Bug]: DTLS 1.3 Client doesn't reject Application message encrypted with handshake_traffic_secret

c-southwest, 2025-04-02:
### Contact Details

GitHub should be enough

### Version

wolfssl-5.7.6

### Description

## Summary
This issue is similar to #8579 , the difference is that the problem "Application with epoch=2" also exists in wolfssl **Client** implementation. However, wolfssl **Client** doesn't have the problem "Finished with epoch=3".

[image]

## Other Information
### How I build wolfSSL:
```bash
bash autogen.sh
AM_CFLAGS='-DHAVE_AES_CBC -DWOLFSSL_AES_128 -DWOLFSSL_DEBUG_TLS' ./configure --enable-dtls --enable-dtls13 --enable-keylog-export --enable-psk --enable-rsa --enable-sha --enable-debug C_EXTRA_FLAGS=-DWOLFSSL_STATIC_PSK
```
### How I launch wolfSSL server:
```bash
# PSK authentication
/home/west/dtls-fuzzer/suts/wolfssl-5.7.6/examples/client/client  -i -x -p 28092 -s -u -v 4 -l TLS13-AES128-GCM-SHA256
```

### Reproduction steps

Before sending Application message, I deliberately set write epoch to 2, so as a Server I can send Application with epoch=2.

You can find Wireshark capture, sslkeylog.log, and also wolfssl client debug logs.
[captures_and_logs.zip](https://github.com/user-attachments/files/19573957/captures_and_logs.zip)

### Relevant log output

```shell
wolfSSL Leaving wolfSSL_dtls_get_current_timeout, return 320
wolfSSL Entering EmbedReceiveFrom
wolfSSL Entering wolfSSL_dtls_get_using_nonblock
Data received
	2e 39 1e 00 1f 68 80 24 1b bd 51 00 f4 fb 2f 61 |.9...h.$..Q.../a
	9b 71 ee c4 88 07 82 da 07 79 8e ef c9 4d bb 5f |.q.......y...M._
	61 31 66 50                                     |a1fP
reconstructed epoch number: 2
wolfSSL Entering Dtls13SetEpochKeys
Provisioning DECRYPT key
	bf 08 cb 18 02 6c c4 fb 62 8a 17 77 ee f8 5a 71 |.....l..b..w..Zq
Provisioning AES Record Number dec key:
	e3 13 7a ae 10 93 3d d1 1a a3 f7 d3 ed f3 89 7d |..z...=........}
reconstructed seq number: 2
wolfSSL Entering DecryptTls13
Data to decrypt
	68 80 24 1b bd 51 00 f4 fb 2f 61 9b 71 ee c4    |h.$..Q.../a.q..
Additional Authentication Data
	2e 00 02 00 1f                                  |.....
Authentication tag
	88 07 82 da 07 79 8e ef c9 4d bb 5f 61 31 66 50 |.....y...M._a1fP
Dtls13GetSeq(): using seq: 2
TLS 1.3 Write Sequence 0 2
Nonce
	92 14 25 3c ff bb 3b 04 95 dc f8 2d             |..%<..;....-
Decrypted data
	68 65 6c 6c 6f 20 77 6f 6c 66 73 73 6c 21 17    |hello wolfssl!.
wolfSSL Entering Dtls13CheckWindow
wolfSSL Entering Dtls13UpdateWindow
received record layer msg
got app DATA
wolfSSL Entering Dtls13DoScheduledWork
Shrinking input buffer
wolfSSL Leaving ReceiveData(), return 14
wolfSSL Leaving wolfSSL_read_internal, return 14
```

c-southwest, 2025-04-02:
Hi @julek-wolfssl ,

Since you are working on the similar issue #8579 , so I think you can also consider this issue, I believe they can be fixed in a similar way.

julek-wolfssl, 2025-04-02:
Thanks. I'll take a look.

julek-wolfssl, 2025-04-09:
@c-southwest reading through this I think https://github.com/wolfSSL/wolfssl/pull/8631 should fix it as well. It has additional checks for application data epochs.

c-southwest, 2025-04-10:
@julek-wolfssl 
I tested all the bug cases, and all bugs no longer exist, so I think your fix is correct. Below are my test results if you want to check.

### Testing DTLS 1.3 Server:
I send Application with epoch=2, and wolfSSL server send alert

[image]

### Test DTLS 1.3 Client:
I send Application with epoch=2, and wolfssl client terminate without sending anything
```
wolfSSL Leaving wolfSSL_dtls_get_current_timeout, return 1
wolfSSL Entering EmbedReceiveFrom
wolfSSL Entering wolfSSL_dtls_get_using_nonblock
Data received
	2e 75 0e 00 1f af 20 8c e1 e4 52 d9 33 4c 91 db |.u.... ...R.3L..
	bf 4f a4 01 01 27 ea a8 d9 7f 08 c4 56 f0 8a ba |.O...'......V...
	a7 ee 4d 1a                                     |..M.
reconstructed epoch number: 2
reconstructed seq number: 3
wolfSSL Entering DecryptTls13
Data to decrypt
	af 20 8c e1 e4 52 d9 33 4c 91 db bf 4f a4 01    |. ...R.3L...O..
Additional Authentication Data
	2e 00 03 00 1f                                  |.....
Authentication tag
	01 27 ea a8 d9 7f 08 c4 56 f0 8a ba a7 ee 4d 1a |.'......V.....M.
Dtls13GetSeq(): using seq: 3
TLS 1.3 Write Sequence 0 3
Nonce
	36 3e 19 77 92 87 09 7f 2f 24 e7 5c             |6>.w..../$.\
Decrypted data
	68 65 6c 6c 6f 20 77 6f 6c 66 73 73 6c 21 17    |hello wolfssl!.
wolfSSL Entering Dtls13CheckWindow
wolfSSL Entering Dtls13UpdateWindow
received record layer msg
got app DATA
wolfSSL error occurred, error = -394
wolfSSL error occurred, error = -394
wolfSSL Leaving wolfSSL_read_internal, return -394
wolfSSL Entering wolfSSL_get_error
wolfSSL Leaving wolfSSL_get_error, return -394
wolfSSL Entering wolfSSL_ERR_error_string
SSL_read reply error -394, Sanity Check on message order Error
wolfSSL error: SSL_read failed
```

[image]

### Testing DTLS 1.3 Server:
I send Finished with epoch=3, and wolfSSL client cannot decrypt

```
wolfSSL Leaving wolfSSL_dtls_get_current_timeout, return 1
wolfSSL Entering EmbedReceiveFrom
wolfSSL Entering wolfSSL_dtls_get_using_nonblock
Data received
	2f d9 04 00 3d 4f 2b 16 59 b6 ef d7 d9 a3 38 67 |/...=O+.Y.....8g
	78 02 ca 63 7e a4 70 20 64 76 08 53 7a ac bf 54 |x..c~.p dv.Sz..T
	9d fe 5c 9a e3 6a 7c 22 3c c4 41 53 5e b5 13 64 |..\..j|"<.AS^..d
	b9 cf 7c 0e 48 24 ad 18 d2 d2 81 08 81 10 ec 24 |..|.H$.........$
	86 87                                           |..
reconstructed epoch number: 3
reconstructed seq number: 56765
wolfSSL Entering DecryptTls13
Data to decrypt
	4f 2b 16 59 b6 ef d7 d9 a3 38 67 78 02 ca 63 7e |O+.Y.....8gx..c~
	a4 70 20 64 76 08 53 7a ac bf 54 9d fe 5c 9a e3 |.p dv.Sz..T..\..
	6a 7c 22 3c c4 41 53 5e b5 13 64 b9 cf          |j|"<.AS^..d..
Additional Authentication Data
	2f dd bd 00 3d                                  |/...=
Authentication tag
	7c 0e 48 24 ad 18 d2 d2 81 08 81 10 ec 24 86 87 ||.H$.........$..
Dtls13GetSeq(): using seq: 56765
TLS 1.3 Write Sequence 0 56765
Nonce
	20 de f2 96 95 66 89 e9 c8 0e bf 8d             | ....f......
Decrypted data
	0b 19 4b 52 0b 17 07 15 c7 a1 9b 76 e4 df 56 48 |..KR.......v..VH
	d4 27 fa 61 c1 49 b4 cb 0c 2b f9 3e 2f 14 a1 ac |.'.a.I...+.>/...
	1c 28 cb 46 9e 9a e7 01 58 2a 4b b6 03          |.(.F....X*K..
Decrypt failed
DTLS: Ignoring failed decryption
wolfSSL Entering Dtls13DoScheduledWork
wolfSSL Entering RetrySendAlert
wolfSSL Leaving wolfSSL_dtls_get_current_timeout, return 1
wolfSSL Entering EmbedReceiveFrom
wolfSSL Entering wolfSSL_dtls_get_using_nonblock
	Would block
Embed Receive From error
wolfSSL Entering wolfSSL_dtls_get_using_nonblock
wolfSSL Leaving wolfSSL_dtls_get_current_timeout, return 1
wolfSSL Entering EmbedReceiveFrom
wolfSSL Entering wolfSSL_dtls_get_using_nonblock
```

[image]

julek-wolfssl, 2025-04-14:
@c-southwest in your logs I see `wolfSSL Entering wolfSSL_shutdown` from the wolfSSL server. It appears that our server is still calling shutdown on an error. This alert doesn't come from our internal `SendAlert` call directly but from a call to `wolfSSL_shutdown`. I think it is alright if the user wants to send a closing alert anyway.

c-southwest, 2025-04-14:
@julek-wolfssl OK, now I understand the Alert comes from a shutdown action instead of error handling. I think you are right.

closed 2025-04-18 by douzzer