wolfSSL/wolfssl #8666: [Bug]: Record numbers in DTLS ACK are not ordered

SpecLad, 2025-04-13:
### Version

5.7.6

### Description

According to RFC 9147, section 7, the `record_numbers` field in an ACK message is:

> A list of the records containing handshake messages in the current flight which the endpoint has received and either processed or buffered, **in numerically increasing order**.

(emphasis mine)

However, I checked the ACK message produced by wolfSSL, and the record numbers there are not numerically increasing. In fact, they appear to be numerically _decreasing_.

My OS is Windows 11; the only custom flags I added are `SHOW_SECRETS`, `WOLFSSL_SSLKEYLOGFILE` and `HAVE_SECRET_CALLBACK`.

### Reproduction steps

1. Build the `client` and `server` examples.
2. Run `.\examples\server\Debug\x64\server.exe -v 4 -u`.
3. Run `.\examples\client\Debug\x64\client.exe -v 4 -u` while capturing traffic.
4. Find the ACK message in the captured traffic and look at the order of the record numbers.

Here's an archive with an example capture and the key log file necessary to decrypt it: [seqnum-repro.zip](https://github.com/user-attachments/files/19726617/seqnum-repro.zip). The ACK message in this capture looks like this:

```
Acknowledgement Message
    Record Number Length: 48
    RecordNumbers (3 record numbers)
        RecordNumber: epoch 2, sequence number 2
        RecordNumber: epoch 2, sequence number 1
        RecordNumber: epoch 2, sequence number 0
```

### Relevant log output

```shell

```

embhorn, 2025-04-14:
Hi @SpecLad 

Thanks for contacting wolfSSL Support. I have requested a review of this issue.

Could you tell us a bit about your project using wolfSSL? Feel free to email us at support@wolfssl.com for a more private discussion.

Thanks,
@embhorn - wolfSSL Support

julek-wolfssl, 2025-04-14:
Hi @SpecLad, it appears that we prepend the record numbers when receiving them. I'll look into how we can arrange them in order to follow the spec more closely.

Sincerely
Juliusz

SpecLad, 2025-04-14:
@embhorn I was just using wolfSSL to generate DTLS packet traces to test a patch I was going to submit to Wireshark. I needed to have a DTLS 1.3 trace, as well as 1.2 with the CID extension, and wolfSSL happens to support both.

julek-wolfssl, 2025-04-17:
Hi @SpecLad please find a fix at https://github.com/wolfSSL/wolfssl/pull/8689.

Sincerely
Juliusz

closed 2025-04-18 by JacobBarthelmeh