wolfSSL/wolfssl #8676: [Bug]: WolfSSL parsed a CRL file with a negative CRL number value.

onepeople158, 2025-04-16:
### Version

I am using version 0.1.7 of the wolfssl Command Line Utility. Linked to wolfSSL version 5.7.6

### Description

Hello Developer, I have a CRL file with a CRL Number value of -36. According to RFC5280, the CRL Number value should be a non-negative integer, but wolfSSL successfully printed this CRL file and displayed the CRL Number value as 220.

### Reproduction steps

[crl_file_test_.zip](https://github.com/user-attachments/files/19773181/crl_file_test_.zip)

### Relevant log output

```shell
wolfSSL Entering wolfSSL_Init
wolfSSL Entering wolfCrypt_Init
RNG_HEALTH_TEST_CHECK_SIZE = 128
sizeof(seedB_data)         = 128
opened /dev/urandom.
rnd read...
wolfSSL Entering wolfSSL_BIO_new_file
wolfSSL Entering wolfSSL_BIO_s_file
wolfSSL Entering wolfSSL_BIO_new
wolfSSL Entering wolfSSL_BIO_set_fp
wolfSSL Entering wolfSSL_BIO_get_len
wolfSSL Entering wolfSSL_BIO_get_fp
wolfSSL Entering wolfSSL_BIO_read
wolfSSL Entering wolfSSL_d2i_X509_CRL
wolfSSL Entering InitCRL
wolfSSL Entering BufferLoadCRL
InitDecodedCRL
ParseCRL
About to verify CRL signature
Did NOT find CRL issuer CA
ERR TRACE: wolfcrypt/src/asn.c L 38604 ASN_CRL_NO_SIGNER_E (-190)
wolfSSL Entering AddCRL
wolfSSL Entering InitCRL_Entry
wolfSSL Entering wolfSSL_d2i_X509_NAME
Getting Name
Getting Cert Name
wolfSSL Entering wolfSSL_X509_NAME_new_ex
wolfSSL Entering wolfSSL_X509_NAME_add_entry_by_NID
Found place for name entry
wolfSSL Entering wolfSSL_sk_X509_NAME_new
wolfSSL Entering wolfSSL_sk_new_node
wolfSSL Entering wolfSSL_sk_push
wolfSSL Entering wolfSSL_sk_insert
wolfSSL Entering wolfSSL_X509_NAME_ENTRY_free
wolfSSL Entering wolfSSL_X509_NAME_add_entry_by_NID
Found place for name entry
wolfSSL Entering wolfSSL_sk_push
wolfSSL Entering wolfSSL_sk_insert
wolfSSL Entering wolfSSL_sk_new_node
wolfSSL Entering wolfSSL_X509_NAME_ENTRY_free
wolfSSL Entering wolfSSL_X509_NAME_add_entry_by_NID
Found place for name entry
wolfSSL Entering wolfSSL_sk_push
wolfSSL Entering wolfSSL_sk_insert
wolfSSL Entering wolfSSL_sk_new_node
wolfSSL Entering wolfSSL_X509_NAME_ENTRY_free
wolfSSL Entering wolfSSL_X509_NAME_add_entry_by_NID
Found place for name entry
wolfSSL Entering wolfSSL_sk_push
wolfSSL Entering wolfSSL_sk_insert
wolfSSL Entering wolfSSL_sk_new_node
wolfSSL Entering wolfSSL_X509_NAME_ENTRY_free
wolfSSL Entering wolfSSL_X509_NAME_add_entry_by_NID
Found place for name entry
wolfSSL Entering wolfSSL_sk_push
wolfSSL Entering wolfSSL_sk_insert
wolfSSL Entering wolfSSL_sk_new_node
wolfSSL Entering wolfSSL_X509_NAME_ENTRY_free
wolfSSL Entering wolfSSL_X509_NAME_add_entry_by_NID
Found place for name entry
wolfSSL Entering wolfSSL_sk_push
wolfSSL Entering wolfSSL_sk_insert
wolfSSL Entering wolfSSL_sk_new_node
wolfSSL Entering wolfSSL_X509_NAME_ENTRY_free
wolfSSL Entering wolfSSL_X509_NAME_new_ex
wolfSSL Entering wolfSSL_X509_NAME_copy
wolfSSL Entering wolfSSL_sk_X509_NAME_new
wolfSSL Entering wolfSSL_sk_new_node
wolfSSL Entering wolfSSL_sk_push
wolfSSL Entering wolfSSL_sk_insert
wolfSSL Entering wolfSSL_sk_push
wolfSSL Entering wolfSSL_sk_insert
wolfSSL Entering wolfSSL_sk_new_node
wolfSSL Entering wolfSSL_sk_push
wolfSSL Entering wolfSSL_sk_insert
wolfSSL Entering wolfSSL_sk_new_node
wolfSSL Entering wolfSSL_sk_push
wolfSSL Entering wolfSSL_sk_insert
wolfSSL Entering wolfSSL_sk_new_node
wolfSSL Entering wolfSSL_sk_push
wolfSSL Entering wolfSSL_sk_insert
wolfSSL Entering wolfSSL_sk_new_node
wolfSSL Entering wolfSSL_sk_push
wolfSSL Entering wolfSSL_sk_insert
wolfSSL Entering wolfSSL_sk_new_node
wolfSSL Entering wolfSSL_X509_NAME_free
wolfSSL Entering wolfSSL_sk_free
FreeDecodedCRL
wolfSSL Entering wolfSSL_BIO_s_file
wolfSSL Entering wolfSSL_BIO_new
wolfSSL Entering wolfSSL_BIO_set_fp
wolfSSL Entering wolfSSL_BIO_write
Certificate Revocation List (CRL):
wolfSSL Entering wolfSSL_BIO_write
wolfSSL Entering wolfSSL_BIO_write
        Version: 2 (0x1)
wolfSSL Entering wolfSSL_X509_CRL_get_signature
wolfSSL Entering wolfSSL_X509_CRL_get_signature
wolfSSL Entering wolfSSL_BIO_write
wolfSSL Entering wolfSSL_OBJ_obj2txt
wolfSSL Entering wolfSSL_OBJ_nid2ln
wolfSSL Entering wolfSSL_BIO_write
        Signature Algorithm: sha256WithRSAEncryption
wolfSSL Entering wolfSSL_BIO_write
wolfSSL Entering wolfSSL_X509_NAME_print_ex
wolfSSL Entering wolfSSL_BIO_write
wolfSSL Entering wolfSSL_X509_NAME_entry_count
wolfSSL Leaving wolfSSL_X509_NAME_entry_count, return 6
wolfSSL Entering wolfSSL_X509_NAME_ENTRY_get_data
wolfSSL Entering wolfSSL_BIO_write
wolfSSL Entering wolfSSL_X509_NAME_ENTRY_get_data
wolfSSL Entering wolfSSL_BIO_write
wolfSSL Entering wolfSSL_X509_NAME_ENTRY_get_data
wolfSSL Entering wolfSSL_BIO_write
wolfSSL Entering wolfSSL_X509_NAME_ENTRY_get_data
wolfSSL Entering wolfSSL_BIO_write
wolfSSL Entering wolfSSL_X509_NAME_ENTRY_get_data
wolfSSL Entering wolfSSL_BIO_write
wolfSSL Entering wolfSSL_X509_NAME_ENTRY_get_data
wolfSSL Entering wolfSSL_BIO_write
wolfSSL Entering wolfSSL_BIO_write
        Issuer:  C=US, ST=US, L=US, O=US, CN=US, OU=US
wolfSSL Entering wolfSSL_BIO_write
wolfSSL Entering wolfSSL_BIO_write
wolfSSL Entering wolfSSL_BIO_write
        Last Update: Jan  1 00:00:00 2025 GMT
wolfSSL Entering wolfSSL_BIO_write
wolfSSL Entering wolfSSL_BIO_write
wolfSSL Entering wolfSSL_BIO_write
        Next Update: Dec  1 00:00:00 2025 GMT
wolfSSL Entering wolfSSL_BIO_write
        CRL extensions:
wolfSSL Entering wolfSSL_BIO_write
            X509v3 CRL Number:
wolfSSL Entering wolfSSL_BIO_write
                220
wolfSSL Entering wolfSSL_BIO_write
Revoked Certificates:
wolfSSL Entering wolfSSL_X509_REVOKED_get_serial_number
wolfSSL Entering wolfSSL_BIO_write
wolfSSL Entering wolfSSL_BIO_write
    Serial Number:
        1c80022ef81f2405ee96a612dcb61fe0ac701e5e
wolfSSL Entering wolfSSL_BIO_write
wolfSSL Entering wolfSSL_BIO_write
wolfSSL Entering wolfSSL_BIO_write
        Revocation Date: Mar 13 02:44:40 2025 GMT
wolfSSL Entering wolfSSL_X509_CRL_get_signature
wolfSSL Entering wolfSSL_X509_CRL_get_signature
wolfSSL Entering wolfSSL_BIO_write
wolfSSL Entering wolfSSL_OBJ_obj2txt
wolfSSL Entering wolfSSL_OBJ_nid2ln
wolfSSL Entering wolfSSL_BIO_write
    Signature Algorithm: sha256WithRSAEncryption
wolfSSL Entering wolfSSL_BIO_write
wolfSSL Entering wolfSSL_BIO_write
         68:36:3d:8c:17:40:26:20:2e:8d:49:1a:2d:82:d5:b7:33:56:
wolfSSL Entering wolfSSL_BIO_write
         17:fd:12:c4:3e:42:07:87:58:21:c6:4c:aa:d3:ca:2e:7e:72:
wolfSSL Entering wolfSSL_BIO_write
         91:cc:64:5e:f9:d1:6d:58:a1:27:e2:a7:00:0b:fd:16:49:f9:
wolfSSL Entering wolfSSL_BIO_write
         8b:08:fb:ec:41:b0:c4:d8:f2:66:4b:50:e2:00:26:70:c8:42:
wolfSSL Entering wolfSSL_BIO_write
         4c:11:1c:00:76:e6:8b:dd:ad:1e:db:68:b7:d4:ab:e3:8f:82:
wolfSSL Entering wolfSSL_BIO_write
         37:ed:0d:69:a4:03:39:f9:48:79:5c:3b:66:2a:fd:d1:35:ae:
wolfSSL Entering wolfSSL_BIO_write
         7e:34:9c:cb:cf:de:ec:59:15:9b:e6:83:e4:28:9c:ad:b2:56:
wolfSSL Entering wolfSSL_BIO_write
         aa:87:b6:d6:90:75:43:58:d2:e3:d8:8d:ad:9c:ea:67:6b:f0:
wolfSSL Entering wolfSSL_BIO_write
         1c:b8:aa:34:0c:e9:79:cc:70:52:28:7f:60:f7:b9:f8:20:64:
wolfSSL Entering wolfSSL_BIO_write
         e4:da:b8:bc:80:9e:89:e1:95:0b:f2:4e:f6:be:52:91:d0:f0:
wolfSSL Entering wolfSSL_BIO_write
         59:04:e5:d0:8d:e4:48:ae:a7:e0:98:7b:e7:71:66:21:e9:fc:
wolfSSL Entering wolfSSL_BIO_write
         d0:5d:99:66:e6:6e:e3:f7:e1:27:b0:b8:ae:5a:fa:5b:d8:ba:
wolfSSL Entering wolfSSL_BIO_write
         16:b2:b4:ea:ce:66:93:53:de:60:51:ca:84:29:30:23:cc:29:
wolfSSL Entering wolfSSL_BIO_write
         f1:c2:2e:74:94:03:94:bb:0a:da:ee:02:4d:cb:93:29:d6:c3:
wolfSSL Entering wolfSSL_BIO_write
         2e:cb:33:60
wolfSSL Entering wolfSSL_BIO_write
-----BEGIN X509 CRL-----
MIIB3jCBxwIBATANBgkqhkiG9w0BAQsFADBOMQswCQYDVQQGEwJVUzELMAkGA1UE
CAwCVVMxCzAJBgNVBAcMAlVTMQswCQYDVQQKDAJVUzELMAkGA1UEAwwCVVMxCzAJ
BgNVBAsMAlVTFw0yNTAxMDEwMDAwMDBaFw0yNTEyMDEwMDAwMDBaMDUwMwIUHIAC
LvgfJAXulqYS3LYf4KxwHl4XDTI1MDMxMzAyNDQ0MFowDDAKBgNVHRUEAwoBBqAO
MAwwCgYDVR0UBAMCAdwwDQYJKoZIhvcNAQELBQADggEBAGg2PYwXQCYgLo1JGi2C
1bczVhf9EsQ+QgeHWCHGTKrTyi5+cpHMZF750W1YoSfipwAL/RZJ+YsI++xBsMTY
8mZLUOIAJnDIQkwRHAB25ovdrR7baLfUq+OPgjftDWmkAzn5SHlcO2Yq/dE1rn40
nMvP3uxZFZvmg+QonK2yVqqHttaQdUNY0uPYja2c6mdr8By4qjQM6XnMcFIof2D3
ufggZOTauLyAnonhlQvyTva+UpHQ8FkE5dCN5Eiup+CYe+dxZiHp/NBdmWbmbuP3
4SewuK5a+lvYuhaytOrOZpNT3mBRyoQpMCPMKfHCLnSUA5S7CtruAk3LkynWwy7L
M2A=
-----END X509 CRL-----
wolfSSL Entering wolfSSL_X509_CRL_free
wolfSSL Entering FreeCRL
wolfSSL Entering FreeCRL_Entry
wolfSSL Entering wolfSSL_sk_free
wolfSSL Entering wolfSSL_BIO_free
wolfSSL Entering wolfSSL_BIO_free
wolfSSL Entering wolfSSL_Cleanup
wolfSSL Entering wolfCrypt_Cleanup
```

lealem47, 2025-05-01:
Hi @onepeople158,

Note that RFC5280 states the following:

```
Note: Non-conforming CAs may issue certificates with serial numbers
   that are negative or zero.  Certificate users SHOULD be prepared to
   gracefully handle such certificates.
```

I've added a commit to PR https://github.com/wolfSSL/wolfssl/pull/8587 to ensure the serial number gets printed as a signed int. If you test it, you should see the serial number printed as `-36` instead of `220`

onepeople158, 2025-05-02:
> Hi [@onepeople158](https://github.com/onepeople158),
>
> Note that RFC5280 states the following:
>
> ```
> Note: Non-conforming CAs may issue certificates with serial numbers
>    that are negative or zero. [...]

Thanks.

closed 2025-06-17 by lealem47