wolfSSL/wolfssl #8704: [Bug]: Extension SEQUENCE SIZE(1..MAX) and Extension value of zero bytes

dulanshuangqiao, 2025-04-23:
### Contact Details

bingxiaoganel@gmail.com

### Version

ubutun 5.7.6

### Description

./configure
make
sudo make install
./testsuite/testsuite.test
wolfSSL is configured and built by default

### Reproduction steps

```
// gcc -g verify.c -o verify -lwolfssl
#include <stdlib.h>
#include <wolfssl/ssl.h>
#include <wolfssl/options.h>
#include <wolfssl/wolfcrypt/error-crypt.h>

int wolfSSL_Verify_PEM(char * cac, char * ec){
int ret = 0;

// to create a new wolfSSL cert manager
WOLFSSL_CERT_MANAGER* cm;
cm = wolfSSL_CertManagerNew();
if (cm == NULL){
	printf("Creating a new wolfSSL_CertManager failed!\n");
	exit(1);
}

// to load cac to the created wolfSSL_CertManager
ret = wolfSSL_CertManagerLoadCA(cm, cac, NULL);
if (ret != SSL_SUCCESS){
	printf("Loading cac to the created wolfSSL_CertManager failed!\n");
	exit(2);
}

// to verify the ec in the created wolfSSL_CertManager
ret = wolfSSL_CertManagerVerify(cm, ec, SSL_FILETYPE_PEM);
if (ret != SSL_SUCCESS){
	printf("wolfSSL_CertManagerVerify filed and with return code %d and error message %s\n",
		 ret,
		 wolfSSL_ERR_reason_error_string(ret));
}
else{
	printf("The target cert has passed through verification.\n");
}

// to free cm
wolfSSL_CertManagerFree(cm);

return ret;
}

int main(int argc, char ** argv){
char * cac = argv[1];
char * ec = argv[2];
wolfSSL_Verify_PEM(cac, ec);
return 0;
}
```
I used the certificate validation script provided above for testing.

In issue # https://github.com/wolfSSL/wolfssl/issues/8564 (@ColtonWilley ), I received feedback regarding the validity of the encoding in question, described as follows: Although OpenSSL does validate these certificates, I have determined that this encoding is invalid under RFC 5280. According to Section 4.2.2.1, the encoding for Authority Information Access is defined as:
AuthorityInfoAccessSyntax ::=  
SEQUENCE SIZE (1..MAX) OF AccessDescription

As you can see, it is specified as a sequence of size (1..MAX), so a zero-byte extension value is not a valid encoding. OpenSSL appears to be more permissive in allowing this, but we will not extend our code to support invalid encodings.

Based on this feedback, I performed related tests. For the PolicyMappings extension with a null value, wolfSSL should reject it, because RFC 5280 specifies that the PolicyMappings extension must also be a SEQUENCE OF size (1..MAX). As described above, a zero-byte extension value is not valid encoding. However, wolfSSL incorrectly accepts it during validation.

### Relevant log output

```shell

```

dulanshuangqiao, 2025-05-24:
There is no staff to handle this report for a long time.I hope a developer can review my report，This is very important for my work.

ColtonWilley, 2025-06-02:
Hello @dulanshuangqiao 

Unfortunately myself and the team has been busy with more critical work. This is still on my list of things to fix, and will definitely be addressed when I have the time to do so. If you want to propose specific code fixes to address the problem I would be happy to review them.