wolfSSL/wolfssl #8731: [Bug]: 5.8.0 fails to build with `-Werror`: `benchmark.c:8633: warning: declaration of ‘index’ shadows a global declaration`

barracuda156, 2025-05-02:
### Version

5.8.0

### Description

5.8.0 errors out when being built with `-Werror`.

### Reproduction steps

Fetch 5.8.0 from release commit, run the build.

### Relevant log output

```shell
depbase=`echo wolfcrypt/benchmark/benchmark.o | sed 's|[^/]*$|.deps/&|;s|\.o$||'`;\
    /usr/bin/gcc-4.2 -std=gnu99 -DHAVE_CONFIG_H -I.  -DWOLFSSL_HAVE_ASSERT_H -DWOLFSSL_WOLFSSH   -isystem/opt/local/include/LegacySupport -I/opt/local/include -fvisibility=hidden -DNDEBUG -DNO_DO178 -DWOLFSSL_ASN_TEMPLATE -DHAVE_CRL_IO -DHAVE_IO_TIMEOUT -DWOLFSSL_DER_LOAD -DKEEP_OUR_CERT -DKEEP_PEER_CERT -DWOLFSSL_SUBJ_DIR_ATTR -DWOLFSSL_FPKI -DWOLFSSL_SUBJ_INFO_ACC -DWOLFSSL_CERT_NAME_ALL -DWOLFSSL_VERBOSE_ERRORS -DHAVE_ECC_CDH -DHAVE_ECC_KOBLITZ -DHAVE_ECC_SECPR2 -DHAVE_ECC_SECPR3 -DWOLFSSL_DES_ECB -DHAVE_AES_DECRYPT -DHAVE_AES_ECB -DWOLFSSL_ALT_NAMES -DHAVE_FFDHE_2048 -DHAVE_FFDHE_3072 -DWOLFSSL_ASN_ALL -DWOLFSSL_DH_EXTRA -DWOLFSSL_ECDSA_DETERMINISTIC_K_VARIANT -DWOLFSSL_HAVE_ISSUER_NAMES -pthread -DHAVE_ECH -DWOLFSSL_DTLS -DWOLFSSL_DTLS_MTU -DWOLFSSL_QUIC -DHAVE_EX_DATA -DWOLFSSL_POST_HANDSHAKE_AUTH -DWOLFSSL_SEND_HRR_COOKIE -DWOLFSSL_LIBWEBSOCKETS -DHAVE_EX_DATA -DOPENSSL_NO_EC -DWOLFSSL_OPENSSH -DHAVE_EX_DATA -DWOLFSSL_BASE16 -DWOLFSSL_ERROR_CODE_OPENSSL -DTFM_TIMING_RESISTANT -DECC_TIMING_RESISTANT -DWC_RSA_BLINDING -DFORTRESS -DWOLFSSL_ALWAYS_VERIFY_CB -DWOLFSSL_AES_COUNTER -DWOLFSSL_AES_DIRECT -DWOLFSSL_DER_LOAD -DWOLFSSL_KEY_GEN -DPERSIST_SESSION_CACHE -DPERSIST_CERT_CACHE -DATOMIC_USER -DHAVE_PK_CALLBACKS -DWOLFSSL_AES_CBC_LENGTH_CHECKS -DHAVE_AESCCM -DWOLFSSL_AES_EAX -DWOLFSSL_AES_OFB -DWOLFSSL_AES_DIRECT -DWOLFSSL_AES_CFB -DWOLFSSL_ARMASM_NO_HW_CRYPTO -DWOLFSSL_USE_ALIGN -DHAVE_CAMELLIA -DWOLFSSL_MD2 -DHAVE_NULL_CIPHER -DWOLFSSL_RIPEMD -DHAVE_BLAKE2 -DHAVE_BLAKE2B -DHAVE_BLAKE2S -DWOLFSSL_SHA224 -DWOLFSSL_SHA512 -DWOLFSSL_SHA384 -DSESSION_CERTS -DWOLFSSL_SEP -DKEEP_PEER_CERT -DHAVE_HKDF -DHAVE_HPKE -DHAVE_X963_KDF -DHAVE_ECC -DECC_SHAMIR -DECC_MIN_KEY_SZ=224 -DHAVE_ECC_BRAINPOOL -DHAVE_CURVE25519 -DFP_ECC -DHAVE_ECC_ENCRYPT -DWOLFCRYPT_HAVE_ECCSI -DWOLFSSL_PUBLIC_MP -DWOLFCRYPT_HAVE_SAKKE -DNO_OLD_TLS -DWOLFSSL_QT -DSESSION_CERTS -DOPENSSL_NO_SSL2 -DWOLFSSL_KEY_GEN -DHAVE_EX_DATA -DWOLFSSL_CUSTOM_CURVES -DHAVE_ECC_SECPR2 -DHAVE_ECC_SECPR3 -DHAVE_ECC_BRAINPOOL -DHAVE_ECC_KOBLITZ -DWC_RSA_PSS -DWOLFSSL_PSS_LONG_SALT -DHAVE_ANON -DWOLFSSL_ASN_PRINT -DWOLFSSL_DTLS13 -DWOLFSSL_W64_WRAPPER -DWOLFSSL_AES_DIRECT -DWOLFSSL_DTLS_CID -DWOLFSSL_DTLS_CH_FRAG -DWOLFSSL_BASE64_ENCODE -DWOLFSSL_BASE16 -DWOLFSSL_SIPHASH -DWOLFSSL_CMAC -DWOLFSSL_AES_DIRECT -DHAVE_WEBSERVER -DWOLFSSL_AES_XTS -DWOLFSSL_AES_DIRECT -DWOLFSSL_CUSTOM_CURVES -DHAVE_ECC_SECPR2 -DHAVE_ECC_SECPR3 -DHAVE_ECC_BRAINPOOL -DHAVE_ECC_KOBLITZ -DHAVE_ECC_CDH -DHAVE_CURVE448 -DHAVE_ED448 -DWOLFSSL_ED448_STREAMING_VERIFY -DWC_SRTP_KDF -DHAVE_AES_ECB -DWOLFSSL_AES_DIRECT -DWOLFSSL_SHA3 -DWOLFSSL_SHAKE128 -DWOLFSSL_SHAKE256 -DHAVE_POLY1305 -DHAVE_CHACHA -DHAVE_XCHACHA -DHAVE_HASHDRBG -DHAVE_OPENSSL_CMD -DHAVE_TLS_EXTENSIONS -DHAVE_CERTIFICATE_STATUS_REQUEST -DHAVE_TLS_EXTENSIONS -DHAVE_CERTIFICATE_STATUS_REQUEST_V2 -DHAVE_CRL -DHAVE_CRL_MONITOR -DHAVE_TLS_EXTENSIONS -DHAVE_SNI -DHAVE_TLS_EXTENSIONS -DHAVE_ALPN -DHAVE_TLS_EXTENSIONS -DHAVE_MAX_FRAGMENT -DHAVE_TLS_EXTENSIONS -DHAVE_TRUSTED_CA -DHAVE_TLS_EXTENSIONS -DHAVE_TRUNCATED_HMAC -DHAVE_FALLBACK_SCSV -DHAVE_KEYING_MATERIAL -DHAVE_TLS_EXTENSIONS -DHAVE_SUPPORTED_CURVES -DHAVE_FFDHE_2048 -DHAVE_SUPPORTED_CURVES -DWOLFSSL_TLS13 -DHAVE_TLS_EXTENSIONS -DHAVE_TLS_EXTENSIONS -DHAVE_SESSION_TICKET -DHAVE_EXTENDED_MASTER -DHAVE_TLS_EXTENSIONS -DHAVE_SNI -DHAVE_MAX_FRAGMENT -DHAVE_TRUNCATED_HMAC -DHAVE_ALPN -DHAVE_TRUSTED_CA -DHAVE_SUPPORTED_CURVES -DWOLFSSL_EARLY_DATA -DHAVE_SMIME -DWOLFCRYPT_HAVE_SRP -DASN_BER_TO_DER -DWOLFSSL_HAVE_CERT_SERVICE -DHAVE_LIGHTY -DHAVE_WOLFSSL_SSL_H=1 -DHAVE_EX_DATA -DOPENSSL_ALL -DWOLFSSL_KEY_GEN -DWOLFSSL_NGINX -DWOLFSSL_SIGNER_DER_CERT -DOPENSSL_COMPATIBLE_DEFAULTS -DWOLFSSL_ERROR_CODE_OPENSSL -DWOLFSSL_OPENVPN -DHAVE_KEYING_MATERIAL -DWOLFSSL_DES_ECB -DHAVE_EX_DATA -DWOLFSSL_KEY_GEN -DWOLFSSL_ALWAYS_VERIFY_CB -DWOLFSSL_ALWAYS_KEEP_SNI -DKEEP_OUR_CERT -DKEEP_PEER_CERT -DHAVE_EXT_CACHE -DHAVE_EX_DATA -DWOLFSSL_CERT_GEN -DWOLFSSL_ASIO -DASIO_USE_WOLFSSL -DWOLFSSL_KEY_GEN -DBOOST_ASIO_USE_WOLFSSL -DHAVE_EX_DATA -DSSL_TXT_TLSV1_2 -DOPENSSL_NO_SSL2 -DOPENSSL_NO_SSL3 -DHAVE_ENCRYPT_THEN_MAC -DHAVE_STUNNEL -DWOLFSSL_ALWAYS_VERIFY_CB -DWOLFSSL_ALWAYS_KEEP_SNI -DHAVE_EX_DATA -DWOLFSSL_DES_ECB -DWOLFSSL_SIGNER_DER_CERT -DOPENSSL_COMPATIBLE_DEFAULTS -DWOLFSSL_TICKET_HAVE_ID -DWOLFSSL_ALT_CERT_CHAINS -DWOLFSSL_IP_ALT_NAME -DNO_SESSION_CACHE_REF -DWOLFSSL_DES_ECB -DWOLFSSL_TICKET_NONCE_MALLOC -DWOLFSSL_ENCRYPTED_KEYS -DHAVE_SCRYPT -DWOLFSSL_SP_MATH_ALL -DWOLFSSL_SP_PPC -DWOLF_CRYPTO_CB -DWC_NO_ASYNC_THREADING -DHAVE_AES_KEYWRAP -DWOLFSSL_AES_DIRECT -DNO_OLD_RNGNAME -DNO_OLD_WC_NAMES -DNO_OLD_SSL_NAMES -DNO_OLD_SHA_NAMES -DNO_OLD_MD5_NAME -DWOLFSSL_HASH_FLAGS -DRSA_MAX_SIZE=4096 -DFP_MAX_BITS=8192 -DSP_INT_BITS=4096 -DWOLFSSL_TRUST_PEER_CERT -DNO_SESSION_CACHE_REF -DWOLFSSL_TLS13_NO_PEEK_HANDSHAKE_DONE -DWOLFSSL_ALT_CERT_CHAINS -DWOLFSSL_PRIORITIZE_PSK -DWOLFSSL_CHECK_ALERT_ON_ERR -DWOLFSSL_TICKET_HAVE_ID -DHAVE_RPK -DWOLFSSL_KEY_GEN -DWOLFSSL_CERT_REQ -DWOLFSSL_CERT_GEN -DWOLFSSL_CERT_EXT -DHAVE_ED25519 -DHAVE_OCSP -DWOLFSSL_ED25519_STREAMING_VERIFY -DOPENSSL_ALL -DWOLFSSL_EITHER_SIDE -DWC_RSA_NO_PADDING -DWC_RSA_PSS -DWOLFSSL_PSS_LONG_SALT -DWOLFSSL_TICKET_HAVE_ID -DWOLFSSL_ERROR_CODE_OPENSSL -DWOLFSSL_CERT_NAME_ALL -DWOLFSSL_AES_SIV -DOPENSSL_EXTRA -DWOLFSSL_HAVE_WOLFSCEP -DHAVE_PKCS7 -DNO_DES3_TLS_SUITES -DHAVE_AESGCM -DWOLFSSL_AESGCM_STREAM -DWOLFSSL_AESXTS_STREAM -DWOLFSSL_SRTP -DWOLFSSL_MULTICAST -DWOLFSSL_PUBLIC_MP -DHAVE_TLS_EXTENSIONS -DHAVE_SERVER_RENEGOTIATION_INFO -DHAVE_COMP_KEY -DWOLFSSL_ALLOW_RC4 -DWOLFSSL_TLS_OCSP_MULTI -Wall -Wno-unused -O2  -DHAVE_WC_INTROSPECTION -include ./.build_params   -pipe -Os -arch ppc   -Werror -Wno-pragmas -Wall -Wextra -Wunknown-pragmas --param=ssp-buffer-size=1 -Waddress -Wbad-function-cast -Wchar-subscripts -Wcomment -Wfloat-equal -Wformat-security -Wformat=2 -Wmissing-field-initializers -Wmissing-noreturn -Wmissing-prototypes -Wnested-externs -Wnormalized=id -Woverride-init -Wpointer-arith -Wpointer-sign -Wshadow -Wshorten-64-to-32 -Wsign-compare -Wstrict-overflow=1 -Wstrict-prototypes -Wswitch-enum -Wundef -Wunused -Wunused-variable -Wwrite-strings -fwrapv -MT wolfcrypt/benchmark/benchmark.o -MD -MP -MF $depbase.Tpo -c -o wolfcrypt/benchmark/benchmark.o wolfcrypt/benchmark/benchmark.c &&\
    mv -f $depbase.Tpo $depbase.Po
cc1: warnings being treated as errors
wolfcrypt/benchmark/benchmark.c: In function ‘bench_srtpkdf’:
wolfcrypt/benchmark/benchmark.c:8633: warning: declaration of ‘index’ shadows a global declaration
/usr/include/string.h:125: warning: shadowed declaration is here
make[2]: *** [wolfcrypt/benchmark/benchmark.o] Error 1
```

anhu, 2025-05-09:
Hi @barracuda156 , 

Sorry for my delayed response. This is an interesting problem you've found!!  My system seems to be a bit different than yours. Can you please verify my proposed solution https://github.com/wolfSSL/wolfssl/pull/8759 fixes your issue? 

Warm regards, Anthony

barracuda156, 2025-05-10:
@anhu This gave a rather surprising result, no gcc-4.2 just chokes on that source file:
```
depbase=`echo wolfcrypt/benchmark/benchmark.o | sed 's|[^/]*$|.deps/&|;s|\.o$||'`;\
    /usr/bin/gcc-4.2 -std=gnu99 -DHAVE_CONFIG_H -I.  -DWOLFSSL_HAVE_ASSERT_H -DWOLFSSL_WOLFSSH   -isystem/opt/local/include/LegacySupport -I/opt/local/include -fvisibility=hidden -DNDEBUG -DNO_DO178 -DTFM_NO_ASM -DWOLFSSL_NO_ASM -DWOLFSSL_ASN_TEMPLATE -DHAVE_CRL_IO -DHAVE_IO_TIMEOUT -DWOLFSSL_DER_LOAD -DKEEP_OUR_CERT -DKEEP_PEER_CERT -DWOLFSSL_SUBJ_DIR_ATTR -DWOLFSSL_FPKI -DWOLFSSL_SUBJ_INFO_ACC -DWOLFSSL_CERT_NAME_ALL -DWOLFSSL_VERBOSE_ERRORS -DHAVE_ECC_CDH -DHAVE_ECC_KOBLITZ -DHAVE_ECC_SECPR2 -DHAVE_ECC_SECPR3 -DWOLFSSL_DES_ECB -DHAVE_AES_DECRYPT -DHAVE_AES_ECB -DWOLFSSL_ALT_NAMES -DHAVE_FFDHE_2048 -DHAVE_FFDHE_3072 -DWOLFSSL_ASN_ALL -DWOLFSSL_DH_EXTRA -DWOLFSSL_ECDSA_DETERMINISTIC_K_VARIANT -DWOLFSSL_HAVE_ISSUER_NAMES -pthread -DHAVE_ECH -DWOLFSSL_DTLS -DWOLFSSL_DTLS_MTU -DWOLFSSL_QUIC -DHAVE_EX_DATA -DWOLFSSL_POST_HANDSHAKE_AUTH -DWOLFSSL_SEND_HRR_COOKIE -DWOLFSSL_LIBWEBSOCKETS -DHAVE_EX_DATA -DOPENSSL_NO_EC -DWOLFSSL_OPENSSH -DHAVE_EX_DATA -DWOLFSSL_BASE16 -DWOLFSSL_ERROR_CODE_OPENSSL -DTFM_TIMING_RESISTANT -DECC_TIMING_RESISTANT -DWC_RSA_BLINDING -DFORTRESS -DWOLFSSL_ALWAYS_VERIFY_CB -DWOLFSSL_AES_COUNTER -DWOLFSSL_AES_DIRECT -DWOLFSSL_DER_LOAD -DWOLFSSL_KEY_GEN -DPERSIST_SESSION_CACHE -DPERSIST_CERT_CACHE -DATOMIC_USER -DHAVE_PK_CALLBACKS -DWOLFSSL_AES_CBC_LENGTH_CHECKS -DHAVE_AESCCM -DWOLFSSL_AES_EAX -DWOLFSSL_AES_OFB -DWOLFSSL_AES_DIRECT -DWOLFSSL_AES_CFB -DWOLFSSL_ARMASM_NO_HW_CRYPTO -DWOLFSSL_USE_ALIGN -DHAVE_CAMELLIA -DWOLFSSL_MD2 -DHAVE_NULL_CIPHER -DWOLFSSL_RIPEMD -DHAVE_BLAKE2 -DHAVE_BLAKE2B -DHAVE_BLAKE2S -DWOLFSSL_SHA224 -DWOLFSSL_SHA512 -DWOLFSSL_SHA384 -DSESSION_CERTS -DWOLFSSL_SEP -DKEEP_PEER_CERT -DHAVE_HKDF -DHAVE_HPKE -DHAVE_X963_KDF -DHAVE_ECC -DECC_SHAMIR -DECC_MIN_KEY_SZ=224 -DHAVE_ECC_BRAINPOOL -DHAVE_CURVE25519 -DFP_ECC -DHAVE_ECC_ENCRYPT -DWOLFCRYPT_HAVE_ECCSI -DWOLFSSL_PUBLIC_MP -DWOLFCRYPT_HAVE_SAKKE -DNO_OLD_TLS -DWOLFSSL_QT -DSESSION_CERTS -DOPENSSL_NO_SSL2 -DWOLFSSL_KEY_GEN -DHAVE_EX_DATA -DWOLFSSL_CUSTOM_CURVES -DHAVE_ECC_SECPR2 -DHAVE_ECC_SECPR3 -DHAVE_ECC_BRAINPOOL -DHAVE_ECC_KOBLITZ -DWC_RSA_PSS -DWOLFSSL_PSS_LONG_SALT -DHAVE_ANON -DWOLFSSL_ASN_PRINT -DWOLFSSL_DTLS13 -DWOLFSSL_W64_WRAPPER -DWOLFSSL_AES_DIRECT -DWOLFSSL_DTLS_CID -DWOLFSSL_DTLS_CH_FRAG -DWOLFSSL_BASE64_ENCODE -DWOLFSSL_BASE16 -DWOLFSSL_SIPHASH -DWOLFSSL_CMAC -DWOLFSSL_AES_DIRECT -DHAVE_WEBSERVER -DWOLFSSL_AES_XTS -DWOLFSSL_AES_DIRECT -DWOLFSSL_CUSTOM_CURVES -DHAVE_ECC_SECPR2 -DHAVE_ECC_SECPR3 -DHAVE_ECC_BRAINPOOL -DHAVE_ECC_KOBLITZ -DHAVE_ECC_CDH -DHAVE_CURVE448 -DHAVE_ED448 -DWOLFSSL_ED448_STREAMING_VERIFY -DWC_SRTP_KDF -DHAVE_AES_ECB -DWOLFSSL_AES_DIRECT -DWOLFSSL_SHA3 -DWOLFSSL_SHAKE128 -DWOLFSSL_SHAKE256 -DHAVE_POLY1305 -DNO_CHACHA_ASM -DHAVE_CHACHA -DHAVE_XCHACHA -DHAVE_HASHDRBG -DHAVE_OPENSSL_CMD -DHAVE_TLS_EXTENSIONS -DHAVE_CERTIFICATE_STATUS_REQUEST -DHAVE_TLS_EXTENSIONS -DHAVE_CERTIFICATE_STATUS_REQUEST_V2 -DHAVE_CRL -DHAVE_CRL_MONITOR -DHAVE_TLS_EXTENSIONS -DHAVE_SNI -DHAVE_TLS_EXTENSIONS -DHAVE_ALPN -DHAVE_TLS_EXTENSIONS -DHAVE_MAX_FRAGMENT -DHAVE_TLS_EXTENSIONS -DHAVE_TRUSTED_CA -DHAVE_TLS_EXTENSIONS -DHAVE_TRUNCATED_HMAC -DHAVE_FALLBACK_SCSV -DHAVE_KEYING_MATERIAL -DHAVE_TLS_EXTENSIONS -DHAVE_SUPPORTED_CURVES -DHAVE_FFDHE_2048 -DHAVE_SUPPORTED_CURVES -DWOLFSSL_TLS13 -DHAVE_TLS_EXTENSIONS -DHAVE_TLS_EXTENSIONS -DHAVE_SESSION_TICKET -DHAVE_EXTENDED_MASTER -DHAVE_TLS_EXTENSIONS -DHAVE_SNI -DHAVE_MAX_FRAGMENT -DHAVE_TRUNCATED_HMAC -DHAVE_ALPN -DHAVE_TRUSTED_CA -DHAVE_SUPPORTED_CURVES -DWOLFSSL_EARLY_DATA -DHAVE_SMIME -DWOLFCRYPT_HAVE_SRP -DASN_BER_TO_DER -DWOLFSSL_HAVE_CERT_SERVICE -DHAVE_LIGHTY -DHAVE_WOLFSSL_SSL_H=1 -DHAVE_EX_DATA -DOPENSSL_ALL -DWOLFSSL_KEY_GEN -DWOLFSSL_NGINX -DWOLFSSL_SIGNER_DER_CERT -DOPENSSL_COMPATIBLE_DEFAULTS -DWOLFSSL_ERROR_CODE_OPENSSL -DWOLFSSL_OPENVPN -DHAVE_KEYING_MATERIAL -DWOLFSSL_DES_ECB -DHAVE_EX_DATA -DWOLFSSL_KEY_GEN -DWOLFSSL_ALWAYS_VERIFY_CB -DWOLFSSL_ALWAYS_KEEP_SNI -DKEEP_OUR_CERT -DKEEP_PEER_CERT -DHAVE_EXT_CACHE -DHAVE_EX_DATA -DWOLFSSL_CERT_GEN -DWOLFSSL_ASIO -DASIO_USE_WOLFSSL -DWOLFSSL_KEY_GEN -DBOOST_ASIO_USE_WOLFSSL -DHAVE_EX_DATA -DSSL_TXT_TLSV1_2 -DOPENSSL_NO_SSL2 -DOPENSSL_NO_SSL3 -DHAVE_ENCRYPT_THEN_MAC -DHAVE_STUNNEL -DWOLFSSL_ALWAYS_VERIFY_CB -DWOLFSSL_ALWAYS_KEEP_SNI -DHAVE_EX_DATA -DWOLFSSL_DES_ECB -DWOLFSSL_SIGNER_DER_CERT -DOPENSSL_COMPATIBLE_DEFAULTS -DWOLFSSL_TICKET_HAVE_ID -DWOLFSSL_ALT_CERT_CHAINS -DWOLFSSL_IP_ALT_NAME -DNO_SESSION_CACHE_REF -DWOLFSSL_DES_ECB -DWOLFSSL_TICKET_NONCE_MALLOC -DWOLFSSL_ENCRYPTED_KEYS -DHAVE_SCRYPT -DWOLF_CRYPTO_CB -DWC_NO_ASYNC_THREADING -DHAVE_AES_KEYWRAP -DWOLFSSL_AES_DIRECT -DNO_OLD_RNGNAME -DNO_OLD_WC_NAMES -DNO_OLD_SSL_NAMES -DNO_OLD_SHA_NAMES -DNO_OLD_MD5_NAME -DWOLFSSL_HASH_FLAGS -DRSA_MAX_SIZE=4096 -DFP_MAX_BITS=8192 -DSP_INT_BITS=4096 -DWOLFSSL_TRUST_PEER_CERT -DNO_SESSION_CACHE_REF -DWOLFSSL_TLS13_NO_PEEK_HANDSHAKE_DONE -DWOLFSSL_ALT_CERT_CHAINS -DWOLFSSL_PRIORITIZE_PSK -DWOLFSSL_CHECK_ALERT_ON_ERR -DWOLFSSL_TICKET_HAVE_ID -DHAVE_RPK -DWOLFSSL_KEY_GEN -DWOLFSSL_CERT_REQ -DWOLFSSL_CERT_GEN -DWOLFSSL_CERT_EXT -DHAVE_ED25519 -DHAVE_OCSP -DWOLFSSL_ED25519_STREAMING_VERIFY -DOPENSSL_ALL -DWOLFSSL_EITHER_SIDE -DWC_RSA_NO_PADDING -DWC_RSA_PSS -DWOLFSSL_PSS_LONG_SALT -DWOLFSSL_TICKET_HAVE_ID -DWOLFSSL_ERROR_CODE_OPENSSL -DWOLFSSL_CERT_NAME_ALL -DWOLFSSL_AES_SIV -DOPENSSL_EXTRA -DWOLFSSL_HAVE_WOLFSCEP -DHAVE_PKCS7 -DNO_DES [... cut ...]
{standard input}:16:non-relocatable subtraction expression, "_aes_aad_options" minus "L00000000001$pb"
{standard input}:16:symbol: "_aes_aad_options" can't be undefined in a subtraction expression
make[2]: *** [wolfcrypt/benchmark/benchmark.o] Error 1
make[2]: *** Waiting for unfinished jobs....
```

This does not necessarily mean there is something wrong though. Maybe it just cannot handle it, and we should require a modern gcc for building WolfSSL (this is not a problem, we have gcc14 on macOS 10.4+ including powerpc systems).

dgarske, 2025-05-15:
Hi @barracuda156 ,

Did you figure out those errors? To me it seems related to Position Independent Code (PIC). Can you try adding `CFLAGS="-fPIC"`?

Thanks,
David Garske, wolfSSL

closed 2025-05-15 by dgarske

barracuda156, 2025-05-15:
@dgarske `-fPIC` is the default on macOS, but let me just try building with gcc14 instead of the old Xcode one.

barracuda156, 2025-05-17:
@dgarske With gcc14 it builds normally without disabling `-Werror`. So all good with this fix. Now the only issue left is re-enabling powerpc assembler.