wolfSSL/wolfssl #8741: esp32/rom/aes.h: No such file or directory

diplfranzhoepfinger, 2025-05-07:
using: 

https://components.espressif.com/components/wolfssl/wolfssl/versions/5.7.6?language=en

and ESP-IDF 5.4

on a ESP32-C3

Compile log: 
https://gist.github.com/diplfranzhoepfinger/d93b40dabf57ce8bdff98012f6cc6ab0

snippet: 

```
idf/wolfssl__wolfssl/CMakeFiles/__idf_wolfssl__wolfssl.dir/src/tls13.c.obj -MF esp-idf/wolfssl__wolfssl/CMakeFiles/__idf_wolfssl__wolfssl.dir/src/tls13.c.obj.d -o esp-idf/wolfssl__wolfssl/CMakeFiles/__idf_wolfssl__wolfssl.dir/src/tls13.c.obj -c /home/franz/git/hm/lab02/aes-wolfssl/managed_components/wolfssl__wolfssl/src/tls13.c
In file included from /home/franz/git/hm/lab02/aes-wolfssl/managed_components/wolfssl__wolfssl/wolfssl/wolfcrypt/sha256.h:85,
                 from /home/franz/git/hm/lab02/aes-wolfssl/managed_components/wolfssl__wolfssl/wolfssl/wolfcrypt/random.h:109,
                 from /home/franz/git/hm/lab02/aes-wolfssl/managed_components/wolfssl__wolfssl/wolfssl/wolfcrypt/asn_public.h:36,
                 from /home/franz/git/hm/lab02/aes-wolfssl/managed_components/wolfssl__wolfssl/wolfssl/ssl.h:36,
                 from /home/franz/git/hm/lab02/aes-wolfssl/managed_components/wolfssl__wolfssl/wolfssl/internal.h:28,
                 from /home/franz/git/hm/lab02/aes-wolfssl/managed_components/wolfssl__wolfssl/src/tls13.c:115:
/home/franz/git/hm/lab02/aes-wolfssl/managed_components/wolfssl__wolfssl/wolfssl/wolfcrypt/port/Espressif/esp32-crypt.h:723:18: fatal error: esp32/rom/aes.h: No such file or directory
  723 |         #include "esp32/rom/aes.h"
      |                  ^~~~~~~~~~~~~~~~~
compilation terminated.
```

diplfranzhoepfinger, 2025-05-07:
ah, it seems this is already here: 

https://github.com/wolfSSL/wolfssl/pull/8380

but why is it not on https://components.espressif.com/components/wolfssl/wolfssl ?

gojimmypi, 2025-05-07:
Hi @diplfranzhoepfinger 

Thank you for using the wolfSSL Managed Component and for taking the time to point out the problem.

> but why is it not on https://components.espressif.com/components/wolfssl/wolfssl ?

The current wolfSSL Managed Component is version `5.7.6`, published a couple of months ago (~ Feb 23). It included some post-release changes from https://github.com/wolfSSL/wolfssl/pull/8418, but not https://github.com/wolfSSL/wolfssl/pull/8380. That v5.7.6 version is based on the [wolfSSL Release 5.7.6 (Dec 31, 2024)](https://github.com/wolfSSL/wolfssl/releases/tag/v5.7.6-stable).

The current release version of wolfSSL is [5.8.0](https://github.com/wolfSSL/wolfssl/releases/tag/v5.8.0-stable), tagged last month on April 24.

As you've encountered a highly undesired problem, I've published [wolfSSL Managed Component 5.8.0-preview1a](https://components.espressif.com/components/wolfssl/wolfssl/versions/5.8.0-preview1a?language=en) for you to try.

Here's an example `ESP32-C3` build for the [wolfSSL Crypt Test Example](https://components.espressif.com/components/wolfssl/wolfssl/versions/5.8.0-preview1a/examples/wolfssl_test) :

```bash
WRK_IDF_PATH=/mnt/c/SysGCC/esp32-14.2r1/esp-idf/v5.4
cd $WRK_IDF_PATH
. ./export.sh

cd /mnt/c/temp
mkdir test-5.8.0
cd test-5.8.0
idf.py create-project-from-example "wolfssl/wolfssl=5.8.0-preview1a:wolfssl_test"

cd wolfssl_test
idf.py set-target esp32c3
idf.py build
```

Normally there's a bit of delay between the wolfSSL release and the final Espressif publish for testing. 

I've not done any testing yet, other than confirming the ESP32-C3 target compiles using the `5.8.0-preview1a` wolfSSL Managed Component using ESP-IDF v5.4.

Note that I encountered a compile-time problem, and needed to add `#define HAVE_HKDF` for the` wolfssl_test` app. I'll create an upstream PR soon.

Please let me know if this version works properly for you. Any additional suggestions will be appreciated.

Cheers, 

Jim