wolfSSL/wolfssl #8754: Fragmented TLS 1.3 client hello is not correctly handled

VoodooChild99, 2025-05-09:
### Version

5.8.0 (b077c81eb635392e694ccedbab8b644297ec0285)

### Description

If the TLS 1.3 client hello message is fragmented multiple small records, the wolfSSL server fails to assemble the whole handshake message and immediately reports an error:
```
got HANDSHAKE
wolfSSL Entering wolfSSL_get_options
wolfSSL Entering DoTls13HandShakeMsg
wolfSSL error occurred, error = -306
wolfSSL error occurred, error = -306
wolfSSL Entering wolfSSL_get_error
wolfSSL Leaving wolfSSL_get_error, return -306
wolfSSL Entering wolfSSL_get_error
wolfSSL Leaving wolfSSL_get_error, return -306
wolfSSL Entering wolfSSL_ERR_error_string
SSL_accept error -306, parse error on header
wolfSSL error: SSL_accept failed
```

This seems to be an "Implementation Pitfall" as pointed out in the Section C.3 of RFC 8446: *"Do you correctly handle handshake messages that are fragmented to multiple TLS records?"*. BTW, OpenSSL 3.6.0 can correctly handle such messages.

The (compressed) `.pcap` file:

[fragmented_client_hello.pcap.gz](https://github.com/user-attachments/files/20116697/fragmented_client_hello.pcap.gz)

embhorn, 2025-05-09:
Hello @VoodooChild99 

Thanks for contacting wolfSSL Support. Could you tell us a bit about your project using wolfSSL?

Kind regards,
@embhorn - wolfSSL Support

VoodooChild99, 2025-05-11:
Hi @embhorn 

Thanks for your timely response. We're doing differential fuzzing to expose divergent behaviors in TLS implementations.

julek-wolfssl, 2025-05-20:
Hi @VoodooChild99 thank you for your report. In `GetHandshakeHeader` we require at least `HANDSHAKE_HEADER_SZ` to be available so that we know how much to allocate for the message. We feel that requiring at least 4 bytes is reasonable and that adding logic to handle one byte at a time may expose us to additional edge cases.

Sincerely
Juliusz

closed 2025-05-20 by julek-wolfssl