wolfSSL/wolfssl #8764: [Bug]: when decrypting a previously encrypted file, wc_AesGcmDecryptFinal() does not generate the right tag.

anthony-d-truong, 2025-05-12:
### Contact Details

anthony.dominic@twosixtech.com

### Version

latest, a clone of main.

### Description

I tried decrypting an AES256GCM encrypted file and at the end called wc_AesGcmDecryptFinal() to retrieve the tag.  The function returns an error code of -180 and a tag that does not match with the one generated after the encryption of the file (by wc_AesGcmEncryptFinal()).

### Relevant log output

```shell

```

anthony-d-truong, 2025-05-13:
> Hello @anthony-d-truong <https://github.com/anthony-d-truong>
>
> Thanks for contacting wolfSSL Support. Are you able to run the wolfcrypt
> test successfully?
> wolfcrypt/test/testwolfcrypt
>
> Could you tell us a bit about your project using wolfSSL?
>
> Kind regards,
> @embhorn <https://github.com/embhorn> - wolfSSL Support

Hello Eric,
If you look into aesgcm-file-encrypt.c at
https://github.com/wolfSSL/wolfssl-examples/blob/master/crypto/aes/aesgcm-file-encrypt.c
, wc_AesGcmEncryptFinal() is used to close out the current decryption
context and to retrieve the supposedly decryption tag.  IS this a bug?
Thanks.

Regards,
Anthony Dominic.

-- 

*Anthony Dominic Truong*
*EMBEDDED SOFTWARE DEVELOPER*
***@***.***
949-278-0583

twosixtech.com

[image: Two Six Technologies]

*    LinkedIn   *
<https://www.linkedin.com/company/twosixtechnologies/posts/>* |**    Blogs
  * <https://twosixtech.com/blog/> *|    Join Our Team! **  .*
<https://twosixtech.com/careers/current-openings/>

embhorn, 2025-05-13:
Hi @anthony-d-truong 

The `authTag` param passed to `wc_AesGcmDecryptFinal` is input. The tag is used to compare to the calculated tag during the decrypt operation. 

In the example, `wc_AesGcmEncryptFinal` is used since AES is symmetric, and the encrypt API actually returns the tag. The next line in the example compares the enc and dec tags:
```
if (ret == 0 && (memcmp(tag_enc, tag_dec, AESGCM_TAG_SIZE) != 0))
```

I agree that the demo is a bit confusing. I modified the example for clarity:
https://github.com/wolfSSL/wolfssl-examples/pull/503

embhorn, 2025-05-14:
I did just that!
https://github.com/wolfSSL/wolfssl-examples/pull/503/files#diff-64e7ee269c4bf290e740502c841a85d80db53e895b1797a46bfd71d6b81e2c30R447

anthony-d-truong, 2025-05-14:
Thanks, Eric.  I did not pay attention to the details.  Sorry.

-- 

*Anthony Dominic Truong*
*EMBEDDED SOFTWARE DEVELOPER*
***@***.***
949-278-0583

twosixtech.com

[image: Two Six Technologies]

*    LinkedIn   *
<https://www.linkedin.com/company/twosixtechnologies/posts/>* |**    Blogs
  * <https://twosixtech.com/blog/> *|    Join Our Team! **  .*
<https://twosixtech.com/careers/current-openings/>