wolfSSL/wolfssl #8767: [Bug]: DTLS Invalid Version Handling fails

bathooman, 2025-05-13:
### Version

v5.8.0-stable

### Description

The DTLS 1.2 RFC states that:

> There is no DTLS 1.1.

> DTLS 1.2 and 1.0 clients MUST use the version to indicate packet formatting.

> Unlike TLS, DTLS is resilient in the face of invalid records (e.g., invalid formatting, **length**, MAC, etc.).  In general, invalid records SHOULD be silently discarded, thus preserving the association; however, an error MAY be logged for diagnostic purposes.

**Actual behavior**    

Since DTLS 1.1 does not exist, any record with that version is considered wrongly formatted and should be discarded. However, for the `CH0` and `CH2`, one can provide the value `fe fe` (1.1) as the version for  the `record version`  and still complete the handshake. Moreover, the server replies with a higher version of the protocol to the client.

In the following, you can find the Wireshark log for both scenarios. This issue was previously discussed with @rizlik and there is an [attempt](https://github.com/wolfSSL/wolfssl/pull/5276) to fix it. However, the issue persists.

[capture.zip](https://github.com/user-attachments/files/20189822/capture.zip)

### Relevant log output

```shell

```

rizlik, 2025-05-14:
Thanks for reporting. 
I'll check it as soon as possible.

rizlik, 2025-05-19:
Hi @bathooman ,

The problem should be fixed in #8777, can you confirm that?
Thanks

bathooman, 2025-05-27:
The PR solves the issue. Thanks.

closed 2025-05-27 by bathooman