wolfSSL/wolfssl #8793: [Bug]: TLS 1.3 session using CCM mode fails

sroy9gmu, 2025-05-21:
### Contact Details

sroy9@gmu.edu

### Version

Master latest

### Description

TLS session using CCM mode for AES encryption is failing with below error:

SSL_accept error -308, error state on socket
wolfSSL error: SSL_accept failed

Configure options are listed below:
```
---
Configuration summary for wolfssl version 5.8.0

   * Installation prefix:        /usr/local
   * System type:                pc-linux-gnu
   * Host CPU:                   x86_64
   * C Compiler:                 gcc
   * C Flags:                    -DNO_AES_192 -DNO_AES_256 -DWOLFSSL_SP_NO_256   -Werror -Wno-pragmas -Wall -Wextra -Wunknown-pragmas --param=ssp-buffer-size=1 -Waddress -Warray-bounds -Wbad-function-cast -Wchar-subscripts -Wcomment -Wfloat-equal -Wformat-security -Wformat=2 -Wmaybe-uninitialized -Wmissing-field-initializers -Wmissing-noreturn -Wmissing-prototypes -Wnested-externs -Wnormalized=id -Woverride-init -Wpointer-arith -Wpointer-sign -Wshadow -Wsign-compare -Wstrict-overflow=1 -Wstrict-prototypes -Wswitch-enum -Wundef -Wunused -Wunused-result -Wunused-variable -Wwrite-strings -fwrapv
   * C++ Compiler:               
   * C++ Flags:                  
   * CPP Flags:                  
   * CCAS Flags:                 -DNO_AES_192 -DNO_AES_256 -DWOLFSSL_SP_NO_256  
   * LD Flags:                   
   * LIB Flags:                   -pie -z relro -z now -Werror 
   * Library Suffix:             
   * Debug enabled:              no
   * Coverage enabled:           
   * Warnings as failure:        yes
   * make -j:                    9
   * VCS checkout:               yes

   Features 
   * Experimental settings:      Forbidden
   * FIPS:                       no
   * Single threaded:            no
   * Filesystem:                 yes
   * OpenSSH Build:              no
   * OpenSSL Extra API:          no
   * OpenSSL Coexist:            no
   * Old Names:                  yes
   * Max Strength Build:         no
   * Distro Build:               no
   * Reproducible Build:         no
   * Side-channel Hardening:     yes
   * Single Precision Math:      no
   * SP implementation:          all
   * Fast Math:                  no
   * Heap Math:                  no
   * Assembly Allowed:           yes
   * sniffer:                    no
   * snifftest:                  no
   * ARC4:                       no
   * AES:                        yes
   * AES-NI:                     no
   * AVX for AES:                no
   * AES-CBC:                    no
   * AES-CBC length checks:      no
   * AES-GCM:                    no
   * AES-GCM streaming:          no
   * AES-CCM:                    yes
   * AES-CTR:                    yes
   * AES-CFB:                    no
   * AES-OFB:                    no
[... 205 lines cut ...]
   * AutoSAR :                   no

---
./configure flags: 'CFLAGS=-DNO_AES_192 -DNO_AES_256 -DWOLFSSL_SP_NO_256' --disable-aesgcm --disable-aescbc --enable-aesccm --enable-aesctr --enable-tls13 --enable-rsapss LIBS=-lm
---
```

### Reproduction steps

1. ./configure CFLAGS="-DNO_AES_192 -DNO_AES_256 -DWOLFSSL_SP_NO_256" --disable-aesgcm --disable-aescbc --enable-aesccm --enable-aesctr --enable-tls13 --enable-rsapss LIBS=-lm
2. make
3. sudo make install

Execute below commands in separate windows:
1. ./examples/server/server -v 4 -c rsa_srv_cert.pem -k rsa_srv_pvt.pem -A rsa_cert.pem -D dh_param.pem
2. ./examples/client/client -h 127.0.0.1 -v 4 -c rsa_cli_cert.pem -k rsa_cli_pvt.pem -A rsa_cert.pem -y

### Relevant log output

```shell
err=-308, server_test, examples/server/server.c, 3601
SSL_accept error -308, error state on socket
server_test, examples/server/server.c, 3630
wolfSSL error: SSL_accept failed
```

dgarske, 2025-05-21:
Hi @sroy9gmu ,

Which ECC curve are you trying to use? I noticed you are supplying `WOLFSSL_SP_NO_256`? Is that because you are trying to disable ECC P256? If so please also add `-DNO_ECC256`.

For TLS v1.3 and AES CCM-8 I don't see supported, but I will let the support captain @anhu run some tests. I only see we support `TLS13-AES256-GCM-SHA384` for AES-256bit.

Thanks,
David Garske, wolfSSL

sroy9gmu, 2025-05-21:
I did a `git pull` again now and can see compilation failing with below reproduction steps:

1. git pull
2. make clean
3. ./configure CFLAGS="-DNO_AES_192 -DNO_AES_256 -DNO_ECC256" --disable-aesgcm --disable-aescbc --enable-aesccm --enable-aesctr --enable-tls13 --enable-rsapss LIBS=-lm
```

---
Configuration summary for wolfssl version 5.8.0

   * Installation prefix:        /usr/local
   * System type:                pc-linux-gnu
   * Host CPU:                   x86_64
   * C Compiler:                 gcc
   * C Flags:                    -DNO_AES_192 -DNO_AES_256 -DNO_ECC256   -Werror -Wno-pragmas -Wall -Wextra -Wunknown-pragmas --param=ssp-buffer-size=1 -Waddress -Warray-bounds -Wbad-function-cast -Wchar-subscripts -Wcomment -Wfloat-equal -Wformat-security -Wformat=2 -Wmaybe-uninitialized -Wmissing-field-initializers -Wmissing-noreturn -Wmissing-prototypes -Wnested-externs -Wnormalized=id -Woverride-init -Wpointer-arith -Wpointer-sign -Wshadow -Wsign-compare -Wstrict-overflow=1 -Wstrict-prototypes -Wswitch-enum -Wundef -Wunused -Wunused-result -Wunused-variable -Wwrite-strings -fwrapv
   * C++ Compiler:               
   * C++ Flags:                  
   * CPP Flags:                  
   * CCAS Flags:                 -DNO_AES_192 -DNO_AES_256 -DNO_ECC256  
   * LD Flags:                   
   * LIB Flags:                   -pie -z relro -z now -Werror 
   * Library Suffix:             
   * Debug enabled:              no
   * Coverage enabled:           
   * Warnings as failure:        yes
   * make -j:                    9
   * VCS checkout:               yes

   Features 
   * Experimental settings:      Forbidden
   * FIPS:                       no
   * Single threaded:            no
   * Filesystem:                 yes
   * OpenSSH Build:              no
   * OpenSSL Extra API:          no
   * OpenSSL Coexist:            no
   * Old Names:                  yes
   * Max Strength Build:         no
   * Distro Build:               no
   * Reproducible Build:         no
   * Side-channel Hardening:     yes
   * Single Precision Math:      no
   * SP implementation:          all
   * Fast Math:                  no
   * Heap Math:                  no
   * Assembly Allowed:           yes
   * sniffer:                    no
   * snifftest:                  no
   * ARC4:                       no
   * AES:                        yes
   * AES-NI:                     no
   * AVX for AES:                no
   * AES-CBC:                    no
   * AES-CBC length checks:      no
   * AES-GCM:                    no
   * AES-GCM streaming:          no
   * AES-CCM:                    yes
   * AES-CTR:                    yes
   * AES-CFB:                    no
   * AES-OFB:                    no
   * AES-XTS:                    no
   * AES-XTS streaming:          no
   * AES-SIV:                    no
   * AES-EAX:                    no
   * AES Bitspliced:             no
   * AES Key Wrap:               no
   * ARIA:                       no
   * ASCON:                      no
   * DES3:                       no
   * DES3 TLS Suites:            no
[... 249 lines cut ...]
wolfcrypt/test/test.c:32363:32: error: ‘ECC_KEYGEN_SIZE’ undeclared (first use in this function); did you mean ‘RC4_KEY_SIZE’?
32363 |     ret = wc_ecc_make_key(rng, ECC_KEYGEN_SIZE, key);
      |                                ^~~~~~~~~~~~~~~
      |                                RC4_KEY_SIZE
wolfcrypt/test/test.c:32401:29: error: ‘eccKeyDerFile’ undeclared (first use in this function); did you mean ‘eccPubKeyDerFile’?
32401 |         XFILE file = XFOPEN(eccKeyDerFile, "rb");
      |                             ^~~~~~~~~~~~~
      |                             eccPubKeyDerFile
wolfcrypt/test/test.c: In function ‘ecc_test_allocator’:
wolfcrypt/test/test.c:33611:32: error: ‘ECC_KEYGEN_SIZE’ undeclared (first use in this function); did you mean ‘RC4_KEY_SIZE’?
33611 |     ret = wc_ecc_make_key(rng, ECC_KEYGEN_SIZE, key);
      |                                ^~~~~~~~~~~~~~~
      |                                RC4_KEY_SIZE
```
I am testing AES-128-CCM mode only as shown by below settings:

```
* AES-GCM:                    no
* AES-GCM streaming:          no
* AES-CCM:                    yes
* AES-CTR:                    yes
```
Previously I tested for AES-128-GCM mode  and FFDHE_2048 successfully using below settings:

./configure CFLAGS="-DNO_AES_192 -DNO_AES_256 -DWOLFSSL_SP_NO_256" --enable-aesgcm --enable-aesctr --enable-tls13 --enable-rsapss LIBS=-lm

anhu, 2025-05-22:
Thanks for all the details.  I'll need to look into this for you.  Please stay tuned. 
Warm regards, Anthony

closed 2025-09-02 by anhu