wolfSSL/wolfssl #8892: wc_RsaPrivateDecrypt_ex returns MISSING_RNG_E

hrkcz001, 2025-06-17:
### Version

v5.8.0

### Description

Calling of
```
int32_t decrypted_len = wc_RsaPrivateDecrypt_ex(
                    ctx->pkg.encrypted_aes_key.data, sizeof(ctx->pkg.encrypted_aes_key.data),
                    ctx->pkg.aes_key.key, sizeof(ctx->pkg.aes_key.key),
                    &ctx->private_key.key,
                    WC_RSA_OAEP_PAD,
                    WC_HASH_TYPE_SHA256,
                    WC_MGF1SHA256,
                    NULL, 0
                );
```
returns -236.

Does it mean MISSING_RNG_E(which is not expected by wc_RsaPrivateDecrypt_ex) or represent another error?

embhorn, 2025-06-18:
Hi @hrkcz001 

If `WC_RSA_BLINDING` is enabled (which it is by default), then the RNG is required. You can either set up the RNG or disable the blinding feature.

Can you tell us a bit about your project using wolfSSL?

Thanks,
@embhorn - wolfSSL Support

hrkcz001, 2025-06-20:
@embhorn 
I'm making a MISRA C Compliant encryptor for my study project
https://github.com/hrkcz001/embedded_encryptor