wolfSSL/wolfssl #8941: Get full OCTET STRING of an extension

ribes96, 2025-06-30:
### Version

5.7.6-stable

### Description

RFC5280 defines an X.509 extension as
~~~asn1
   Extension  ::=  SEQUENCE  {
        extnID      OBJECT IDENTIFIER,
        critical    BOOLEAN DEFAULT FALSE,
        extnValue   OCTET STRING
                    -- contains the DER encoding of an ASN.1 value
                    -- corresponding to the extension type identified
                    -- by extnID
        }
~~~

OpenSSL's function `X509_EXTENSION_get_data` returns the full `extnValue   OCTET STRING`.
But WolfSSL's function `wolfSSL_X509_EXTENSION_get_data` doesn't seem to get the full `OCTET STRING`, but some data specific for each extension.

For example, for `AuthorityKeyIdentifier`, which is defined as
~~~asn1
   AuthorityKeyIdentifier ::= SEQUENCE {
      keyIdentifier             [0] KeyIdentifier           OPTIONAL,
      authorityCertIssuer       [1] GeneralNames            OPTIONAL,
      authorityCertSerialNumber [2] CertificateSerialNumber OPTIONAL  }
~~~
it directly gives the `keyIdentifier` array (in my case `{0x3d, 0x1, 0xd3, 0x87,...`) instead of the SEQUENCE, which should start with `{0x30, 0x16, 0x80, 0x14, 0x3d, 0x1, 0xd3, 0x87,...`

`wolfSSL_X509_EXTENSION_get_data` is aliased to `X509_EXTENSION_get_data`, so I guess the intention is for them to have the same behavior. If they are not, is there a way to get the full `extnValue   OCTET STRING`?

kareem-wolfssl, 2025-07-01:
Hi @ribes96 ,

Can you attach a full reproducer for this?  Are you able to retest on the latest wolfSSL master and confirm if the issue is still present?

ribes96, 2025-07-02:
In the same way, the following code generates a certificate without extensions when compiling against `wolfssl`, but with a single extension when compiling against `openssl`:

~~~c
#include <stdio.h>
#include <err.h>

#ifdef USE_WOLFSSL

#define WOLFSSL_USE_OPTIONS_H

#include <wolfssl/ssl.h>
#include <wolfssl/openssl/pem.h>
#include <wolfssl/openssl/ssl.h>
#include <wolfssl/openssl/buffer.h>

#endif /* USE_WOLFSSL */

#ifdef USE_OPENSSL

#include <openssl/x509.h>
#include <openssl/pem.h>
#include <openssl/err.h>
#include <openssl/evp.h>
#include <openssl/x509v3.h>

#endif /* USE_OPENSSL */

#define ARRAY_LEN(a)           (sizeof (a) / sizeof (a[0]))

static const uint8_t g_data[] =
{
	0x30, 0x16, 0x80, 0x14, 0x60, 0x7B, 0x66, 0x1A,
	0x45, 0x0D, 0x97, 0xCA, 0x89, 0x50, 0x2F, 0x7D,
	0x04, 0xCD, 0x34, 0xA8, 0xFF, 0xFC, 0xFD, 0x4B,
};

static EVP_PKEY *generate_rsa_key ();

int main ()
{
	X509               *x509;
	ASN1_OBJECT        *object;
	ASN1_OCTET_STRING  *octet_string;
	X509_EXTENSION     *ext;
	EVP_PKEY           *pkey;
	X509_NAME          *name;

	if ((pkey = generate_rsa_key ()) == NULL)
		err (-1, "generate_rsa_key");

	if ((x509 = X509_new ()) == NULL)
		err (-1, "X509_new");

	if (X509_set_version (x509, 2) == 0)
		err (-1, "X509_set_version");

	ASN1_INTEGER_set (X509_get_serialNumber (x509), 1);

	if (X509_gmtime_adj (X509_get_notBefore (x509), 0) == NULL)
		err (-1, "X509_get_notBefore");

	if (X509_gmtime_adj (X509_get_notAfter (x509), 60*60*24*365) == NULL)
		err (-1, "X509_get_notAfter");

	if (X509_set_pubkey (x509, pkey) == 0)
		err (-1, "X509_set_pubkey");

	if ((name = X509_get_subject_name (x509)) == NULL)
		err (-1, "X509_get_subject_name");

	X509_NAME_add_entry_by_txt (name, "CN", MBSTRING_ASC, (const unsigned char *)"Dummy Cert", -1, -1, 0);

	if (X509_set_issuer_name(x509, name) == 0)
		err (-1, "X509_set_issuer_name");

	if ((object = OBJ_txt2obj ("2.5.29.35", 1)) == NULL)
		err (-1, "OBJ_txt2obj");

	if ((octet_string = ASN1_OCTET_STRING_new ()) == NULL)
		err (-1, "ASN1_OCTET_STRING_new");

	if (ASN1_OCTET_STRING_set (octet_string, g_data, ARRAY_LEN (g_data)) == 0)
		err (-1, "ASN1_OCTET_STRING_set");

	if ((ext = X509_EXTENSION_create_by_OBJ (NULL, object, 0, octet_string)) == NULL)
		err (-1, "X509_EXTENSION_create_by_OBJ");

	if (X509_add_ext (x509, ext, -1) == 0)
		err (-1, "X509_add_ext");

	if (X509_sign (x509, pkey, EVP_sha256 ()) == 0)
		err (-1, "X509_sign");

	if (PEM_write_X509 (stdout, x509) == 0)
		err (-1, "PEM_write_X509");

	X509_free (x509);
	return 0;
}

static EVP_PKEY *generate_rsa_key ()
{
	RSA        *rsa;
	EVP_PKEY   *pkey;
	BIGNUM     *e;

	if ((pkey = EVP_PKEY_new ()) == NULL)
		err (-1, "EVP_PKEY_new");

	if ((rsa = RSA_new ()) == NULL)
		err (-1, "RSA_new");

	if ((e = BN_new ()) == NULL)
		err (-1, "BN_new");

	if (BN_set_word (e, RSA_F4) == 0)
		err (-1, "BN_set_word");

	if (RSA_generate_key_ex (rsa, 2048, e, NULL) == 0)
		err (-1, "RSA_generate_key_ex");

	if (EVP_PKEY_assign_RSA (pkey, rsa) == 0)
		err (-1, "EVP_PKEY_assign_RSA");

	BN_free (e);

	return pkey;
}
~~~