wolfSSL/wolfssl #8951: Outside array bounds warnings from GCC: wolfcrypt/src/sp_int.c

tjko, 2025-07-02:
### Version

5.8.0

### Description

I noticed following warnings when compiling wolfSSL:

```
In function '_sp_init_size',
    inlined from 'sp_init_size' at /home/tjko/git/fanpico/libs/wolfssl/wolfcrypt/src/sp_int.c:5126:9,
    inlined from '_sp_sqrmod' at /home/tjko/git/fanpico/libs/wolfssl/wolfcrypt/src/sp_int.c:17303:15,
    inlined from 'sp_sqrmod' at /home/tjko/git/fanpico/libs/wolfssl/wolfcrypt/src/sp_int.c:17359:15,
    inlined from '_sp_exptmod_ex' at /home/tjko/git/fanpico/libs/wolfssl/wolfcrypt/src/sp_int.c:13055:19:
/home/tjko/git/fanpico/libs/wolfssl/wolfcrypt/src/sp_int.c:5105:13: warning: array subscript 'sp_int[0]' is partly outside array bounds of 'sp_int[0]' [-Warray-bounds=]
 5105 |     a->size = (sp_size_t)size;
      |     ~~~~~~~~^~~~~~~~~~~~~~~~~
In file included from /home/tjko/git/fanpico/libs/wolfssl/wolfssl/wolfcrypt/libwolfssl_sources.h:46,
                 from /home/tjko/git/fanpico/libs/wolfssl/wolfcrypt/src/sp_int.c:30:
In function '_sp_sqrmod',
    inlined from 'sp_sqrmod' at /home/tjko/git/fanpico/libs/wolfssl/wolfcrypt/src/sp_int.c:17359:15,
    inlined from '_sp_exptmod_ex' at /home/tjko/git/fanpico/libs/wolfssl/wolfcrypt/src/sp_int.c:13055:19:
/home/tjko/git/fanpico/libs/wolfssl/wolfssl/wolfcrypt/types.h:577:61: note: object of size [8, 772] allocated by 'malloc'
  577 |         #define XMALLOC(s, h, t)     ((void)(h), (void)(t), malloc((size_t)(s))) /* native heap */
      |                                                             ^~~~~~~~~~~~~~~~~~~
/home/tjko/git/fanpico/libs/wolfssl/wolfcrypt/src/sp_int.c:148:28: note: in expansion of macro 'XMALLOC'
  148 |             (n) = (sp_int*)XMALLOC(MP_INT_SIZEOF(s), (h),                      \
      |                            ^~~~~~~
/home/tjko/git/fanpico/libs/wolfssl/wolfcrypt/src/sp_int.c:17301:5: note: in expansion of macro 'ALLOC_SP_INT'
17301 |     ALLOC_SP_INT(t, a->used * 2, err, NULL);
      |     ^~~~~~~~~~~~
In function '_sp_init_size',
    inlined from 'sp_init_size' at /home/tjko/git/fanpico/libs/wolfssl/wolfcrypt/src/sp_int.c:5126:9,
    inlined from '_sp_sqrmod' at /home/tjko/git/fanpico/libs/wolfssl/wolfcrypt/src/sp_int.c:17303:15,
    inlined from 'sp_sqrmod' at /home/tjko/git/fanpico/libs/wolfssl/wolfcrypt/src/sp_int.c:17359:15,
    inlined from 'sp_prime_miller_rabin.part.0.constprop' at /home/tjko/git/fanpico/libs/wolfssl/wolfcrypt/src/sp_int.c:19071:23:
/home/tjko/git/fanpico/libs/wolfssl/wolfcrypt/src/sp_int.c:5105:13: warning: array subscript 'sp_int[0]' is partly outside array bounds of 'sp_int[0]' [-Warray-bounds=]
 5105 |     a->size = (sp_size_t)size;
      |     ~~~~~~~~^~~~~~~~~~~~~~~~~
In function '_sp_sqrmod',
    inlined from 'sp_sqrmod' at /home/tjko/git/fanpico/libs/wolfssl/wolfcrypt/src/sp_int.c:17359:15,
    inlined from 'sp_prime_miller_rabin.part.0.constprop' at /home/tjko/git/fanpico/libs/wolfssl/wolfcrypt/src/sp_int.c:19071:23:
/home/tjko/git/fanpico/libs/wolfssl/wolfssl/wolfcrypt/types.h:577:61: note: object of size [8, 772] allocated by 'malloc'
  577 |         #define XMALLOC(s, h, t)     ((void)(h), (void)(t), malloc((size_t)(s))) /* native heap */
      |                                                             ^~~~~~~~~~~~~~~~~~~
/home/tjko/git/fanpico/libs/wolfssl/wolfcrypt/src/sp_int.c:148:28: note: in expansion of macro 'XMALLOC'
  148 |             (n) = (sp_int*)XMALLOC(MP_INT_SIZEOF(s), (h),                      \
      |                            ^~~~~~~
/home/tjko/git/fanpico/libs/wolfssl/wolfcrypt/src/sp_int.c:17301:5: note: in expansion of macro 'ALLOC_SP_INT'
17301 |     ALLOC_SP_INT(t, a->used * 2, err, NULL);
      |     ^~~~~~~~~~~~

```

user_options.h can be found here: https://github.com/tjko/fanpico/blob/7f7d89c3d2eb6f7764bbbd6cbb8f64be8dada5c2/src/wolfssl/user_settings.h

dgarske, 2025-07-03:
Hi @tjko ,

Please try adding `WOLFSSL_SP_NO_DYN_STACK`. Please let us know if that helps.

Thanks, David Garske, wolfSSL

dgarske, 2025-07-03:
I don't think it will help, but please comment out these two, since you have DH and RSA disabled.

#define WOLFSSL_HAVE_SP_RSA
#define WOLFSSL_HAVE_SP_DH

Also consider manually setting `#define SP_INT_BITS 256`.

dgarske, 2025-07-03:
Also @tjko can you please tell us about your project? If you would like to keep that information private please email facts at wolfssl dot com and reference this github issue.

tjko, 2025-07-03:
> Please try adding `WOLFSSL_SP_NO_DYN_STACK`. Please let us know if that helps.
> 
> Thanks, David Garske, wolfSSL

This didn't have any effect on the compiler warnings.

tjko, 2025-07-03:
> I don't think it will help, but please comment out these two, since you have DH and RSA disabled.
> 
> #define WOLFSSL_HAVE_SP_RSA #define WOLFSSL_HAVE_SP_DH
> 
> Also consider manually setting `#define SP_INT_BITS 256`.

Thanks for the tip (btw, these didn't affect the warnings GCC generates either).

Configuration I have is based on the example one from wolfssl-examples (https://github.com/wolfSSL/wolfssl-examples/blob/master/RPi-Pico/config/user_settings.h). I have only made some minor tweaks to get wolfSSH working on extremely limited memory on RP2040...

tjko, 2025-07-03:
> Also [@tjko](https://github.com/tjko) can you please tell us about your project? If you would like to keep that information private please email facts at wolfssl dot com and reference this github issue.

@dgarske,  I'm experimenting implementing SSH support using wolfSSH in my OSHW project: https://github.com/tjko/fanpico

So far, wolfSSH seems to work pretty well with ECC algorithms on RPi Pico.   RSA didn't seem to work "out of the box", likely running out of stack memory (or something) since key generation was causing hang...  RP2040 might be too "slow" to run RSA anyways, so haven't looked into that further yet...

dgarske, 2025-07-03:
Hi @tjko ,

Thanks for sharing details about your project! I cannot reproduce this issue using your build settings with v5.8.0-stable or latest master using `arm-none-eabi-gcc (15:12.2.rel1-1) 12.2.1 20221205`. I tried a few different ways (first with the wolfssl-examples/RPi-Pico and second with our IDE/GCC-ARM using your user_settings.h). 

Can you share the steps to build and reproduce these warnings with your fanpico project?

Thanks,
David Garske, wolfSSL

dgarske, 2025-07-10:
Hi @tjko , we were able to reproduce the warning, but currently do not believe there is an issue. @sebastian-carpenter is continuing to look into it to see if there is a way to appease the compiler. Thanks

sebastian-carpenter, 2025-07-10:
Hey @tjko, could you check if PR #8985 fixes the warning. I got it to work on my machine using your fanpico make system, just want to make sure it works on your end.

tjko, 2025-07-10:
Hi @sebastian-carpenter, yes this seems to fix the warning, thanks!

I tested using:  ```arm-none-eabi-gcc (Arm GNU Toolchain 14.3.Rel1 (Build arm-14.174)) 14.3.1 20250623```

closed 2025-07-10 by dgarske