wolfSSL/wolfssl #8958: CWE-190: wolfssl/wolfcrypt/src/sp_int.c:7616

tjko, 2025-07-03:
### Version

5.8.0

### Description

CodeQL flagged this:

>Comparison of narrow type with wide type in loop condition:

>In a loop condition, comparison of a value of a narrow type with a value of a wide type may result in unexpected behavior if the wider value is sufficiently large (or small). This is because the narrower value may overflow. This can lead to an infinite loop.

libs/wolfssl/wolfcrypt/src/sp_int.c:7616: Comparison between i of type sp_size_t and o of wider type unsigned int.
```
   7614     /* Need to copy digits up to offset into result. */
   7615     if (r != a) {
   7616         for (; (i < o) && (i < a->used); i++) {
   7617             r->dp[i] = a->dp[i];
   7618         }
   7619     }
```

Not sure if this is causing any issue in this case? But technically finding seems correct, if _sp_sub_off() is ever callew with o parameter set to >= 2^16, then comparison "i < o" will always be true regardless value of i...

SparkiDev, 2025-07-03:
Hi @tjko 

I've put up a fix for this: #8963
(Explanation of fix in PR description.)

Let us know if CodeQL still has an issue with this code.

Sean

SparkiDev, 2025-07-04:
Hi @tjko,

Is there anything more for this issue?

Thanks,
Sean

tjko, 2025-07-04:
Seems good to me.  (I only reported it since it seemed like something you might want to look into, if you think it warranted fix or anything...)

PR #8963 (2025-07-09): CodeQL: o is larger type and could cause issues
# Description

Make 'o' sp_size_t as the callers are passing 0 or explicit cast to sp_size_t

Fixes #8958

# Testing

Standard

# Checklist

 - [ ] added tests
 - [ ] updated/added doxygen
 - [ ] updated appropriate READMEs
 - [ ] Updated manual and documentation
closed 2025-07-04 by tjko