wolfSSL/wolfssl #8997: [Bug]: Discarding Records with Large Record Sequence Numbers

bathooman, 2025-07-13:
### Version

All versions

### Description

### Configuration:

`./configure --disable-asm --enable-singlethreaded --enable-dtls --enable-dtls-mtu --enable-psk-one-id --enable-psk --enable-static --disable-extended-master --enable-opensslextra --enable-tlsv10 --enable-debug --enable-nullcipher --enable-all-crypto`

### Actual Behavior
During an interaction, if the server receives an initial ClientHello or the second ClientHello with a large record sequence number, it discards the message noting that the `record is from way too far in the future`. Although, according to [DTLS 1.2 RFC [p. 9]](https://datatracker.ietf.org/doc/html/rfc6347#page-9), sequence number wrapping MUST be prevented, the sequence number in this case in not large enough to cause wrapping.

#### PoC
Please check the attached PCAP file for an example interaction. 

[large-rseq.zip](https://github.com/user-attachments/files/21203957/large-rseq.zip)

### Relevant log output

```shell

```

embhorn, 2025-07-14:
Hi @bathooman 

Thanks for submitting this report. I have requested a review of this by our engineers. Could you tell us a bit about your project using wolfSSL?

Kind regards,
@embhorn - wolfSSL Support

bathooman, 2025-07-17:
Hi,

We are using WolfSSL as part of a scientific project. 

Best,
Hooman

rizlik, 2025-07-17:
Thanks for the report @bathooman, 

Other than stressing the RFC do you have another use case to use a large record number on the CH?  (other than stressing RFC obviously). 
We will address this in a future refactoring, but it will be not a priority for the moment.