wolfSSL/wolfssl #9030: [Bug]: Ssl_cipher list order matters on MariaDB linked against WolfSSL

hholzgra, 2025-07-24:
### Contact Details

hartmut@mariadb.com

### Version

WolfSSL 5.7.6

### Description

With MariaDB linked against WolfSSL and specifying a list of accepted ciphers; TLS v1.3 clients can only connect when TLS v1.3 ciphers are listed first. Order does not matter when using OpenSSL instead.

Not sure whether that is an upstream or downstream problem after all, but as it works with OpenSSL it rather seems upstream, so I'm filing it here, too.

Related MariaDB bug report is: 

https://jira.mariadb.org/browse/MDEV-37307

### Relevant log output

```shell

```

anhu, 2025-07-24:
Hi, I noticed you're seeing this in wolfssl 5.7.6 .  Can you try with the latest 5.8.2 to see if it still behaves this way?

hholzgra, 2025-09-29:
Not sure if it is the one actual release binaries were built with, but this is what I get when building the MariaDB server with
`cmake -DWITH_SSL=bundled ..`

```
#ifndef WOLFSSL_USER_SETTINGS_H
#define WOLFSSL_USER_SETTINGS_H

#define HAVE_CRL
#define WOLFSSL_HAVE_ERROR_QUEUE

/*
 Workaround bug in 5.7.6
 WOLFSSL_MYSQL_COMPATIBLE breaks building wolfssl

 But it is needed to avoid redefinition of protocol_version
 when its public header ssl.h is included
*/
#ifndef BUILDING_WOLFSSL
#define WOLFSSL_MYSQL_COMPATIBLE
#endif

#define SP_INT_BITS 8192
#define HAVE_EMPTY_AGGREGATES 0

#define HAVE_ECC
#define ECC_TIMING_RESISTANT
#define HAVE_HASHDRBG
#define WOLFSSL_AES_DIRECT
#define WOLFSSL_SHA384
#define WOLFSSL_SHA512
#define WOLFSSL_SHA224
#define SESSION_CERT
#define KEEP_OUR_CERT
#define WOLFSSL_STATIC_RSA
#define WOLFSSL_USER_IO
#define WC_RSA_BLINDING
#define HAVE_TLS_EXTENSIONS
#define HAVE_AES_ECB
#define HAVE_AESGCM
#define HAVE_CHACHA
#define HAVE_POLY1305
#define HAVE_THREAD_LS
#define WOLFSSL_AES_COUNTER
#define NO_WOLFSSL_STUB
// #define OPENSSL_ALL
#define OPENSSL_EXTRA
#define WOLFSSL_ALLOW_TLSV10
#define NO_OLD_TIMEVAL_NAME
#define HAVE_SECURE_RENEGOTIATION
#define HAVE_EXTENDED_MASTER
/*
  Following is workaround about a WolfSSL 5.6.6 bug.
  The bug is about undefined sessionCtxSz during compilation.
*/
#define WOLFSSL_SESSION_ID_CTX

/* TLSv1.3 definitions (all needed to build) */
#define WOLFSSL_TLS13
#define HAVE_HKDF
#define HAVE_TLS_EXTENSIONS
#define HAVE_SUPPORTED_CURVES
#define HAVE_FFDHE_2048
#define WC_RSA_PSS
/* End of TLSv1.3 defines */

/* Features we exclude */
#define NO_DSA
#define NO_HC128
#define NO_MD4
#define NO_PSK
#define NO_RABBIT
#define NO_RC4

#define RSA_MAX_SIZE 8192
#define WOLFSSL_SP_MATH_ALL
#define WOLFSSL_HAVE_SP_RSA
#ifndef WOLFSSL_SP_4096
#define WOLFSSL_SP_4096
#endif

#define WOLFSSL_AESNI
#define HAVE_INTEL_RDSEED
#define HAVE_INTEL_RDRAND
#define USE_INTEL_SPEEDUP
#define WOLFSSL_X86_64_BUILD
#define WOLFSSL_SP_X86_64
#define WOLFSSL_SP_X86_64_ASM

#endif  /* WOLFSSL_USER_SETTINGS_H */
```

jackctj117, 2025-10-01:
@hholzgra  I have been able to reproduce the issue with MariaDB version 12.0.2 with wolfssl 5.8.0. When I linked against wolfssl 5.8.2 it looks like the error is still present. I am looking to more into the issue to find why this is occurring.

jackctj117, 2025-11-13:
Hello @hartmut-mariadb! I have a [PR](https://github.com/wolfSSL/wolfssl/pull/9420) up right now for a fix.