wolfSSL/wolfssl #9059: [Bug]: i2d_PUBKEY inconsistent with OpenSSL

ribes96, 2025-08-04:
### Contact Details

aribes96@gmail.com

### Version

5.8.2, and also current master

### Description

[man3(i2d_PUBKEY)](https://docs.openssl.org/1.1.1/man3/X509_PUBKEY_new/) says this about function `i2d_PUBKEY`

~~~man
d2i_PUBKEY() and i2d_PUBKEY() decode and encode an EVP_PKEY structure using
SubjectPublicKeyInfo format. They otherwise follow the conventions of other
ASN.1 functions such as d2i_X509().
~~~

~~~asn1
SubjectPublicKeyInfo  ::=  SEQUENCE  {
     algorithm            AlgorithmIdentifier,
     subjectPublicKey     BIT STRING
}

AlgorithmIdentifier  ::=  SEQUENCE  {
     algorithm               OBJECT IDENTIFIER,
     parameters              ANY DEFINED BY algorithm OPTIONAL
}
~~~

`wolfSSL`s DER output is `SubjectPublicKeyInfo` format only if the public key is RSA. When the public key is EC, the DER is directly the `BIT STRING`. This is inconsistent with the behavior of OpenSSL. This can be observed in the following program:

~~~c
#include <stdio.h>
#include <err.h>

#ifdef USE_WOLFSSL

#define WOLFSSL_USE_OPTIONS_H

#include <wolfssl/ssl.h>
#include <wolfssl/openssl/pem.h>
#include <wolfssl/openssl/ssl.h>
#include <wolfssl/openssl/buffer.h>

#endif /* USE_WOLFSSL */

#ifdef USE_OPENSSL

#include <openssl/x509.h>
#include <openssl/pem.h>
#include <openssl/err.h>

#endif /* USE_OPENSSL */

int main (int argc, char **argv)
{
	const char     *filename;
	EVP_PKEY       *pkey;
	int            len;
	int            i;
	uint8_t        *buf;
	uint8_t        *p;
	FILE           *f;

	if (argc < 2)
		errx (-1, "Error: Usage: <filename>");

	filename = argv[1];

	if ((f = fopen (filename, "r")) == NULL)
		err (-1, "%s", filename);

	if ((pkey = PEM_read_PUBKEY (f, NULL, NULL, NULL)) == NULL)
		errx (-1, "%s", "PEM_read_PUBKEY");

	fclose (f);

	if ((len = i2d_PUBKEY (pkey, NULL)) <= 0)
	{
		errx (-1, "Error: i2d_PUBKEY returned %d", len);
		return -1;
	}

	buf = calloc (1, len);
	p = buf;

	if (i2d_PUBKEY (pkey, &p) <= 0)
	{
		errx (-1, "Error: i2d_PUBKEY returned <= 0");
		return -1;
	}

	printf ("pubkey={");
	for (i = 0; i < len; ++i)
		printf ("%02X:", (unsigned char) buf[i]);
	printf ("}\n");

	free (buf);
	EVP_PKEY_free (pkey);

	return 0;
}
~~~

For example with these keys
~~~openssl

# rsa.pub
-----BEGIN PUBLIC KEY-----
MIIBIjANBgkqhkiG9w0BAQEFAAOCAQ8AMIIBCgKCAQEA+I+Nf1uaeZf609q2nn4L
ZcZ+ALiZcWfHo6cnVaQNpEtIpdhiTTA+vroPYL8t+Py5j9IcUIG+Dj0GsbWUJzLb
d+DcGSb5BbSwEQkE+xtEn3YCgsZoiwzhXDxbj9qe/2pzHsDtVGg+vBVrwWjBs+6c
Qup+h0qWLIhksVvArk9mo0ztKUnHYzMaZ2U8fppdu0lEvbRHC11ghZF/yqY2XJOL
fsJ2OZSPDmJp/RzikE93t4A0iJp59F4iC4M+rX270kIZWKYd/9NB06niMUBeb6mB
L9Hiooiu08pyldr4iYzvxMdj1DJ6JODtTfGaAYgFotY9kdVF/1uX2dANw39d6vLq
6QIDAQAB
-----END PUBLIC KEY-----

# ec.pub
-----BEGIN PUBLIC KEY-----
MFkwEwYHKoZIzj0CAQYIKoZIzj0DAQcDQgAEDU6Ne8Coeii0Tncf0sM0m0O+s7G/
SJCZkdeqgmIpxqZ+K/4KpovE9F9tRif0AzbSQHpHAh4PQzWPvv6l8xJUsA==
-----END PUBLIC KEY-----
~~~

You can compile the program against openssl with
~~~sh
gcc -o openssl_derpub prog.c -lssl -lcrypto -DUSE_OPENSSL
~~~

And against wolfssl with
~~~sh
gcc -o wolfssl_derpub prog.c -lwolfssl -DUSE_WOLFSSL
~~~

Ouput with openssl:
~~~sh
$ ./openssl_derpub rsa.pub
pubkey={30:82:01:22:30:0D:06:09:2A:86:48:86:F7:0D:01:01:01:05:00:03:82:01:0F:00:30:82:01:0A:02:82:01:01:00:F8:8F:8D:7F:5B:9A:79:97:FA:D3:DA:B6:9E:7E:0B:65:C6:7E:00:B8:99:71:67:C7:A3:A7:27:55:A4:0D:A4:4B:48:A5:D8:62:4D:30:3E:BE:BA:0F:60:BF:2D:F8:FC:B9:8F:D2:1C:50:81:BE:0E:3D:06:B1:B5:94:27:32:DB:77:E0:DC:19:26:F9:05:B4:B0:11:09:04:FB:1B:44:9F:76:02:82:C6:68:8B:0C:E1:5C:3C:5B:8F:DA:9E:FF:6A:73:1E:C0:ED:54:68:3E:BC:15:6B:C1:68:C1:B3:EE:9C:42:EA:7E:87:4A:96:2C:88:64:B1:5B:C0:AE:4F:66:A3:4C:ED:29:49:C7:63:33:1A:67:65:3C:7E:9A:5D:BB:49:44:BD:B4:47:0B:5D:60:85:91:7F:CA:A6:36:5C:93:8B:7E:C2:76:39:94:8F:0E:62:69:FD:1C:E2:90:4F:77:B7:80:34:88:9A:79:F4:5E:22:0B:83:3E:AD:7D:BB:D2:42:19:58:A6:1D:FF:D3:41:D3:A9:E2:31:40:5E:6F:A9:81:2F:D1:E2:A2:88:AE:D3:CA:72:95:DA:F8:89:8C:EF:C4:C7:63:D4:32:7A:24:E0:ED:4D:F1:9A:01:88:05:A2:D6:3D:91:D5:45:FF:5B:97:D9:D0:0D:C3:7F:5D:EA:F2:EA:E9:02:03:01:00:01:}

$ ./openssl_derpub ec.pub
pubkey={30:59:30:13:06:07:2A:86:48:CE:3D:02:01:06:08:2A:86:48:CE:3D:03:01:07:03:42:00:04:0D:4E:8D:7B:C0:A8:7A:28:B4:4E:77:1F:D2:C3:34:9B:43:BE:B3:B1:BF:48:90:99:91:D7:AA:82:62:29:C6:A6:7E:2B:FE:0A:A6:8B:C4:F4:5F:6D:46:27:F4:03:36:D2:40:7A:47:02:1E:0F:43:35:8F:BE:FE:A5:F3:12:54:B0:}
~~~

Output with wolfssl:
~~~sh
./wolfssl_derpub rsa.pub
pubkey={30:82:01:22:30:0D:06:09:2A:86:48:86:F7:0D:01:01:01:05:00:03:82:01:0F:00:30:82:01:0A:02:82:01:01:00:F8:8F:8D:7F:5B:9A:79:97:FA:D3:DA:B6:9E:7E:0B:65:C6:7E:00:B8:99:71:67:C7:A3:A7:27:55:A4:0D:A4:4B:48:A5:D8:62:4D:30:3E:BE:BA:0F:60:BF:2D:F8:FC:B9:8F:D2:1C:50:81:BE:0E:3D:06:B1:B5:94:27:32:DB:77:E0:DC:19:26:F9:05:B4:B0:11:09:04:FB:1B:44:9F:76:02:82:C6:68:8B:0C:E1:5C:3C:5B:8F:DA:9E:FF:6A:73:1E:C0:ED:54:68:3E:BC:15:6B:C1:68:C1:B3:EE:9C:42:EA:7E:87:4A:96:2C:88:64:B1:5B:C0:AE:4F:66:A3:4C:ED:29:49:C7:63:33:1A:67:65:3C:7E:9A:5D:BB:49:44:BD:B4:47:0B:5D:60:85:91:7F:CA:A6:36:5C:93:8B:7E:C2:76:39:94:8F:0E:62:69:FD:1C:E2:90:4F:77:B7:80:34:88:9A:79:F4:5E:22:0B:83:3E:AD:7D:BB:D2:42:19:58:A6:1D:FF:D3:41:D3:A9:E2:31:40:5E:6F:A9:81:2F:D1:E2:A2:88:AE:D3:CA:72:95:DA:F8:89:8C:EF:C4:C7:63:D4:32:7A:24:E0:ED:4D:F1:9A:01:88:05:A2:D6:3D:91:D5:45:FF:5B:97:D9:D0:0D:C3:7F:5D:EA:F2:EA:E9:02:03:01:00:01:}

./wolfssl_derpub ec.pub
pubkey={04:0D:4E:8D:7B:C0:A8:7A:28:B4:4E:77:1F:D2:C3:34:9B:43:BE:B3:B1:BF:48:90:99:91:D7:AA:82:62:29:C6:A6:7E:2B:FE:0A:A6:8B:C4:F4:5F:6D:46:27:F4:03:36:D2:40:7A:47:02:1E:0F:43:35:8F:BE:FE:A5:F3:12:54:B0:}
~~~

### Reproduction steps

1. `./configure --enable-all`

### Relevant log output

```shell

```

kareem-wolfssl, 2025-08-04:
Hi @ribes96 ,

Thanks for the detailed report, I'm working on reproducing this here and will let you know what I find.

Thanks,
Kareem

ribes96, 2025-08-05:
Yes, that fixes it for me. Thank you!