wolfSSL/wolfssl #9080: wc_SignatureGetSize always returns BAD_FUNC_ARG

ftommasi, 2025-08-08:
### Version

latest master branch (5.8.3)

### Description

I am fairly new to wolfssl, so please forgive me if I am missing something obvious. I am working on evaluating wolf and I ran into an issue with signing functionality. I tried my best to follow the API reference and even to mimic how the tests callthe functions, but I am always getting an error when trying to call `wc_SignatureGenerate`

I tracked it down to the fact that `wc_SignatureGetSize` is always returning `BAD_FUNC_ARG` no matter how I pass the arguments. I have put together the following that consistently reproduces the behavior

```c
void ecc_test() {
  ecc_key ecc;
  WC_RNG rng;

  int status = wc_InitRng(&rng);
  status = wc_ecc_init(&ecc);

  status = wc_ecc_make_key(&rng, 32, &ecc);

  // always fails
  int siglen = wc_SignatureGetSize(WC_SIGNATURE_TYPE_ECC, &ecc, sizeof(ecc));
  byte *signature = (byte *)malloc(siglen);
  assert(siglen > 0); // always returns -173 AKA BAD_FUNC_ARG
                      //
  const char *message = "Sign me\0";
  status = wc_SignatureGenerate(
      WC_HASH_TYPE_SHA256, WC_SIGNATURE_TYPE_ECC, (const byte *)message[0],
      strlen(message), signature, (word32 *)&siglen, &ecc, sizeof(ecc), &rng);
  assert(status == 0);
}

```

I configure the library with 
```bash
./configure --enable-ecc --enable-all --enable-shared
```
And then I run `make -f std_t_static.mk` 

The contents of the makefile are below

```Makefile
WOLFSSL_ROOT ?= $(shell readlink -f .)

Wolfssl_C_Files :=$(WOLFSSL_ROOT)/wolfcrypt/src/aes.c\
                    $(WOLFSSL_ROOT)/wolfcrypt/src/arc4.c\
                    $(WOLFSSL_ROOT)/wolfcrypt/src/asn.c\
                    $(WOLFSSL_ROOT)/wolfcrypt/src/blake2b.c\
                    $(WOLFSSL_ROOT)/wolfcrypt/src/camellia.c\
                    $(WOLFSSL_ROOT)/wolfcrypt/src/coding.c\
                    $(WOLFSSL_ROOT)/wolfcrypt/src/chacha.c\
                    $(WOLFSSL_ROOT)/wolfcrypt/src/chacha20_poly1305.c\
                    $(WOLFSSL_ROOT)/src/crl.c\
                    $(WOLFSSL_ROOT)/wolfcrypt/src/des3.c\
                    $(WOLFSSL_ROOT)/wolfcrypt/src/dh.c\
                    $(WOLFSSL_ROOT)/wolfcrypt/src/tfm.c\
                    $(WOLFSSL_ROOT)/wolfcrypt/src/ecc.c\
                    $(WOLFSSL_ROOT)/wolfcrypt/src/error.c\
                    $(WOLFSSL_ROOT)/wolfcrypt/src/hash.c\
                    $(WOLFSSL_ROOT)/wolfcrypt/src/kdf.c\
                    $(WOLFSSL_ROOT)/wolfcrypt/src/hmac.c\
                    $(WOLFSSL_ROOT)/wolfcrypt/src/integer.c\
                    $(WOLFSSL_ROOT)/src/internal.c\
                    $(WOLFSSL_ROOT)/src/wolfio.c\
                    $(WOLFSSL_ROOT)/src/keys.c\
                    $(WOLFSSL_ROOT)/wolfcrypt/src/logging.c\
                    $(WOLFSSL_ROOT)/wolfcrypt/src/md4.c\
                    $(WOLFSSL_ROOT)/wolfcrypt/src/md5.c\
                    $(WOLFSSL_ROOT)/wolfcrypt/src/memory.c\
                    $(WOLFSSL_ROOT)/src/ocsp.c\
                    $(WOLFSSL_ROOT)/wolfcrypt/src/pkcs7.c\
                    $(WOLFSSL_ROOT)/wolfcrypt/src/pkcs12.c\
                    $(WOLFSSL_ROOT)/wolfcrypt/src/poly1305.c\
                    $(WOLFSSL_ROOT)/wolfcrypt/src/wc_port.c\
                    $(WOLFSSL_ROOT)/wolfcrypt/src/wolfmath.c\
                    $(WOLFSSL_ROOT)/wolfcrypt/src/pwdbased.c\
                    $(WOLFSSL_ROOT)/wolfcrypt/src/random.c\
                    $(WOLFSSL_ROOT)/wolfcrypt/src/ripemd.c\
                    $(WOLFSSL_ROOT)/wolfcrypt/src/rsa.c\
                    $(WOLFSSL_ROOT)/wolfcrypt/src/dsa.c\
                    $(WOLFSSL_ROOT)/wolfcrypt/src/sha.c\
                    $(WOLFSSL_ROOT)/wolfcrypt/src/sha256.c\
                    $(WOLFSSL_ROOT)/wolfcrypt/src/sha512.c\
                    $(WOLFSSL_ROOT)/wolfcrypt/src/signature.c\
                    $(WOLFSSL_ROOT)/wolfcrypt/src/sp_c32.c\
                    $(WOLFSSL_ROOT)/wolfcrypt/src/sp_c64.c\
                    $(WOLFSSL_ROOT)/wolfcrypt/src/sp_int.c\
                    $(WOLFSSL_ROOT)/src/ssl.c\
                    $(WOLFSSL_ROOT)/src/tls.c\
                    $(WOLFSSL_ROOT)/wolfcrypt/src/wc_encrypt.c\
                    $(WOLFSSL_ROOT)/wolfcrypt/src/wolfevent.c\

Wolfssl_Include_Paths := -I$(WOLFSSL_ROOT)/ \
                         -I$(WOLFSSL_ROOT)/wolfcrypt/

Flags_Just_For_C := -Wno-implicit-function-declaration -std=c99
Common_C_Cpp_Flags := $(Wolfssl_Include_Paths) -fPIC -fpie -DHAVE_ECC -I.

Wolfssl_C_Flags := $(Flags_Just_For_C) $(Common_C_Cpp_Flags) $(Wolfssl_C_Extra_Flags)

Wolfssl_C_Objects := $(Wolfssl_C_Files:.c=.o)

override CFLAGS += $(Wolfssl_C_Flags)

.PHONY: all run

all: libwolfssl.stdlstatic.lib.a

libwolfssl.stdlstatic.lib.a: $(Wolfssl_C_Objects)
	ar rcs libwolfssl.std.static.lib.a $(Wolfssl_C_Objects)
	@echo "LINK => $@"

clean:
	@rm -f libwolfssl.stdlstatic.lib.a $(Wolfssl_C_Objects)

```

With the static library built I copy it into a lib directory and then go to build and run my test application with the following `make test` contents below

```Makefile
build:
	g++ -g test.c -std=c++11 -I./wolfssl/ -L./wolfssl/lib -lwolfssl.std.static.lib -DHAVE_ECC -DUSE_FAST_MATH -lpthread -o wolf_test

test: build
	./wolf_test

```

I also tried with `RsaKey` and go the same result

Please let me know if I am using something incorrectly or missing something. Any help or insight would be very helpful. Thanks in advance!

dgarske, 2025-08-08:
Hi @ftommasi ,

Yes that appears to be the issue. You must make sure the build options used for building wolfSSL are also defined when building your application. The ./configure generates a wolfssl/options.h that contains these. Instead of supplying some of them as -D CLAGS just include wolfssl/options.h.

Thanks,
David Garske, wolfSSL

ftommasi, 2025-08-08:
@dgarske Thanks! including wolfssl/options.h is the way to go! That solved my issue. I knew it was something silly so I appreciate the quick response and patience 😄

dgarske, 2025-08-08:
Can you tell us a bit about your project? If you'd like to keep it private you can email support at wolfssl dot com. Thanks!