wolfSSL/wolfssl #9086: [Bug]: Questions about wolfssl verify

onepeople158, 2025-08-11:
### Version

I am using version 0.1.7 of the wolfssl Command Line Utility. Linked to wolfSSL version 5.7.6

### Description

Hello developer, I'm not sure if wolfssl has a command for CRL revocation checks. When I perform a revocation check using openssl, openssl indicates that the certificate has been revoked, but wolfssl does not display any information related to the certificate being revoked.

command:
wolfssl verify -CAfile root_cert_1.pem -CRLfile crl_revokedCert_02.pem ca1.pem

test cases:

[isRevoked.zip](https://github.com/user-attachments/files/21714002/isRevoked.zip)

### Relevant log output

```shell

```

kareem-wolfssl, 2025-08-12:
Hello @onepeople158 ,

How are you building wolfSSL?  You will need to pass in `--enable-crl` to enable CRL support.
Please use wolfSSL 5.8.2 with wolfCLU master, we have had significant changes to both since 0.1.7/5.7.6.

onepeople158, 2025-08-26:
> Hello [@onepeople158](https://github.com/onepeople158) ,
> 
> How are you building wolfSSL? You will need to pass in to enable CRL support. Please use wolfSSL 5.8.2 with wolfCLU master, we have had significant changes to both since 0.1.7/5.7.6.`--enable-crl`

Dear Developers,

Hello. I have installed the latest version of wolfSSL.

The following is the displayed result:
You are using version 0.1.8 of the wolfssl Command Line Utility.
Linked to wolfSSL version 5.8.2

I ran the command below and am not sure if wolfSSL supports CRL revocation checking:
wolfssl verify -CAfile root_cert_1.pem -CRLfile crl_revokedCert_02.pem ca1.pem

The command output was:
verifying certificate file ca1.pem
using CA file root_cert_1.pem
OK

holtrop, 2025-08-26:
Hello @onepeople158, I will take a look at this. I see your test case files in the isRevoked.zip file. Can you include the openssl verify commands you are using for comparison?

miyazakh, 2026-01-09:
Hi @onepeople158 

It seems that there are multiple reasons why this does not work as expected.

1. `wolfCLU` doesn't support `-CRLfile` option yet.
2. Parsing CRL fails when CRL number is large.
3. Error code mapping issues.

For `#1`, could you concatenate the CRL and the CA certificate, and then use the result as the `-CAfile` argument, as shown below?
```
cat your_crl.pem >> crl_rootca.pem
cat root_cert_1.pem >> crl_rootca.pem
./wolfssl verify -CAfile crl_rootca.pem -crl_check ca1.pem
```
For `#2`, I have submitted a fix as [PR#968](https://github.com/wolfSSL/wolfssl/pull/9628). In the meantime, could you try configuring wolfSSL with `CFLAGS="-DCRL_MAX_NUM_SZ=40"`?

For `#3`, I am working on it.

onepeople158, 2026-01-09:
> Hi [@onepeople158](https://github.com/onepeople158)
> 
> It seems that there are multiple reasons why this does not work as expected.
> 
> 1. `wolfCLU` doesn't support  option yet.`-CRLfile`
> 2. Parsing CRL fails when CRL number is large.
> 3. Error code mapping issues.
> 
> For , could you concatenate the CRL and the CA certificate, and then use the result as the argument, as shown below?`#1``-CAfile`
> 
> ```
> cat your_crl.pem >> crl_rootca.pem
> cat root_cert_1.pem >> crl_rootca.pem
> ./wolfssl verify -CAfile crl_rootca.pem -crl_check ca1.pem
> ```
> 
> For , I have submitted a fix as [PR#968](https://github.com/wolfSSL/wolfssl/pull/9628). In the meantime, could you try configuring wolfSSL with ?`#2``CFLAGS="-DCRL_MAX_NUM_SZ=40"`
> 
> For , I am working on it.`#3`

```
cat crl_revokedCert_02.pem root_cert_1.pem > crl_rootca.pem && wolfssl verify -CAfile crl_rootca.pem -crl_check ca1.pem
verifying certificate file ca1.pem
using CA file crl_rootca.pem
Verification Failed
Err (-362): CRL missing, not loaded
Error returned: -1.

```

miyazakh, 2026-01-09:
@onepeople158 
Did you recompile wolfSSL with `CFLAGS=-DCRL_MAX_NUM_SZ=40` because your CRL number is large, and rebuild wolfCLU accordingly?