wolfSSL/wolfssl #9128: [Bug]: make check fails when configuring with CFLAGS=-DWOLFSSL_CHECK_MEM_ZERO

effbiae, 2025-08-24:
### Contact Details

effbiae@gmail.com

### Version

5.8.2 (commit 6f567bbca2bbe981ba48a698d578b5608318e875)

### Description

```sh
$ ./configure --enable-all CFLAGS="-DWOLFSSL_CHECK_MEM_ZERO" && make check
[snip]
FAIL: scripts/unit.test
[snip]
```

### Reproduction steps

./configure --enable-all CFLAGS="-DWOLFSSL_CHECK_MEM_ZERO" && make check

### Relevant log output

```shell
=====================================
   wolfssl 5.8.2: ./test-suite.log
=====================================

# TOTAL: 18
# PASS:  14
# SKIP:  3
# XFAIL: 0
# FAIL:  1
# XPASS: 0
# ERROR: 0
```

effbiae, 2025-08-25:
running git bisect,
```
$ cat bi.sh 
sh autogen.sh && ./configure --enable-all CFLAGS="-DWOLFSSL_CHECK_MEM_ZERO" && make check
$ git bisect start HEAD v5.8.0-stable --
$ git bisect run sh bi.sh
..
0d532cc3f22cabb25abcc2066d9c95d6add21283 is the first bad commit
commit 0d532cc3f22cabb25abcc2066d9c95d6add21283
Author: Juliusz Sosinowicz <juliusz@wolfssl.com>
Date:   Thu Aug 7 14:12:42 2025 +0200

    Test DTLS replay protection
...
```
which introduces a new test `test_dtls_replay`.
```
$ `./configure --enable-all --enable-debug --enable-debug-trace-errcodes CFLAGS="-DWOLFSSL_CHECK_MEM_ZERO" && make check`
$ tail test-suite.log 
 passed (  0.00512)
   1151: test_wolfSSL_dtls_cid_parse                        : passed (  0.00000)
   1152: test_dtls13_epochs                                 : passed (  0.00003)
   1153: test_dtls_rtx_across_epoch_change                  : passed (  0.00143)
   1154: test_dtls_drop_client_ack                          : passed (  0.00141)
   1155: test_dtls_replay                                   :
[MEM_ZERO] TLS 1.3 Decrypted data:0x5df037e69a88 + 0 is not zero
[MEM_ZERO] Checking 0x5df037e69a80:1916
FAIL scripts/unit.test (exit status: 134)
```
note a MEM_ZERO failure in the log (unclear if the `test_dtls_replay` test passed or not) but there are earlier failures:
```
$ grep ^ERROR test-suite.log -A 2
ERROR - tests/api.c line 9457 failed with:
    expected: wolfSSL_read_ex(ssl_s, buf, sizeof(buf), &count) == WOLFSSL_SUCCESS
    result:   5 != 1
--
ERROR - tests/api.c line 49037 failed with:
    expected: wolfSSL_shutdown(ssl_c) == 1
    result:   0 != 1
```
these failures also occur in the latest commit with `-DWOLFSSL_CHECK_MEM_ZERO`

rlm2002, 2025-08-27:
Hi @effbiae 

Thanks again for the report. Please check out PR #9146 to address this issue. It is a `ForceZero()` modification when calling `ShrinkInputBuffer()`. Let me know if it works for you!

effbiae, 2025-08-28:
Thanks @rlm2002! Works for me. Closing...

closed 2025-08-28 by effbiae