wolfSSL/wolfssl #9199: [Bug]: TLS 1.3 client handshake fail with `BAD_FUNC_ARG` (-173)

ycdev1, 2025-09-14:
### Version

5.8.2

### Description

I upgraded wolfssl using vcpkg today to 5.8.2 which supports TLS 1.3 and it was selected automatically. 
But I found that my https tests are now failing with wolfssl and returning: Bad function argument.

After some debugging and inspecting the source codes to know where the error is actually happening, I found It's resulting from `wolfSSL_negotiate` and digging more I found it is this line:
https://github.com/wolfSSL/wolfssl/blob/dc421a0d4ce9c2b4f92509bb9f1a6b7d14927c78/wolfcrypt/src/hpke.c#L244

The reason is that the passed `kem` parameter is 32 and it didn't match any switch branch.

The call stack is as follows:
inside `wolfSSL_negotiate`:
https://github.com/wolfSSL/wolfssl/blob/dc421a0d4ce9c2b4f92509bb9f1a6b7d14927c78/src/ssl.c#L2664
then inside `wolfSSL_connect_TLSv13`:
https://github.com/wolfSSL/wolfssl/blob/dc421a0d4ce9c2b4f92509bb9f1a6b7d14927c78/src/tls13.c#L13304
then inside `SendTls13ClientHello`:
https://github.com/wolfSSL/wolfssl/blob/dc421a0d4ce9c2b4f92509bb9f1a6b7d14927c78/src/tls13.c#L4690
then inside `TLSX_WriteRequest`:
https://github.com/wolfSSL/wolfssl/blob/dc421a0d4ce9c2b4f92509bb9f1a6b7d14927c78/src/tls.c#L15530
then inside `TLSX_Write`:
https://github.com/wolfSSL/wolfssl/blob/dc421a0d4ce9c2b4f92509bb9f1a6b7d14927c78/src/tls.c#L14272
then inside `TLSX_ECH_Write`:
https://github.com/wolfSSL/wolfssl/blob/dc421a0d4ce9c2b4f92509bb9f1a6b7d14927c78/src/tls.c#L13197C13-L13197C72
and finally, in `wc_HpkeInit`:
https://github.com/wolfSSL/wolfssl/blob/dc421a0d4ce9c2b4f92509bb9f1a6b7d14927c78/wolfcrypt/src/hpke.c#L244

I switched back to TLS 1.2 method `wolfTLSv1_2_client_method` and the tests passed again.
The current sources on the main repo are a bit different from the last vcpkg port so I don't know whether this problem was found and solved or not.

Also, I'm making the tcp connection manually and using `wolfSSL_SetIOReadCtx` and `wolfSSL_SetIOWriteCtx` to manage read and write but I think this shouldn't be a problem because:
1- The same is used with previous versions of TLS and working.
2- The handshake fails before even calling my read and write implementations.
3- The error `BAD_FUNC_ARG` means is a logic error which means unexpected value for argument was passed.

Are there additional setup steps for TLS 1.3 that I'm missing?  The problem prevents any attempt to handshake and should have been easily caught during tests, so may be the problem is in my code?

### Relevant log output

```shell

```

ycdev1, 2025-09-14:
I'm trying to connect to github and google DOH.

embhorn, 2025-09-15:
Hello @ycdev1 

Thanks for contacting wolfSSL Support. Could you please enable debug logging and share the log showing the error?
`#define DEBUG_WOLFSSL`

Thanks,
@embhorn - wolfSSL Support

ycdev1, 2025-09-15:
The logging needs to be enabled at runtime!
Here is the logging:

```
wolfSSL Entering wolfSSL_new
wolfSSL Entering ReinitSSL
RNG_HEALTH_TEST_CHECK_SIZE = 128
sizeof(seedB_data)         = 128
wolfSSL Entering SetSSL_CTX
wolfSSL Entering wolfSSL_NewSession
InitSSL done. return 0 (success)
wolfSSL_new InitSSL success
wolfSSL Leaving wolfSSL_new InitSSL =, return 0
wolfSSL Entering wolfSSL_get_SSL_CTX
wolfSSL Entering wolfSSL_new
wolfSSL Entering ReinitSSL
RNG_HEALTH_TEST_CHECK_SIZE = 128
sizeof(seedB_data)         = 128
wolfSSL Entering SetSSL_CTX
wolfSSL Entering wolfSSL_NewSession
InitSSL done. return 0 (success)
wolfSSL_new InitSSL success
wolfSSL Leaving wolfSSL_new InitSSL =, return 0
wolfSSL Entering wolfSSL_free
Free SSL: 00000279545F43F0
Free'ing client ssl
wolfSSL Entering ClientSessionToSession
wolfSSL Entering wolfSSL_FreeSession
wolfSSL_FreeSession full free
wolfSSL Entering wolfSSL_sk_CIPHER_free
wolfSSL Entering wolfSSL_sk_free
wolfSSL Entering wolfSSL_sk_X509_pop_free
wolfSSL Entering wolfSSL_sk_pop_free
wolfSSL Entering wolfSSL_sk_X509_pop_free
wolfSSL Entering wolfSSL_sk_pop_free
wolfSSL Entering wolfSSL_sk_X509_NAME_pop_free
wolfSSL Entering wolfSSL_sk_pop_free
CTX ref count not 0 yet, no free
wolfSSL Leaving wolfSSL_free, return 0
wolfSSL Entering wolfSSL_set_connect_state
wolfSSL Entering wolfSSL_negotiate
wolfSSL Entering wolfSSL_connect_TLSv13
wolfSSL Entering ReinitSSL
wolfSSL Entering RetrySendAlert
wolfSSL Entering SendTls13ClientHello
Adding signature algorithms extension
Adding supported versions extension
wolfSSL Entering EccMakeKey
wolfSSL Leaving EccMakeKey, return 0
growing output buffer
ECH extension to write
TLSX_ECH_Write
wolfSSL error occurred, error = -173
wolfSSL Leaving wolfSSL_negotiate, return -1
wolfSSL Entering wolfSSL_get_error
wolfSSL Leaving wolfSSL_get_error, return -173
```

ycdev1, 2025-09-15:
Also forgot to mention that I'm setting verify mode to none using:
`wolfSSL_CTX_set_ex_data(ctx, 0, nullptr);`
does this affect the TLS 1.3 path since verification is mandatory?

ycdev1, 2025-09-15:
I made a minimal example reproducing the problem:

```
#define WOLFSSL_NO_OPTIONS_H
// also tried with user_settings.h
#include <wolfssl/ssl.h>
#ifdef _WIN32
#define WIN32_LEAN_AND_MEAN
#include <windows.h>
#include <winsock2.h>
#include <ws2tcpip.h>
#else
#include <sys/types.h>
#include <arpa/inet.h>
#include <sys/socket.h>
#include <netdb.h>
#endif // _WIN32
#include <iostream>

namespace {
	void close_socket(int sockfd) noexcept {
		if (sockfd != -1) {
#ifdef _WIN32
			::closesocket(sockfd);
#else
			::close(sockfd);
#endif // _WIN32
		}
	}

	int last_socket_error() noexcept {
#ifdef _WIN32
		return ::WSAGetLastError();
#else
		return errno;
#endif // _WIN32
	}

	char wolfssl_error_buff[1024];

	const char* get_wolfssl_error_msg(int c) {
		std::memset(wolfssl_error_buff, 0, sizeof(wolfssl_error_buff));
		wolfSSL_ERR_error_string_n(c, wolfssl_error_buff, sizeof(wolfssl_error_buff) - 1);
		return wolfssl_error_buff;
	}

	struct clean_on_exit {
		WOLFSSL_CTX* ctx = nullptr;
		WOLFSSL* ssl = nullptr;
		int sockfd = -1;

		~clean_on_exit() {
			if (ssl != nullptr) {
				::wolfSSL_Free(ssl);
			}
			if (ctx != nullptr) {
				::wolfSSL_CTX_free(ctx);
			}
			close_socket(sockfd);
			sockfd = -1;
			wolfSSL_Debugging_OFF();
			wolfSSL_Cleanup();
		}
	};

	int connect_socket(const char* domain_name) {
		int sockfd = static_cast<int>(::socket(AF_INET, SOCK_STREAM, 0));
		if (sockfd == -1) {
			return -1;
		}

		struct addrinfo hint{};
		hint.ai_family = AF_INET;
		hint.ai_socktype = SOCK_STREAM;

		struct addrinfo* results = nullptr;
		const int dns_res = ::getaddrinfo(domain_name, "https", &hint, &results);
		if (dns_res != 0 || results == nullptr) {
			std::cout << "getaddrinfo failed with error: " << dns_res << " !\n";
			close_socket(sockfd);
			return -1;
		}

		bool connected = false;
		for (struct addrinfo* p = results; p != nullptr; p = p->ai_next) {
			char buff[100];
			std::memset(buff, 0, sizeof(buff));
			const char* str = ::inet_ntop(p->ai_family, &(((sockaddr_in*)(p->ai_addr))->sin_addr),
				buff, sizeof(buff) - 1);
			if (str != nullptr) {
				std::cout << "trying to connect to " << domain_name << " (" << buff << ") ...\n";
			}
			const int connect_res = ::connect(sockfd, p->ai_addr, static_cast<int>(p->ai_addrlen));
			if (connect_res == 0) {
				std::cout << "connected to " << domain_name << " (" << buff << ")\n";
				connected = true;
				break;
			}
			else {
				std::cout << "failed to connect to " << domain_name << " (" << buff << ")"
					" with error: " << last_socket_error() << "\n";
			}
		}

		::freeaddrinfo(results);
		if (!connected) {
			std::cout << "couldn't connect to " << domain_name << " !\n";
			close_socket(sockfd);
			return -1;
		}

		return sockfd;
	}

	bool do_ssl_handshake(const char* domain_name, bool use_tls13) {
		std::cout << " ========================= \n";
		wolfSSL_Init();
		wolfSSL_Debugging_ON();
		clean_on_exit on_exit;
		on_exit.sockfd = connect_socket(domain_name);
		if (on_exit.sockfd == -1) {
			return false;
		}
		WOLFSSL_METHOD* method = use_tls13 ? wolfTLS_client_method() 
			: wolfTLSv1_2_client_method();
		on_exit.ctx = wolfSSL_CTX_new(method);
		if (on_exit.ctx == nullptr) {
			std::cout << "failed to create a new wolfssl context !\n";
			return false;
		}

		wolfSSL_CTX_set_verify(on_exit.ctx, 0, nullptr);

		on_exit.ssl = wolfSSL_new(on_exit.ctx);
		if (on_exit.ssl == nullptr) {
			std::cout << "failed to create a new wolfssl session !\n";
			return false;
		}

		wolfSSL_set_fd(on_exit.ssl, on_exit.sockfd);
		wolfSSL_UseSNI(on_exit.ssl, WOLFSSL_SNI_HOST_NAME, domain_name, 
			static_cast<unsigned short>(strlen(domain_name)));

		const int connect_ret = wolfSSL_connect(on_exit.ssl);
		wolfSSL_Debugging_OFF();
		if (connect_ret != SSL_SUCCESS) {
			// fails here when use_tls13 is true!
			const char* error_msg = get_wolfssl_error_msg(wolfSSL_get_error(on_exit.ssl, connect_ret));
			std::cout << "failed to do handshake : " << error_msg << "\n";
			return false;
		}
		else {
			const char* version = wolfSSL_get_version(on_exit.ssl);
			std::cout << "done the handshake, protocol version: " << version << "\n";
			wolfSSL_shutdown(on_exit.ssl);
			return true;
		}
	}
}

namespace tests_fn {
	bool do_wolfssl_client_test() {
		const bool res1 = do_ssl_handshake("github.com", true);
		const bool res2 = do_ssl_handshake("github.com", false);
		return res1 && res2;
	}
}

```

The logs for the failed case:

```
=========================
trying to connect to github.com (140.82.121.4) ...
connected to github.com (140.82.121.4)
wolfSSL Entering TLS_client_method_ex
wolfSSL Entering wolfSSL_CTX_new_ex
wolfSSL Entering wolfSSL_CertManagerNew
heap param is null
DYNAMIC_TYPE_CERT_MANAGER Allocating = 256 bytes
wolfSSL Leaving wolfSSL_CTX_new_ex, return 0
wolfSSL Entering wolfSSL_CTX_set_verify
wolfSSL Entering wolfSSL_new
wolfSSL Entering ReinitSSL
RNG_HEALTH_TEST_CHECK_SIZE = 128
sizeof(seedB_data)         = 128
wolfSSL Entering SetSSL_CTX
wolfSSL Entering wolfSSL_NewSession
InitSSL done. return 0 (success)
wolfSSL_new InitSSL success
wolfSSL Leaving wolfSSL_new InitSSL =, return 0
wolfSSL Entering wolfSSL_set_fd
wolfSSL Entering wolfSSL_set_read_fd
wolfSSL Leaving wolfSSL_set_read_fd, return 1
wolfSSL Entering wolfSSL_set_write_fd
wolfSSL Leaving wolfSSL_set_write_fd, return 1
TLS 1.2 or lower
wolfSSL Entering wolfSSL_connect
wolfSSL Entering ReinitSSL
wolfSSL Entering RetrySendAlert
wolfSSL Entering SendTls13ClientHello
Adding signature algorithms extension
Adding supported versions extension
wolfSSL Entering EccMakeKey
wolfSSL Leaving EccMakeKey, return 0
growing output buffer
ECH extension to write
TLSX_ECH_Write
wolfSSL error occurred, error = -173
failed to do handshake : Bad function argument
```

embhorn, 2025-09-15:
Could you please share the `user_settings.h` you are using to configure the library?

ycdev1, 2025-09-15:
I'm using the vcpkg port installed with this commad:
`vcpkg install wolfssl:x64-windows-static`
the port version is 5.8.2
I tried using the `user_settings.h` file in `IDE\WIN10`  path.
I'm going to try the main branch next, may be this is a bug in the vcpkg port?

ycdev1, 2025-09-15:
Here the settings file:

```
#ifndef _WIN_USER_SETTINGS_H_
#define _WIN_USER_SETTINGS_H_

/* For FIPS 140-2 3389 build set to "#if 1" */
#if 0
#undef HAVE_FIPS
#define HAVE_FIPS
#undef HAVE_FIPS_VERSION
#define HAVE_FIPS_VERSION 2
#undef HAVE_FIPS_VERSION_MINOR
#define HAVE_FIPS_VERSION_MINOR 0
#endif

/* Set the following to 1 for WCv5.0-RC12 build. */
#if 0
#undef HAVE_FIPS
#define HAVE_FIPS
#undef HAVE_FIPS_VERSION
#define HAVE_FIPS_VERSION 5
#undef HAVE_FIPS_VERSION_MINOR
#define HAVE_FIPS_VERSION_MINOR 2
#endif

/* For FIPS Ready, uncomment the following: */
/* #define WOLFSSL_FIPS_READY */
#ifdef WOLFSSL_FIPS_READY
    #undef HAVE_FIPS
    #define HAVE_FIPS
    #undef HAVE_FIPS_VERSION
    #define HAVE_FIPS_VERSION 5
    #undef HAVE_FIPS_VERSION_MINOR
    #define HAVE_FIPS_VERSION_MINOR 3
#endif

/* Verify this is Windows */
#ifndef _WIN32
#error This user_settings.h header is only designed for Windows
#endif

/* Configurations */
#if defined(HAVE_FIPS)
    /* FIPS */
    #define OPENSSL_EXTRA
    #define HAVE_THREAD_LS
    #define WOLFSSL_KEY_GEN
    #define HAVE_AESGCM
    #define HAVE_HASHDRBG
    #define WOLFSSL_SHA384
    #define WOLFSSL_SHA512
    #define NO_PSK
    #define NO_RC4
    #define NO_DSA
    #define NO_MD4

    #if defined(HAVE_FIPS_VERSION) && (HAVE_FIPS_VERSION >= 2)
        #define WOLFSSL_SHA224
        #define WOLFSSL_SHA3
        #define WC_RSA_PSS
        #define WC_RSA_NO_PADDING
        #define HAVE_ECC
        #define HAVE_ECC384
        #define HAVE_ECC521
        #define HAVE_SUPPORTED_CURVES
        #define HAVE_TLS_EXTENSIONS
        #define ECC_SHAMIR
        #define HAVE_ECC_CDH
        #define ECC_TIMING_RESISTANT
        #define TFM_TIMING_RESISTANT
        #define WOLFSSL_AES_COUNTER
        #define WOLFSSL_AES_DIRECT
        #define HAVE_AES_ECB
        #define HAVE_AESCCM
        #define WOLFSSL_CMAC
        #define HAVE_HKDF
        #define WOLFSSL_VALIDATE_ECC_IMPORT
        #define WOLFSSL_VALIDATE_FFC_IMPORT
        #define HAVE_FFDHE_Q
        #define HAVE_PUBLIC_FFDHE
    #ifdef _WIN64
        #define WOLFSSL_AESNI
        #define HAVE_INTEL_RDSEED
    #endif
        #define FORCE_FAILURE_RDSEED
    #endif /* FIPS v2 */
    #if defined(HAVE_FIPS_VERSION) && (HAVE_FIPS_VERSION >= 5)
        #undef WOLFSSL_AESNI /* Comment out if using PAA */
        #undef HAVE_INTEL_RDSEED
        #undef FORCE_FAILURE_RDSEED
        #undef HAVE_PUBLIC_FFDHE

        #define NO_DES
        #define NO_DES3
        #define NO_MD5
        #define NO_OLD_TLS

        #define WOLFSSL_TLS13
        #define HAVE_TLS_EXTENSIONS
        #define HAVE_SUPPORTED_CURVES
        #define GCM_TABLE_4BIT
        #define WOLFSSL_NO_SHAKE256
        #define WOLFSSL_VALIDATE_ECC_KEYGEN
        #define WOLFSSL_ECDSA_SET_K
        #define WOLFSSL_WOLFSSH
        #define WOLFSSL_PUBLIC_MP
        #define WC_RNG_SEED_CB
        #define TFM_ECC256
        #define ECC_USER_CURVES
        #define HAVE_ECC192
        #define HAVE_ECC224
        #define HAVE_ECC256
        #define HAVE_ECC384
        #define HAVE_ECC521
        #define HAVE_FFDHE_2048
        #define HAVE_FFDHE_3072
        #define HAVE_FFDHE_4096
        #define HAVE_FFDHE_6144
        #define HAVE_FFDHE_8192
        #define WOLFSSL_AES_OFB
        #define FP_MAX_BITS 16384
    #endif /* FIPS v5 */
    #if defined(HAVE_FIPS_VERSION) && (HAVE_FIPS_VERSION >= 6)
        #define WOLFSSL_AES_XTS
    #endif
#else
    /* Enables blinding mode, to prevent timing attacks */
    #define WC_RSA_BLINDING

    #if defined(WOLFSSL_LIB)
        /* The lib */
        #define OPENSSL_EXTRA
        #define WOLFSSL_RIPEMD
        #define NO_PSK
        #define HAVE_EXTENDED_MASTER
        #define WOLFSSL_SNIFFER
        #define HAVE_SECURE_RENEGOTIATION

        #define HAVE_AESGCM
        #define WOLFSSL_AES_XTS
        #define WOLFSSL_SHA384
        #define WOLFSSL_SHA512

        #define HAVE_SUPPORTED_CURVES
        #define HAVE_TLS_EXTENSIONS

        #define HAVE_ECC
        #define ECC_SHAMIR
        #define ECC_TIMING_RESISTANT
    #else
        /* The servers and clients */
        #define OPENSSL_EXTRA
        #define NO_PSK
    #endif
#endif /* HAVE_FIPS */

#endif /* _WIN_USER_SETTINGS_H_ */

```

embhorn, 2025-09-15:
Hang on... It's been a while since I used the vcpkg install. The vcpkg installer actually puts the wolfssl config into `<vcpkg>/installed/x64-windows/include/wolfssl/options.h`. Here is the one I generated:
[options.h.txt](https://github.com/user-attachments/files/22351924/options.h.txt)

It does have TLS 1.3 enabled by default.

embhorn, 2025-09-15:
So your application should be including that file. Also remove the `#define WOLFSSL_NO_OPTIONS_H`

ycdev1, 2025-09-15:
Changed the code with options file you sent and still the same issue:

```
#include "wolfssl_options.h"
#include <wolfssl/ssl.h>
#ifdef _WIN32
#define WIN32_LEAN_AND_MEAN
#include <windows.h>
#include <winsock2.h>
#include <ws2tcpip.h>
#else
#include <sys/types.h>
#include <arpa/inet.h>
#include <sys/socket.h>
#include <netdb.h>
#endif // _WIN32
#include <iostream>

namespace {
	void close_socket(int sockfd) noexcept {
		if (sockfd != -1) {
#ifdef _WIN32
			::closesocket(sockfd);
#else
			::close(sockfd);
#endif // _WIN32
		}
	}

	int last_socket_error() noexcept {
#ifdef _WIN32
		return ::WSAGetLastError();
#else
		return errno;
#endif // _WIN32
	}

	char wolfssl_error_buff[1024];

	const char* get_wolfssl_error_msg(int c) {
		std::memset(wolfssl_error_buff, 0, sizeof(wolfssl_error_buff));
		wolfSSL_ERR_error_string_n(c, wolfssl_error_buff, sizeof(wolfssl_error_buff) - 1);
		return wolfssl_error_buff;
	}

	struct clean_on_exit {
		WOLFSSL_CTX* ctx = nullptr;
		WOLFSSL* ssl = nullptr;
		int sockfd = -1;

		~clean_on_exit() {
			if (ssl != nullptr) {
				::wolfSSL_Free(ssl);
			}
			if (ctx != nullptr) {
				::wolfSSL_CTX_free(ctx);
			}
			close_socket(sockfd);
			sockfd = -1;
			wolfSSL_Debugging_OFF();
			wolfSSL_Cleanup();
		}
	};

	int connect_socket(const char* domain_name) {
		int sockfd = static_cast<int>(::socket(AF_INET, SOCK_STREAM, 0));
		if (sockfd == -1) {
			return -1;
		}

		struct addrinfo hint{};
		hint.ai_family = AF_INET;
		hint.ai_socktype = SOCK_STREAM;

		struct addrinfo* results = nullptr;
		const int dns_res = ::getaddrinfo(domain_name, "https", &hint, &results);
		if (dns_res != 0 || results == nullptr) {
			std::cout << "getaddrinfo failed with error: " << dns_res << " !\n";
			close_socket(sockfd);
			return -1;
		}

		bool connected = false;
		for (struct addrinfo* p = results; p != nullptr; p = p->ai_next) {
			char buff[100];
			std::memset(buff, 0, sizeof(buff));
			const char* str = ::inet_ntop(p->ai_family, &(((sockaddr_in*)(p->ai_addr))->sin_addr),
				buff, sizeof(buff) - 1);
			if (str != nullptr) {
				std::cout << "trying to connect to " << domain_name << " (" << buff << ") ...\n";
			}
			const int connect_res = ::connect(sockfd, p->ai_addr, static_cast<int>(p->ai_addrlen));
			if (connect_res == 0) {
				std::cout << "connected to " << domain_name << " (" << buff << ")\n";
				connected = true;
				break;
			}
			else {
				std::cout << "failed to connect to " << domain_name << " (" << buff << ")"
					" with error: " << last_socket_error() << "\n";
			}
		}

		::freeaddrinfo(results);
		if (!connected) {
			std::cout << "couldn't connect to " << domain_name << " !\n";
			close_socket(sockfd);
			return -1;
		}

		return sockfd;
	}

	bool do_ssl_handshake(const char* domain_name, bool use_tls13) {
		std::cout << " ========================= \n";
		wolfSSL_Init();
		wolfSSL_Debugging_ON();
		clean_on_exit on_exit;
		on_exit.sockfd = connect_socket(domain_name);
		if (on_exit.sockfd == -1) {
			return false;
		}
		WOLFSSL_METHOD* method = use_tls13 ? wolfTLS_client_method() 
			: wolfTLSv1_2_client_method();
		on_exit.ctx = wolfSSL_CTX_new(method);
		if (on_exit.ctx == nullptr) {
			std::cout << "failed to create a new wolfssl context !\n";
			return false;
		}

		wolfSSL_Debugging_OFF();
		wolfSSL_CTX_set_default_verify_paths(on_exit.ctx);
		wolfSSL_Debugging_ON();

		on_exit.ssl = wolfSSL_new(on_exit.ctx);
		if (on_exit.ssl == nullptr) {
			std::cout << "failed to create a new wolfssl session !\n";
			return false;
		}

		wolfSSL_set_fd(on_exit.ssl, on_exit.sockfd);
		wolfSSL_UseSNI(on_exit.ssl, WOLFSSL_SNI_HOST_NAME, domain_name, 
			static_cast<unsigned short>(strlen(domain_name)));

		const int connect_ret = wolfSSL_connect(on_exit.ssl);
		wolfSSL_Debugging_OFF();
		if (connect_ret != SSL_SUCCESS) {
			// fails here when use_tls13 is true!
			const char* error_msg = get_wolfssl_error_msg(wolfSSL_get_error(on_exit.ssl, connect_ret));
			std::cout << "failed to do handshake : " << error_msg << "\n";
			return false;
		}
		else {
			const char* version = wolfSSL_get_version(on_exit.ssl);
			std::cout << "done the handshake, protocol version: " << version << "\n";
			wolfSSL_shutdown(on_exit.ssl);
			return true;
		}
	}
}

namespace tests_fn {
	bool do_wolfssl_client_test() {
		const bool res1 = do_ssl_handshake("github.com", true);
		const bool res2 = do_ssl_handshake("github.com", false);
		return res1 && res2;
	}
}
```

The logs of the first try:

```
 =========================
trying to connect to github.com (140.82.121.3) ...
connected to github.com (140.82.121.3)
wolfSSL Entering TLS_client_method_ex
wolfSSL Entering wolfSSL_CTX_new_ex
wolfSSL Entering wolfSSL_CertManagerNew
heap param is null
DYNAMIC_TYPE_CERT_MANAGER Allocating = 256 bytes
wolfSSL Leaving wolfSSL_CTX_new_ex, return 0
wolfSSL Entering wolfSSL_new
wolfSSL Entering ReinitSSL
RNG_HEALTH_TEST_CHECK_SIZE = 128
sizeof(seedB_data)         = 128
wolfSSL Entering SetSSL_CTX
wolfSSL Entering wolfSSL_NewSession
InitSSL done. return 0 (success)
wolfSSL_new InitSSL success
wolfSSL Leaving wolfSSL_new InitSSL =, return 0
wolfSSL Entering wolfSSL_set_fd
wolfSSL Entering wolfSSL_set_read_fd
wolfSSL Leaving wolfSSL_set_read_fd, return 1
wolfSSL Entering wolfSSL_set_write_fd
wolfSSL Leaving wolfSSL_set_write_fd, return 1
TLS 1.2 or lower
wolfSSL Entering wolfSSL_connect
wolfSSL Entering ReinitSSL
wolfSSL Entering RetrySendAlert
wolfSSL Entering SendTls13ClientHello
Adding signature algorithms extension
Adding supported versions extension
wolfSSL Entering EccMakeKey
wolfSSL Leaving EccMakeKey, return 0
growing output buffer
ECH extension to write
TLSX_ECH_Write
wolfSSL error occurred, error = -173
failed to do handshake : Bad function argument
```

ycdev1, 2025-09-15:
I downloaded the master branch, and put the code above in the client project after editing it to be compatible with c and now it is working for tls 1.3 so I think it is a vcpkg port bug?
Why the client project not accepting:

```
struct clean_on_exit {
    WOLFSSL_CTX* ctx;
    WOLFSSL* ssl;
    int sockfd;
};

```
but I must write:

```
typedef struct clean_on_exit {
    WOLFSSL_CTX* ctx;
    WOLFSSL* ssl;
    int sockfd;
} clean_on_exit;
```

and the default verify load is not working so I had to disable verification.

ycdev1, 2025-09-17:
I remembered that it is Encrypted Client Hello. I didn't know wolfssl supports that.
It is safe to disable it.

embhorn, 2025-11-04:
Hi @ycdev1 

Please submit a support ticket for this issue by emailing `support@wolfssl.com`