wolfSSL/wolfssl #9214: [Bug]: configure: error: Header file inconsistency detected -- error including wolfssl/openssl/aes.h.

RAGHURH1, 2025-09-18:
### Contact Details

raghavendrarh4@gmail.com

### Version

5.8.2

### Description

Configuration error combining a) --enable-opensslall and --enable-opensslextra b) --enable-dtls13 c) --enable-mlkem.
I can't Configure all 1,2,3 options at a time.
but any 2 of 3 options works fine.
I need all three a,b,c options in my project

OS: Ubuntu 24.04 LTS 
 

### Reproduction steps

```sh
# erorr
./configure --enable-mlkem --enable-dtls --enable-dtls13 --enable-opensslall --enable-opensslextra

# working 
./configure --enable-dtls --enable-dtls13 --enable-opensslall --enable-opensslextra
./configure --enable-mlkem --enable-dtls --enable-opensslall --enable-opensslextra
./configure --enable-mlkem --enable-dtls --enable-dtls13 
```

### Relevant log output

```shell
Error output

checking whether to enable maintainer-specific portions of Makefiles... no
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking whether gcc understands -c and -o together... yes
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking target system type... x86_64-pc-linux-gnu
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports the include directive... yes (GNU style)
checking whether make supports nested variables... yes
checking whether UID '1002' is supported by ustar format... yes
checking whether GID '1002' is supported by ustar format... yes
checking how to create a ustar tar archive... gnutar
checking dependency style of gcc... gcc3
checking whether make supports nested variables... (cached) yes
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
[... 162 lines cut ...]
config.status: creating Makefile
config.status: creating wolfssl/version.h
config.status: creating wolfssl/options.h
config.status: creating support/wolfssl.pc
config.status: creating debian/control
config.status: creating debian/changelog
config.status: creating rpm/spec
config.status: creating wolfcrypt/test/test_paths.h
config.status: creating scripts/unit.test
config.status: creating config.h
config.status: config.h is unchanged
config.status: executing depfiles commands
config.status: executing libtool commands
config.status: executing wolfssl/wolfcrypt/async.h commands
config.status: executing wolfssl/wolfcrypt/fips.h commands
config.status: executing wolfssl/wolfcrypt/port/cavium/cavium_nitrox.h commands
config.status: executing wolfssl/wolfcrypt/port/intel/quickassist.h commands
config.status: executing wolfssl/wolfcrypt/port/intel/quickassist_mem.h commands
configure: ---
configure: Running make clean...
configure: ---
configure: Generating user options header...
checking for wolfssl/openssl/aes.h... no
configure: error: Header file inconsistency detected -- error including wolfssl/openssl/aes.h.
```

kareem-wolfssl, 2025-09-18:
Hi @RAGHURH1 ,

I am able to reproduce this here, I am seeing the following error in config.log:
```./wolfssl/wolfcrypt/settings.h:4309:2: error: #warning "Using DTLS 1.3 + pqc without WOLFSSL_DTLS_CH_FRAG will probably" "fail.Use --enable-dtls-frag-ch to enable it." [-Werror=cpp]```

With `--enable-dtls-frag-ch` added to my configure line, I am able to build wolfSSL: `./configure --enable-mlkem --enable-dtls --enable-dtls13 --enable-opensslall --enable-opensslextra --enable-dtls-frag-ch`

Please give it a try and let me know if it helps.

Are you able to share any information about your project?  Are you using wolfSSL in a personal or commercial project?

RAGHURH1, 2025-09-19:
> Hi [@RAGHURH1](https://github.com/RAGHURH1) ,
>
> I am able to reproduce this here, I am seeing the following error in config.log: `./wolfssl/wolfcrypt/settings.h:4309:2: error: #warning "Using DTLS [...]

Thank you sir, its working now.

Its for my personal experimental project for PQC.

closed 2025-09-19 by kareem-wolfssl